solvespace/solvespace #2: Coincident vs distance = 0

whitequark, 2015-04-25:
I've noticed that in some cases, coincident constraint on points doesn't work whereas setting distance to 0 does... but only before the first regeneration. This looks like a bug.

The case where I most often encounter this is setting difference extrusion depth. This is actually very tricky to do without overconstraining the sketch if you want to make it using constraints and not just drag and leave it wherever.

whitequark, 2015-04-25:
Sample file: http://rghost.net/7LXlrP7xn.

jwesthues, 2015-04-25:
Point-point distance (or length) of zero won't usually do anything good. In general, that constraint removes one DOF, but when the distance is zero it removes three or two (in 3d/2d). The solver currently writes an equation like x^2 + y^2 = 0 in that case, which attempts to constraint multiple degrees of freedom in a single equation, which doesn't work.

That could be modified to write a different set of equations when the distance was below a tolerance, but that's potentially confusing too, as the structure of the sketch changes with what appears to be just a parameter change. The best solution would perhaps be just to warn when the user tries to enter that.

How else are you trying to constrain the extrusion? That depth is 1 DOF, so point-on-point or point-on-line in 3d overconstrains. Point-on-face (or point-on-line projected into 2d) should work, though.

A greater range of constraints would work if the solver permitted redundant but not inconsistent equations--basically, if it were okay for the Jacobian to not have full rank, as long as we could still satisfy all the equations to within some tolerance. That's less geometrically clean, but it may be closer to what people expect.

whitequark, 2015-04-25:
Ah, I thought it'd be something like this. Warning the user makes sense. The issue is more with complexity of constraining extrusions than the distance = 0, it's just that setting distance to 0 appeared to momentarily work for my case.

It's trivial in this case (just constrain it onto a face), but lathe groups don't have faces yet, and there are other cases. For example, I needed to make a hole in a wall of a cylinder that goes through the entire depth of the wall. Constraining that so that the sketch is not overconstrained requires a ridiculous amount of construction.

(As a side note, why does constraining a point in 3D to a circle that is the result of extrusion actually constrains it to the cylinder going through that circle?)

I think allowing overconstrained but consistent sketches would be a good possibility. I'm not sure if this is something that should be enabled by default; especially in 2D, overconstraining is usually a really good indication that something's wrong with your understanding of the geometry you're trying to define. In 3D however and also in 2D if you use symmetry constraints heavily, it's more trouble than it's worth, in my experience.

Perhaps a per-group solver setting that's off by default for 2D and on for 3D?

whitequark, 2015-04-25:
After reflection, I really like the idea of that setting. I often get to fight the solver when constraining 3D assemblies--especially in the cases where "same orientation" is too restrictive, so I use a parallel constraint on the normal, but then it can rotate around it and often there's nothing handy to constrain the rotation against. And I think there were some other cases I can't immediately recall in detail.

jwesthues, 2015-04-27:
It seems reasonable. The basic change would be to permit a row of zeros when we orthogonalize the Jacobian in CalculateRank(), as long as the same sequence of row operations applied to the b vector would also result in zero. It shouldn't be too hard to set a tolerance on that, since the equations already all return something roughly like a length. The numerical implications of that downstream (or the correct heuristic for choosing which equations to discard) would be the hard part.

whitequark, 2015-04-28:
I see. I'm afraid that's out of my depth.

Evil-Spirit, 2015-07-18:
i have many ideas how to improve sketch solving. for example, it is good idea to drag sketch through something like distance=0 or fix in new mouse position instead of directly parameter chaneging. this is more stable, but causes dof problems sometimes. also, for point=point you can firstly add distance=0 constraint, when it solved ok just substitute it with x=x, y=y for more PERFOMANCE.

Evil-Spirit, 2015-07-18:
Also we have tried pseudo-solutions for linear systems, but pofit it is not yet compleatley understood.

jwesthues, 2015-07-18:
There certainly are ways to improve convergence of the solver. I don't see how writing distance=0 (i.e., (xb^2 - xa^2) + (yb^2 - ya^2) = 0 instead of xa = xb and ya = yb) helps, though. That's one equation that attempts to fix 2 DOF, and the linearized constraint goes to zero as the constraint is satisfied. If you really have a sketch where that "helps", then I'd guess that something is badly wrong elsewhere.

Evil-Spirit, 2015-07-19:
Mysterious are the ways of The Solver. So, distance=0 can have better convergene not because of its formula but because it goes into Jacobian matrix. so, at the moment of a constraint creation, we can add equation x=x, y=y to solve using Jacobian, so it converges, after it we can use substitution again.

Evil-Spirit, 2015-07-19:
The main idea is to start from parameters which satsfy all the equations except one, not from many broken contraints how it performed when dragging points directly changes coordintes, or when equations solved by substitution.

jwesthues, 2015-07-19:
Can you post an example that exhibits this behavior? As dx^2 + dy^2 = 0 gets satisfied, dx and dy must both go to zero, which means that the linearized constraint (i.e., the corresponding row in the Jacobian) also goes to zero, and the Jacobian goes singular. So perhaps something unexpected is happening, but it doesn't make sense for that to do anything but hurt.

I also don't see why solving xa = xb in the Jacobian would perform better than consolidating those by forward-substitution. Ignoring rounding error in the linear system solve, they're exactly equivalent. That rounding error is (a) generally insignificant, and (b) if significant, generally bad.

Evil-Spirit, 2015-07-19:
at the moment all old files are dead, someone broken file format. so i cant recall this. when i get it again, you will be first who will know. but generally i dont speak about dx^2 + dy^2 = 0 but speak about sketch-modifing stratagy - allways start from maximally converged parameters. it means when you need to drag point, you should not to modify cordinates directly & solve sketch after it, you should apply new constraint (like as in "where dragged" constraint) and solve sketch. if you need to apply new constraint, you need to add equation in Jacobian anyway (even if it can be solved by substitution). because if you add it like substitution, it cause sketch constraints not converged at the start point of solving.

Evil-Spirit, 2015-07-19:
also, you have "step dimension". all dimensions can be parameters which added to system of equations when you trying to modify it with the corresponding equation. & every label value changeing can be performed like this.
also you can perform satisfing of constraints through solve (actually works for me)

jwesthues, 2015-07-19:
I think I get what you're saying about substitution vs. Newton-solving. You mean where xa is initially not equal to xb, like when a point-coincident constraint is first applied to two non-coincident constraints? In that case, I agree that the behavior will be different--substitution requires you to pick a single initial value, but the Newton's method will work its way in from both.

When a point with 1 DOF gets dragged, are you writing two equations (and accepting that it won't generally be possible to satisfy both), or a single equation that projects the mouse motion onto a fixed line (with that line calculated, for example, based on the previous solution), or a single nonlinear equation?

"Step dimension" or anything else that breaks a big change into many little changes is definitely helpful. I don't do it automatically because it's slow, but that will pretty much always improve convergence.

Evil-Spirit, 2015-07-19:
yes, you have get it. substitution will work unpredictable way, you dont know how will it shoot - xa will be xb, or xb will be xa. so, you have at least two variants. & you should consider, you will get not converged equations which envolved xa when it turns to xb, or vice versa. so, i think it is better to make initial solve different way when constraint just created.

Evil-Spirit, 2015-07-19:
so, what about how to perform sketch dragging - at now i dont understand completely. i have tried to make it through x - mouse.x, y-mouse.y & get surprising results - at most cases all working fine, more natural way. but sometimes, when i need to drag 1dof objects i get didnt converge result. so, this contraint still must be variable (not rigid, not strict).

Evil-Spirit, 2015-07-19:
ok now i understand what you have written about 1dof, so tomorrow i will try different approaches for dragged. thank you.

jwesthues, 2015-07-19:
Well, forward-substitution works very predictably; it's just a question of the code that picks which initial value to take. When we drag, for example, we always take the dragged point and discard the other one(s). That doesn't lend itself to "step dimension" style convergence aids, though.

dx^2 + dy^2 = 0 is interesting because the linearized constraint does the right thing until the constraint is satisfied. So it can get you closer to a solution, but never all the way. The solver needs to know in what direction to move the point, and until the points are coincident, the answer is "towards each other"; but once they're coincident, that state is fundamentally lost.

A constraint on distance instead of d^2 (which is what we actually write) does something slightly better, because the derivatives have the form dx/sqrt(dx^2 + dy^2), with both the numerator and the denominator going to zero. So as the constraint is satisfied, the slope along which the point is constrained turns to noise (but not zero). That potentially causes the point to move away from a correct solution; but if it moves far enough away, then the constraint starts to work again. So that seems like it could perhaps be made practically convergent, but it's icky. In particular, if that direction ends up perpendicular to the point's actual degree of freedom (because the user dragged it that way, or because of numerical noise in a nearly-satisfied constraint), then the Jacobian will lose rank, but that shouldn't be reported as an error; the point just shouldn't move.

You could also keep state on that direction outside the parameters (for example, by recording the direction the last time the distance was greater than some threshold, or computing a running average weighted by that distance), and blend from (dx, dy) calculated from the parameters to that saved (dx, dy) as the distance constrained gets small. That seems like it avoids the fundamental problem, but it's a weird bolt-on. Or, at least for dragging with the mouse (where imprecision is okay), you could just use a fixed slope based on the initial drag direction. I looked at various approaches of this form when I was first writing the solvers; but that was a long time ago now.

Evil-Spirit, 2015-07-19:
probably we ve started from the version, where you use sqrt(dx^2+dy^2)-d equation for the distance. i have many experimets with squared and normal distances, but still not understand which is better. normal distance reaches converge in only one step, but if you have many constraints, it will cause large jumps far away from solution (i made visualisation inside newton step to draw solver path with lines). example is triangle with constrained point distances. the function is like cone https://www.google.ru/webhp?sourceid=chrome-instant&ion=1&espv=2&ie=UTF-8#newwindow=1&q=sqrt(x%5E2+%2B+y%5E2)+-+10. solution is intersection with zero plane. derivative will fall us directly in solution. another case is length square equation https://www.google.ru/webhp?sourceid=chrome-instant&ion=1&espv=2&ie=UTF-8#newwindow=1&q=x%5E2+%2B+y%5E2+-+100. this is like we have to perform some newton steps to reach solution. but in practics, jumps from solution not so big as in one-step sqrt(dx^2 + dy^2) - 10. So I am confused.  One-step solution is not better not because of result, because of the way it was reached. Simple expressions is not better than complex... an so on. Ive exprimentating with solver so much last six months :)

jwesthues, 2015-07-19:
The basins of attraction of a Newton's method are classic fractals, after all. In this application, it's quite unfortunate.

Are you doing the equivalent of "step dimension" concurrently with the Newton's method yet, without rewriting the symbolic Jacobian (or necessarily converging the solution all the way to our usual tolerance) at each step? That's a noticeable speedup that I never got around to making for real.

Evil-Spirit, 2015-07-19:
Can you form your question another way?  for visualisation i draw point coordinates (parameters) at each newton solver step. For trying to solve system any way, i have made something like "step dimension" but for dragged weight. so through this process, i dont have to rewrite symbolic expressions because i dont use unstable equations where you need to calculate something to choose the right eq (like parallel equation you have use) . for parallel, perpendicular & angle i use atan2. not so fast, but something near stable.

jwesthues, 2015-07-19:
I mean something similar to what you're saying, I think; but you also don't necessarily have to converge the Newton's method all the way before stepping the dimension.

So what are you writing for a 3d parallel constraint? That's the only constraint where the symbolic equations written are a function of the initial parameter values, and I'm pretty sure the hairy ball theorem makes a discontinuous function necessary there.

Evil-Spirit, 2015-07-20:
3d angle (as parallel and perpendicular) now written through atan2(length(cross(a, b), dot(a, b))
dont test enough it yet.

jwesthues, 2015-07-20:
3d angle generally constrains one DOF, but 3d parallel is a special case that constrains two; so a second equation is needed, and I'm pretty sure those equations must contain a discontinuity. Do you (a) see why parallel is a special case in 3d, and (b) see why the hairy ball theorem is relevant?

Evil-Spirit, 2015-07-20:
no, so i need to discuss this with our math-man. he loves such things)

jwesthues, 2015-07-20:
(a) is pretty intuitive. If you think of all the vectors that make some angle &theta; with some axis, then if:
* &theta; = 90&deg;: they sweep out a plane (i.e., a surface, 2 DOF remaining)
* 0&deg; < &theta; < 90&deg;: they sweep out a cone (still a surface, 2 DOF)
* &theta; = 0&deg;: they sweep out a line (1 DOF)

(b) takes a bit more thought...

Evil-Spirit, 2015-07-20:
ok, ive tested. i can assmble box in sketch 3d using parallel, perpendicular and this is hard but solveable

Evil-Spirit, 2015-07-20:
double getAngle3d(const dVec3 &d0, const dVec3 &d1) {
		return atan2(length(cross(d0, d1)), dot(d0, d1));
		//return acos(dot(d0, d1) / (length(d0) * length(d1)));
	}

Evil-Spirit, 2015-07-20:
Exp expressionAngle3d(const ExprVector &d0, const ExprVector &d1) {
		return atan2(length(cross(d0, d1)), dot(d0, d1));
		//return acos(dot(d0, d1) / (length(d0) * length(d1)));
	}

Evil-Spirit, 2015-07-20:
<math><mrow><mrow><mi>atan2</mi><mfenced><mrow><msqrt><mrow><mrow><msup><mfenced><mrow><mrow><mfenced><mrow><mi>py</mi><mo>-</mo><mi>py</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>pz</mi><mo>-</mo><mi>pz</mi></mrow></mfenced></mrow><mo>-</mo><mrow><mfenced><mrow><mi>pz</mi><mo>-</mo><mi>pz</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>py</mi><mo>-</mo><mi>py</mi></mrow></mfenced></mrow></mrow></mfenced><mn>2</mn></msup><mo>+</mo><msup><mfenced><mrow><mrow><mfenced><mrow><mi>pz</mi><mo>-</mo><mi>pz</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>px</mi><mo>-</mo><mi>px</mi></mrow></mfenced></mrow><mo>-</mo><mrow><mfenced><mrow><mi>px</mi><mo>-</mo><mi>px</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>pz</mi><mo>-</mo><mi>pz</mi></mrow></mfenced></mrow></mrow></mfenced><mn>2</mn></msup></mrow><mo>+</mo><msup><mfenced><mrow><mrow><mfenced><mrow><mi>px</mi><mo>-</mo><mi>px</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>py</mi><mo>-</mo><mi>py</mi></mrow></mfenced></mrow><mo>-</mo><mrow><mfenced><mrow><mi>py</mi><mo>-</mo><mi>py</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>px</mi><mo>-</mo><mi>px</mi></mrow></mfenced></mrow></mrow></mfenced><mn>2</mn></msup></mrow></msqrt><mo>,</mo><mrow><mrow><mrow><mfenced><mrow><mi>px</mi><mo>-</mo><mi>px</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>px</mi><mo>-</mo><mi>px</mi></mrow></mfenced></mrow><mo>+</mo><mrow><mfenced><mrow><mi>py</mi><mo>-</mo><mi>py</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>py</mi><mo>-</mo><mi>py</mi></mrow></mfenced></mrow></mrow><mo>+</mo><mrow><mfenced><mrow><mi>pz</mi><mo>-</mo><mi>pz</mi></mrow></mfenced><mo>*</mo><mfenced><mrow><mi>pz</mi><mo>-</mo><mi>pz</mi></mrow></mfenced></mrow></mrow></mrow></mfenced></mrow><mo>-</mo><mn>1.5708</mn></mrow></math>

Evil-Spirit, 2015-07-20:
atan2(sqrt(sqr((py - py) * (pz - pz) - (pz - pz) * (py - py)) + sqr((pz - pz) * (px - px) - (px - px) * (pz - pz)) + sqr((px - px) * (py - py) - (py - py) * (px - px))), (px - px) * (px - px) + (py - py) * (py - py) + (pz - pz) * (pz - pz)) - 1.5708

Evil-Spirit, 2015-10-06:
i have made some improvements about atan2 angle (parallel, perpendicular, angle) with signed angle preserving relative position. 
also, i ve made some improvements about 
https://github.com/jwesthues/solvespace/issues/2#issuecomment-122671870
the repo
https://github.com/Evil-Spirit/solvespace-1

whitequark, 2016-01-22:
Setting mentioned in https://github.com/jwesthues/solvespace/issues/2#issuecomment-96240683 implemented in whitequark/solvespace@46ab541444bdf4c9a16af3057c043ca4507bc63e.

closed 2016-01-22 by whitequark

ruevs, 2020-10-23:
Note to self - wisdom lies in here :-)