solvespace/solvespace #5: Mac OS X: Occasional SIGSEGV on mouse (Magic Trackpad) move

michthom, 2016-05-28:
Process:               solvespace [30608]
Path:                  /Applications/solvespace.app/Contents/MacOS/solvespace
Identifier:            solvespace
Version:               2.1 (2.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           solvespace [30608]
User ID:               502

Date/Time:             2016-05-29 00:24:12.402 +0100
OS Version:            Mac OS X 10.11.5 (15F34)
Report Version:        11
Anonymous UUID:        77A06C71-7219-FFF3-7683-72B79C8D32EE

Sleep/Wake UUID:       9B50071F-8246-4CC9-B847-AF35F9AB8A65

Time Awake Since Boot: 140000 seconds
Time Since Wake:       2000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000002000000020
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x2000000020:
    CoreAnimation          000000098dd22000-00000009913a1000 [ 54.5M] rw-/rwx SM=PRV  
--> 
    STACK GUARD            0000700000083000-0000700000084000 [    4K] ---/rwx SM=NUL  stack guard for thread 1

Global Trace Buffer (reverse chronological seconds):
464.718595   libextension.dylib        	0x00007fff91303533 tearing down extension request for pid 30645

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	0x0000002000000020 0 + 137438953504
1   solvespace                    	0x000000010a460376 SolveSpace::TextWindow::MouseEvent(bool, bool, double, double) + 566
2   solvespace                    	0x000000010a3ddd44 -[TextWindowView mouseMoved:] + 100
3   com.apple.AppKit              	0x00007fff937131e4 -[NSTrackingArea mouseMoved:] + 152
4   com.apple.AppKit              	0x00007fff93c99bd1 -[NSWindow _handleMouseMovedEvent:] + 266
5   com.apple.AppKit              	0x00007fff93c9ab15 -[NSWindow _reallySendEvent:isDelayedEvent:] + 2108
6   com.apple.AppKit              	0x00007fff936d9539 -[NSWindow sendEvent:] + 517
7   com.apple.AppKit              	0x00007fff936594db -[NSApplication sendEvent:] + 1167
8   com.apple.AppKit              	0x00007fff934c0df2 -[NSApplication run] + 796
9   solvespace                    	0x000000010a3df458 main + 296
10  solvespace                    	0x000000010a3c2e54 start + 52

whitequark, 2016-05-28:
I don't have any Apple hardware at all. Is this related just to pointer movement? Can it be reproduced somehow?

michthom, 2016-05-29:
I've not found a reproducible scenario - might be tied to the multi-finger gestures on the trackpad - I will see if using a 'regular' mouse is less prone to tickling the bug.

whitequark, 2016-05-29:
Any chance you could run SolveSpace under a debugger? It should have all debugging symbols, and give a far more informative backtrace, which would confirm some ideas I have about the cause of this bug.

michthom, 2016-05-29:
I'll see what I can do - I have Xcode installed, so should have LLDB - but no clue how to operate it! Will try reading this first: https://developer.apple.com/library/mac/documentation/IDEs/Conceptual/gdb_to_lldb_transition_guide/document/lldb-terminal-workflow-tutorial.html#//apple_ref/doc/uid/TP40012917-CH4-SW1

michthom, 2016-05-29:
michthom$ **lldb /Applications/solvespace.app/Contents/MacOS/solvespace**
(lldb) **process launch**

michthom, 2016-06-01:
Process 38892 launched: '/Applications/solvespace.app/Contents/MacOS/solvespace' (x86_64)
failed to look up item 00000000, searched 27 items
oops at line 302, file /Users/travis/build/solvespace/solvespace/src/dsc.h

Process 38892 exited with status = 255 (0x000000ff)

whitequark, 2016-06-11:
@michthom Unfortunately that still doesn't have a backtrace...

whitequark, 2016-06-11:
This apparently can be grabbed using the `thread backtrace` lldb command.

michthom, 2016-06-13:
Thanks - I'll see if I can get more details with this. i had thought that 'bt' would generate a backtrace, but the stop hook seems to be ignored when the app dies - perhaps it only fires on a breakpoint being hit?

whitequark, 2016-06-23:
@michthom Sorry, missed your last comment. I now understand your problem--lldb doesn't stop on abort. I think that a more robust way to fix this, rather than messing with hooks, is to simply set a breakpoint on `abort`, which would let you grab a backtrace from there.

michthom, 2016-06-25:
Aha.
```
(lldb) breakpoint set -b abort
Breakpoint 1: where = libsystem_c.dylib`abort, address = 0x000000000005e666

```
However, the backtrace that got me here in the first place didn't actually *call* abort, so I may use both this and the technique in my previous comment.

michthom, 2016-06-25:
I'm feeling veeeeery stupid at this point, but are the DSYM files included in the 2.1 dmg? I'm looking under /Applications/solvespace.app/Contents/Resources/ and I can't see anything apart from some nibs, icons and a png?
```
Mikes-2015-iMac:Resources michthom$ pwd
/Applications/solvespace.app/Contents/Resources
Mikes-2015-iMac:Resources michthom$ ls -l
total 160
-rw-r--r--@ 1 michthom  admin  63141 17 May 17:59 AppIcon.icns
-rw-r--r--@ 1 michthom  admin   4021 17 May 17:59 MainMenu.nib
-rw-r--r--@ 1 michthom  admin   4941 17 May 17:59 SaveFormatAccessory.nib
-rw-r--r--@ 1 michthom  admin    352 17 May 17:59 solvespace-48x48.png

```
So when I load a core file into lldb I'm not sure where it's going to get the debugging symbols from?
Or have I been dumb and need to grab the source and build it myself to get the symbols to give you any useful feedback?

whitequark, 2016-06-26:
> I'm feeling veeeeery stupid at this point, but are the DSYM files included in the 2.1 dmg?

I think they should be in the executable. The CI builds are performed with CMAKE_BUILD_TYPE as either Debug or RelWithDebInfo, and the default build type is equivalent to Debug, so you should have the symbols. No idea what dSYM stuff is about.

whitequark, 2016-06-26:
Oh, looks like dSYM is for *stripped* symbols. I.e. you take your proprietary executable and separate the debugging symbols into another file, which you save, and ship the executable. They are inapplicable in case of SolveSpace.

lucdesign, 2016-06-26:
same error. crashes while following the tutorial. in this form unusable for me.

whitequark, 2016-06-27:
@michthom Ok, I've figured this out. On OS X (only?), you have to build with -DCMAKE_BUILD_TYPE=Debug to get any debug symbols at all.

@lucdesign As I've mentioned above, I do not have any Apple hardware and cannot reproduce this on my own. You can help me fix this by providing a symbolized backtrace, by following the [build instructions](https://github.com/solvespace/solvespace#building-on-mac-os-x) (replacing `cmake ..` with `cmake .. -DCMAKE_BUILD_TYPE=Debug`), running `lldb build/src/solvespace.app`, then in LLDB prompt entering `run` and `bt` once it crashes.

michthom, 2016-06-29:
Sorry I've been away on business, will rebuild a debug version and see if I can reproduce soon.

whitequark, 2016-07-19:
@michthom ping

michthom, 2016-07-19:
Argh! Sorry - I've been swamped with work. I have a debug version, I have been able to get it to crash and see the backtrace with symbols, but I've no idea what I was doing at the time. I'll have more time to use SolveSpace in the coming weeks, if you're willing to wait, else close this one and I'll reopen once I have some debug info to provide?

whitequark, 2016-07-20:
@michthom Sure--I'll wait. No particular rush here, just making sure this issue doesn't die out as it seems serious and I've no means to reproduce it myself.

whitequark, 2016-08-07:
@michthom Any luck with that backtrace? It shouldn't take too long... I would like to get this fixed in 2.2, which is scheduled in 1-2 weeks from now, as there will likely be no release for a long time after that.

michthom, 2016-08-07:
Sorry I've failed. My mac is with Apple trying to debug a random freeze/crash and I'm currently on holiday, back August 15th so I think you'll have to go without me for now.

whitequark, 2016-10-07:
@michthom ping?

michthom, 2016-10-07:
<blush> Sorry - this task has rather slipped my mind. I'm still battling with Apple - they're currently debating replacement of my iMac as it has got worse not better after three trips to the Genius Bar.
I really don't think I can commit to helping out in a reasonable timescale, so if you're waiting for me to help find that crash bug before releasing 2.2, it may be some time off yet...
If on the other hand you have the patience of Job I will pick this up once I have a stable machine and can spend enough time in a debug build of Solvespace to reproduce the issue. Not having a set of steps that triggers it reliably, it'll be me flailing in the dark for a while before it happens again.

whitequark, 2016-10-08:
@michthom Understood--I wish you luck with your troubles. It's not that I am impatient, this is just normal triage process, making sure that the project is going forward.

whitequark, 2016-10-10:
Fixed in master and 2.x (via #63).

closed 2016-10-10 by whitequark