solvespace/solvespace #6: Add 32bit binary deb's in release

Symbian9, 2016-06-12:
Please, make 32bit binary too for current and next releases!

whitequark, 2016-06-12:
This (and any other weird architecture you might like to run it on) will be handled by the Debian packagers, which are already working on getting 2.1 into Debian repositories. The Debian packages provided through SolveSpace's build infrastructure are not, and cannot, cover all the range of Linux distributions and the different configurations they use.

There are also technical problems with building 32-bit packages on Travis, which we use.

whitequark, 2016-06-12:
Why do you need 32-bit binaries anyway? There's almost no 32-bit x86 hardware left.

closed 2016-06-12 by whitequark

Symbian9, 2016-06-12:
Many notebooks and PC's, used by Linux users still have 32bit architectures. Also most PC's in schools and universities still not upgraded to 64bit.
Also application on 64bit architectures need more hardware resources then same apps on 32bit hardware.

Anyway, can you setup OpenSuse OBS, like it work for http://github.com/OpenOrienteering/mapper project ?
* http://download.opensuse.org/repositories/home:/dg0yt/

Or, maybe, you can back in deal Launchpad repo of SolveSpace.
I very hope, that you not break down 32bit.
I use latest PuppyLinux 32bit based on Trusty, and this linux distr give more speed for my notebook.

whitequark, 2016-06-12:
OBS doesn't help a lot, and Launchpad requires me to do manual uploads every time. I would really rather do CAD development and not CAD packaging.