solvespace/solvespace #8: Dependency path problems on Mac OS X

xythobuz, 2016-06-12:
I have all required dependencies installed using MacPorts, which places them in `opt/local/...`. I also have the `Mono.framework` installed on my system, which includes libpng and freetype.

After running cmake, it is in a strange state where it uses the headers from `Mono.framework` and the libraries from `/opt/local/lib`, causing run-time errors with libpng because of mismatching versions.

The only way I have found to fix this is manually editing `build/CMakeCache.txt` after running cmake and replacing all occurrences of:

    /Library/Frameworks/Mono.framework/Headers

with

    /opt/local/include

I then had to run cmake again to recreate the makefiles. Now, the resulting binary works fine.

I'm not sure where exactly the problem is, or how it could be fixed, but I guess this is the best place to document my solution if anyone stumbles upon the same problem in the future.

whitequark, 2016-06-12:
Yeah, I've noticed this before. I think one solution would be to prioritize Homebrew and MacPorts search paths over anything else, let me see how that can be done.

mistydemeo, 2016-06-13:
Cmake has an option to do this; the `-DCMAKE_FIND_FRAMEWORK=LAST` switch causes it to deprioritize frameworks, so it'll use the Macports versions of the libraries first.

whitequark, 2016-06-13:
@mistydemeo Thanks!

@xythobuz I've committed that in master and 2.x (inside CMakeLists.txt). Please reopen if the problem persists.

xythobuz, 2016-06-19:
I still have to add the `-DCMAKE_FIND_FRAMEWORK=LAST` flag manually. I think there's a typo, it should be `FRAMEWORK`, not `FRAMEWORKS`... And I can't reopen this issue.

whitequark, 2016-06-19:
Oops. Does e0283b2 fix it?

xythobuz, 2016-06-19:
Yes, it works now!

closed 2016-06-19 by xythobuz