solvespace/solvespace #9: CI / Travis integration / auto-stats

twitchyliquid64, 2016-06-19:
Any reason there isnt CI setup ATM?
If not, I might try setting on up.

whitequark, 2016-06-19:
There is... https://travis-ci.org/solvespace/solvespace https://ci.appveyor.com/project/whitequark/solvespace

twitchyliquid64, 2016-06-19:
Kek, might wanna get these ones going then

https://docs.travis-ci.com/user/status-images/ etc

whitequark, 2016-06-19:
I don't see any point in those. They seem to be cargo-culted around open-source projects despite not providing any useful information. As a user of the project there is absolutely no reason for you to care about CI and as a contributor, how often do you look at those badges in README? Do you ever?

closed 2016-06-19 by whitequark

twitchyliquid64, 2016-06-19:
Disagree. The badge itself provides credability that the project follows TDD and has some kind of indicative QA around the state of the build.

whitequark, 2016-06-19:
See, you are making an excellent example. SolveSpace doesn't even *have tests*. All that the CI being green is currently demonstrating is that the packages can be built.

twitchyliquid64, 2016-06-19:
Oh. Welp.

whitequark, 2016-11-18:
(SolveSpace has tests now.)