solvespace/solvespace #11: SIGABRT when exporting DXF

sigxcpu76, 2016-06-22:
```
Process:               solvespace [80710]
Path:                  /Applications/solvespace.app/Contents/MacOS/solvespace
Identifier:            solvespace
Version:               3.0 (3.0~e0283b2d)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           solvespace [80710]
User ID:               501

Date/Time:             2016-06-22 19:05:55.643 +0300
OS Version:            Mac OS X 10.11.5 (15F34)
Report Version:        11
Anonymous UUID:        D5F1B9B3-8243-2CC6-6E35-3EEC052C8A6A

Sleep/Wake UUID:       FE8B54CA-06C8-4F55-8C53-B4081747805F

Time Awake Since Boot: 1100000 seconds
Time Since Wake:       29000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Global Trace Buffer (reverse chronological seconds):
18446744028.618950 libextension.dylib        	0x00007fff9903b533 tearing down extension request for pid 80711
18446744050.065845 libextension.dylib        	0x00007fff9903b533 tearing down extension request for pid 80711
18446744055.060570 libextension.dylib        	0x00007fff9903b533 tearing down extension request for pid 80711

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8dfc3f06 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff8890b4ec pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff9231f6e7 abort + 129
3   solvespace                    	0x0000000106384973 SolveSpace::assert_failure(char const*, unsigned int, char const*, char const*, char const*) + 371
4   solvespace                    	0x00000001063e60b4 DxfWriteInterface::writeLTypes() + 9380
5   solvespace                    	0x000000010664dfa5 dxfRW::writeTables() + 10901
6   solvespace                    	0x000000010664a62d dxfRW::write(DRW_Interface*, DRW::Version, bool) + 5421
7   solvespace                    	0x00000001063da5bc SolveSpace::DxfFileWriter::FinishAndCloseFile() + 396
8   solvespace                    	0x00000001063d38ba SolveSpace::VectorFileWriter::OutputLinesAndMesh(SolveSpace::SBezierLoopSetSet*, SolveSpace::SMesh*) + 1914
9   solvespace                    	0x00000001063d25c2 SolveSpace::SolveSpaceUI::ExportLinesAndMesh(SolveSpace::SEdgeList*, SolveSpace::SBezierList*, SolveSpace::SMesh*, SolveSpace::Vector, SolveSpace::Vector, SolveSpace::Vector, SolveSpace::Vector, double, SolveSpace::VectorFileWriter*) + 11346
10  solvespace                    	0x00000001063d2de1 SolveSpace::SolveSpaceUI::ExportViewOrWireframeTo(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) + 1985
11  solvespace                    	0x00000001064c01d2 SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) + 3026
12  solvespace                    	0x000000010638a54c +[MainMenuResponder handleStatic:] + 140
13  libsystem_trace.dylib         	0x00007fff8c23b07a _os_activity_initiate + 75
14  com.apple.AppKit              	0x00007fff82fdcdbd -[NSApplication sendAction:to:from:] + 460
15  com.apple.AppKit              	0x00007fff82fdcb57 -[NSMenuItem _corePerformAction] + 336
16  com.apple.AppKit              	0x00007fff82fdc8b7 -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:] + 114
17  libsystem_trace.dylib         	0x00007fff8c23b07a _os_activity_initiate + 75
18  com.apple.AppKit              	0x00007fff8306ee54 -[NSMenu performActionForItemAtIndex:] + 131
19  com.apple.AppKit              	0x00007fff8306edc7 -[NSMenu _internalPerformActionForItemAtIndex:] + 35
20  com.apple.AppKit              	0x00007fff8306ec1f -[NSCarbonMenuImpl _carbonCommandProcessEvent:handlerCallRef:] + 107
21  com.apple.AppKit              	0x00007fff82f13fd9 NSSLMMenuEventHandler + 708
22  com.apple.HIToolbox           	0x00007fff935807be DispatchEventToHandlers(EventTargetRec*, OpaqueEventRef*, HandlerCallRec*) + 1231
23  com.apple.HIToolbox           	0x00007fff9357fc48 SendEventToEventTargetInternal(OpaqueEventRef*, OpaqueEventTargetRef*, HandlerCallRec*) + 404
24  com.apple.HIToolbox           	0x00007fff935959e6 SendEventToEventTarget + 40
25  com.apple.HIToolbox           	0x00007fff935df99a SendHICommandEvent(unsigned int, HICommand const*, unsigned int, unsigned int, unsigned char, void const*, OpaqueEventTargetRef*, OpaqueEventTargetRef*, OpaqueEventRef**) + 411
26  com.apple.HIToolbox           	0x00007fff9360ad5b SendMenuCommandWithContextAndModifiers + 59
27  com.apple.HIToolbox           	0x00007fff9360ad0c SendMenuItemSelectedEvent + 188
28  com.apple.HIToolbox           	0x00007fff9360abe8 FinishMenuSelection(SelectionData*, MenuResult*, MenuResult*) + 96
29  com.apple.HIToolbox           	0x00007fff9360b596 MenuSelectCore(MenuData*, Point, double, unsigned int, OpaqueMenuRef**, unsigned short*) + 711
30  com.apple.HIToolbox           	0x00007fff9360b230 _HandleMenuSelection2 + 460
31  com.apple.AppKit              	0x00007fff82eff0fa _NSHandleCarbonMenuEvent + 277
32  com.apple.AppKit              	0x00007fff82d7413d _DPSNextEvent + 1906
33  com.apple.AppKit              	0x00007fff82d73226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
34  com.apple.AppKit              	0x00007fff82d67d80 -[NSApplication run] + 682
35  solvespace                    	0x0000000106391bf5 main + 437
36  libdyld.dylib                 	0x00007fff887ef5ad start + 1

```

whitequark, 2016-06-23:
Yeah, known issue on master, please use 2.1 for now.

michthom, 2016-06-25:
...but please leave a comment on how you obtained the aborted thread backtrace as I'm currently too stupid to get one for my bug!

sigxcpu76, 2016-06-25:
I've just copied the top of the text in OS X crash window, no big deal. If you disabled your crash windows, you should find the crash logs in Console.app

Alex

whitequark, 2016-06-29:
Fixed in da1fc3f.

closed 2016-06-29 by whitequark