solvespace/solvespace #12: Crash on viewing anything in g001-#references

kamnxt, 2016-06-23:
To reproduce:
Click on g001-#references
Mouse over or click on r001-#XY or r002-#YZ or r003-#ZX

Running Arch Linux, solvespace compiled from source.

kamnxt, 2016-06-23:
Output:

```
SolveSpace!

File /home/kamnxt/solvespace/src/drawentity.cpp, line 236, function GetReferencePos:
Assertion 'Unexpected entity type' failed: ((false) == false).
Backtrace:
 0: ./src/solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x72) [0x55a60a]
 1: ./src/solvespace(_ZN10SolveSpace6Entity15GetReferencePosEv+0x20a) [0x58bd34]
 2: ./src/solvespace(_ZN10SolveSpace14GraphicsWindow9Selection4DrawEv+0xcf) [0x57b183]
 3: ./src/solvespace(_ZN10SolveSpace14GraphicsWindow5PaintEv+0x19ee) [0x57e88c]
 4: ./src/solvespace(_ZN10SolveSpace14GraphicsWidget10on_gl_drawEv+0x16) [0x560cb6]
 5: ./src/solvespace(_ZN10SolveSpace8GlWidget15on_expose_eventEP15_GdkEventExpose+0x17f) [0x56003f]
 6: /usr/lib/libgtkmm-2.4.so.1(_ZN3Gtk12Widget_Class21expose_event_callbackEP10_GtkWidgetP15_GdkEventExpose+0x54) [0x7f03853094e4]
 7: /usr/lib/libgtk-x11-2.0.so.0(+0x13191c) [0x7f03848b191c]
 8: /usr/lib/libgobject-2.0.so.0(g_closure_invoke+0x145) [0x7f038351afa5]
 9: /usr/lib/libgobject-2.0.so.0(+0x223ad) [0x7f038352d3ad]
10: /usr/lib/libgobject-2.0.so.0(g_signal_emit_valist+0x8df) [0x7f03835356bf]
11: /usr/lib/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7f0383535fff]
12: /usr/lib/libgtk-x11-2.0.so.0(+0x2492dc) [0x7f03849c92dc]
13: /usr/lib/libgtk-x11-2.0.so.0(gtk_main_do_event+0x532) [0x7f03848b06d2]
14: /usr/lib/libgdk-x11-2.0.so.0(+0x422ff) [0x7f03842bf2ff]
15: /usr/lib/libgdk-x11-2.0.so.0(+0x422a5) [0x7f03842bf2a5]
16: /usr/lib/libgdk-x11-2.0.so.0(+0x3edd1) [0x7f03842bbdd1]
17: /usr/lib/libgdk-x11-2.0.so.0(gdk_window_process_all_updates+0x118) [0x7f03842bc708]
18: /usr/lib/libgdk-x11-2.0.so.0(+0x3f769) [0x7f03842bc769]
19: /usr/lib/libgdk-x11-2.0.so.0(+0x1e4e7) [0x7f038429b4e7]
20: /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x15a) [0x7f037d762c8a]
21: /usr/lib/libglib-2.0.so.0(+0x4a040) [0x7f037d763040]
22: /usr/lib/libglib-2.0.so.0(g_main_loop_run+0xc2) [0x7f037d763362]
23: /usr/lib/libgtk-x11-2.0.so.0(gtk_main+0xb7) [0x7f03848af4e7]
24: /usr/lib/libgtkmm-2.4.so.1(_ZN3Gtk4Main3runERNS_6WindowE+0x11f) [0x7f03852a616f]
25: ./src/solvespace(main+0x416) [0x55f23a]
26: /usr/lib/libc.so.6(__libc_start_main+0xf1) [0x7f0381f16741]
27: ./src/solvespace(_start+0x29) [0x5420e9]
[1]    14959 abort (core dumped)  ./src/solvespace

```

whitequark, 2016-06-23:
Fixed in master.

closed 2016-06-23 by whitequark

kamnxt, 2016-06-23:
This was the fastest issue fix I've ever seen on Github. Awesome.