solvespace/solvespace #15: Solvespace 2.1 is slow compard to 2.0

ruevs, 2016-07-05:
First I realize this is a fairly "stupid" bug report but I think it is important.

The 2.1 release of SolveSpace is much slower than 2.0.

Open "jansen-asm.slvs" from here http://solvespace.com/dl/mechanisms.zip (http://solvespace.com/linkage.pl)

Drag the "handle" to make the mechanism "walk" in 2.0 (chord tolerance 0.5; now 2184 triangles) then do the same in 2.1 (chord tolerance 0.01%; 0.03mm; 2184 triangles) - it is much, MUCH slower. 

So slow in fact that it is trivial to cause "SOLVE FAILED! in less than 45 degrees rotation of the handle! See the attached screen shot.
[image]

This is quite unacceptable. Sovespace is/was one of the few fast and lean programs in the 3D world.

Software: Windows XP 32

Hardware: 
CPU: Core 2 Duo E8500 3.2GHz
Graphics: Intel Q45/Q43 Express Chipset (driver v. 6.14.10.5420)
RAM: 4GB

ruevs, 2016-07-05:
I used the official release:
https://github.com/solvespace/solvespace/releases/download/v2.1/solvespace.exe

whitequark, 2016-07-05:
@ruevs Try chord tolerance 0.1%. I think (but haven't confirmed for this case specifically) that this is caused by the poor handling of assembly meshes.

ruevs, 2016-07-06:
Changing the chord tolerance makes no difference (even 5%).

whitequark, 2016-07-06:
Ok, that's a different bug then. Will investigate.

Evil-Spirit, 2016-07-06:
@ruevs 
> The 2.1 release of SolveSpace is much slower than 2.0.

Why you talking about performance? I've clicked "allow redundant" and all things in 2.1 becomes the same as 2.0. I don't think it is problem of performance (for me it's the same), but this problem is in changes performed in solver. Can you check again with "allow redundant"?

ruevs, 2016-07-06:
"Allow redundant" removes the solve failed. However the "FRAME RATE" in 2.1 is still way slower than in 2.0 and this is my complaint. The "solve failed" is just an artifact of easily being able to change a dimension in a big step because of the slow "frame rate".

Evil-Spirit, 2016-07-06:
@ruevs, ok, I have figured out why. We have double-generation for edges. If you want to have better performance, you have to switch off edges. I think it's already fixed in some branches (https://github.com/Evil-Spirit/solvespace/commit/ba84cbcb83163d317ccc81e427c752b76cf0838e#diff-9a5625e73a45d7ece1335647e0fa27d6L96).

> Perhaps worth pointing out again - this is an Intel video card

video card can't be bottleneck for such simple files.

ruevs, 2016-07-07:
You are absolutely right! Turning off the edges makes 2.1 indistinguishable from 2.0 by eye.
When I have a development system I will pull your branch and try it.

whitequark, 2016-08-07:
@Evil-Spirit This also needs to go to 2.2.

Evil-Spirit, 2016-08-09:
ready in https://github.com/Evil-Spirit/solvespace/commit/b6fbc6176cc508fc61465ad1a43a999d9481f570

whitequark, 2016-08-13:
@Evil-Spirit No, this doesn't help. Even with your patch it is very easy to break the assembly, whereas when using the 2.0 that never happens.

whitequark, 2016-08-13:
@Evil-Spirit I figured out what happens. There are two independent problems. The first one is that calculating outlines is slow (which you have fixed, and it works well). The second is that our changed rank test errors out on this assembly, whereas previously that did not happen for some reason. Can you please take a look and figure out if this is a regression in the solver, or a genuine issue in the sketch? If latter, why does the rank test not error out immediately?

Evil-Spirit, 2016-08-13:
@whitequark , ok, i'll take a look into this.

Evil-Spirit, 2016-08-14:
@whitequark , I've tried to understand what's going wrong with this file.
You can try this branch with some fixes (some of them can be merged)
https://github.com/Evil-Spirit/solvespace/tree/redundant-drag
I think the problem is in constraint aliasing conflict which appears sometime. The problem is in g008-link-left, where we have 6dof - 2dof * 2(pt-on-line)  - 1dof * 2(pt-on-face) = 0. 
Without advanced solver debug output I can not say more about this issue. 
May be @jwesthues can say more about it?

whitequark, 2016-08-15:
So why exactly was this not a problem in 2.0?

Evil-Spirit, 2016-08-15:
@whitequark, because checking was before solving. The solve process is run for the skeleton sketch before and causes solver start from not converged state for problematic group. In such conditions we never get problems because we start our analysis from random hand-specified parameters, so chance to fail test is small. When we precisely converge, we get problem rather frequently, but not stable repeatable as you can see with solver fixes while the dragging.

whitequark, 2016-08-15:
Ok, I see.

Evil-Spirit, 2016-08-15:
I think when we have conveged system for g008, the two pt-on-plane equations starts to represent the same DOF, may be because equations is written for the same plane. Even more magic: just click some group checkbox when you have redundant problem - and it will disappear. This is because we start one more Newton-solver step over the converged parameters and get different result.

Evil-Spirit, 2016-10-08:
So, what we have to do?
* Merge outlines from canvas branch (only function with generation of edges and outlines inside only one pass)
* Switch off only outlines while dragging instead of all the edges

whitequark, 2016-10-09:
@Evil-Spirit I think your fix works well, doesn't it? There is only the rank test problem, and there, it seems, is nothing we can easily do with that. So merging `isDegraded` to 2.2 is the way to go, I think.

Evil-Spirit, 2016-10-09:
@whitequark , ok.

Evil-Spirit, 2016-10-09:
@whitequark, Problem with rank test may be in equations or the form we written it for this case. Or may be just not properly adjusted epsilons for rank test...

whitequark, 2016-10-09:
If you have any idea what's wrong with the rank test, then please look into it.

jwesthues, 2016-10-09:
@Evil-Spirit Why do you think the two pt-on-plane constraints become the same DOF? The link has a rotational degree of freedom without them, as you can see if you delete one. The rank test seems just wrong here.

whitequark, 2016-10-10:
Degraded mode merged in 2.x (not master) but I would still like to understand why the rank test fails, so this is open for now. The rank test issue should be the last in 2.x series, yay!

Evil-Spirit, 2016-10-12:
@whitequark,
I can't figure out what is wrong with it. May be there is problems with quaternion representation. May be 
some two dofs began to express the same dof in converged state (as like case of triangle with a+b=c).
@jwesthues, what do you think about it?

jwesthues, 2016-10-12:
@Evil-Spirit What's the simplest change that takes this from working to broken? And beyond the fact that it fails with your changes, why do you think there's a fundamental problem?

whitequark, 2016-10-12:
@jwesthues `git bisect` points me to c011444045.

Evil-Spirit, 2016-10-12:
@jwesthues, because now rank test performed twice: before and after solving and this change causes the rank test fail. Before this change, there was cases when rank test can pass on redundant constraints, and stay pass always if we drag the sketch. Only when we cause Solve() to start without sketch modification, rank test starts to fail. This problem was because we performed rank test over not converged(damaged by dragging) parameters, so it passes. When sketch becomes converged and the dof degeneration appears, solver just reports "OKAY", but actually this state is redundant. The next drag cause parameters changing and rank test again passed. But if we don't drag sketch but cause Solve() to run, the rank test will fail.

Evil-Spirit, 2016-10-12:
@jwesthues 
To demonstrate this, you can try 
'''
To reproduce the bug, do this:
  * Draw a triangle.
  * Create a length constraint for all sides.
  * Set side lengths to a, b, and c such that a + b = c.
  * Add a line segment.
'''

Evil-Spirit, 2016-10-12:
@jwesthues, I guess the case with jasen-asm can has the same problem but this was hidden by problem with early rank test.

Evil-Spirit, 2016-10-12:
@whitequark 
@jwesthues 
try with this commints:
https://github.com/Evil-Spirit/solvespace/commit/a24d869dc47637f1c71ebe975c833f4fef2e544b
https://github.com/Evil-Spirit/solvespace/commit/8700aa1a56439153443ea1e8482dfada64d3b4ea
This makes message REDUNDAND_OKAY just error message without breaking the dragging abilities.
Also, for DIDNT_CONVERGE this makes ability to drag smooth and catch state where sketch becomes solveable again (this is  useable for 3d assemblies).
Here you can see what jasen-asm don't have constant redundant error, but this appears sometimes. And the next solve() (it can be performed by pressing "space") fixes this problem. 
May be for this sketch dofs is near to degrade but accidently fall into it because of numeric error noise. Anyway, this needs to be researched in details with more verbose debugging output.

jwesthues, 2016-10-12:
@Evil-Spirit I'm thinking it's something numerical, as you say, since (a) I can't see anything wrong with those constraints geometrically, (b) if the problem is with the final solution, then I don't understand why the solver works for small changes but fails for big ones, and (c) re-solving with space, which should if anything get you closer to the final solution, _fixes_ the problem (and, before c011444, at least didn't break).

So between the failing rank test at the end of solve k, and the passing rank test at the beginning of solve k+1, what changes?

Evil-Spirit, 2016-10-12:
The changes between k and k+1 is likely dragging (not so much cases with ui when we call solve() like by pressing space). This dragging causes regenerating sketch. During this process, the master-2d-sketch (g004-sketch-in-plane) solved in new state and causes big disconvergence for all other groups with 3d details, so this can't fail because of numerical noise.

Evil-Spirit, 2016-10-12:
Before https://github.com/solvespace/solvespace/commit/c0114440450a30f75e7f08425af3f33173acce04 there was no any chance to fail: space or other ui action without dragging re-solves sketch with more precision and fix the problem with rank-test fail. Any dragging causes big disconvergence an can't cause this problems...

jwesthues, 2016-10-12:
I mean re-solving with space, without moving anything. If the problem is that we really should lose rank when we're converged, then why does converging more _fix_ the problem?

jwesthues, 2016-10-12:
Random thought: Is `VectorsParallel()` pivoting wrong due to the big move, converging anyways but ending up singular, and then re-pivoting correctly on the new solution when you re-solve?

Evil-Spirit, 2016-10-13:
@whitequark,
1) https://github.com/Evil-Spirit/solvespace/commit/a24d869dc47637f1c71ebe975c833f4fef2e544b
2) https://github.com/Evil-Spirit/solvespace/commit/8700aa1a56439153443ea1e8482dfada64d3b4ea
Can you think about merging this commits?

1) The reason of this is fact, what REDUNDANT_OKAY prouduces the right sketch state, but when it happens, sketch becomes broken. Isn't it better to set it to solved parameters instead of broken?

2) In this case, while we dragging we interrupt moving. This does not prevents further dragging (you sill can perform one more step on the dragging). This is so annoying. In 3d assemblies we just can't move anything to get solution, and have to press "relax" button, which is annoying too.

jwesthues, 2016-10-13:
>Random thought: Is `VectorsParallel()` pivoting wrong due to the big move, converging anyways but ending up singular, and then re-pivoting correctly on the new solution when you re-solve?

Yes, and that's the problem here. The constraints are fine geometrically, and the linearization is fine. The problem is that `VectorsParallel()` wrote constraint equations with the point-on-line constraint projected into the wrong two planes. Options to fix are:

1. Replace that ugly pivoting with something that effectively re-chooses those two planes with each Newton iteration.
    - But that's ugly too, since you can't do that with a continuous function. It's easy to imagine weird convergence due to that.
2. If the rank test fails, then regenerate the equations, rewrite the Jacobian, and retry the rank test.

The second option is hacky, but easier, and I don't see any obvious harm it could cause.

Evil-Spirit, 2016-10-13:
@jwesthues,
Can you explain again, why we can't use some other equation for vectors parallel in 3d?
```atan(dot(a, b), length(cross(a, b))) == (0 || 180)```
```normalized(a) == normalized(b)```
may the something else...

jwesthues, 2016-10-13:
https://en.wikipedia.org/wiki/Hairy_ball_theorem

For the Newton's method, we'd like two continuous functions such that by writing the linearized equations for f(u, v) = 0 and g(u, v) = 0, we constrain the 3-vectors u and v to be parallel. If we had those, then we could treat u as the constant input, and v as the output. So the Jacobian is the partial derivatives of each function with respect to each component of v.

If we did that, then for any input u, the Jacobian defines two planes intersecting in a line parallel to u. Either plane's normal is always perpendicular to the input u. So if we let u be the vector from the center of a sphere to its surface, then each row of the Jacobian always gives us a nonzero vector tangent to the sphere. The hairy ball theorem says you can't do that, so the functions that we want don't exist.

A discontinuous function can exist; that's what we have now. I'm not entirely sure what happens when you cross the discontinuity inside the Newton's method, and either a convincing theoretical explanation or considerable experiment would be required to be confident that's not bad.

whitequark, 2016-10-13:
@Evil-Spirit So,  Evil-Spirit/solvespace@a24d869 looks fine to me, and it's actually a better solution than 3ff236c4941c8d7f501dd1d4d6ae81870ab337ca currently used to fix that problem, so we should replace that with it.  On the other hand Evil-Spirit/solvespace@8700aa1 doesn't look fine at all; we should fix errors in the solver and not paper over them.

I think @jwesthues's option number 2 is better here, and it is also safe to go into 2.x branch. On the other hand a change of how vector parallel constraint works isn't so safe for 2.x, I think.

Evil-Spirit, 2016-10-13:
@whitequark,
Yes, the option 2. is the way to resolve it temporary. For equations changing we have to do a lot of tests. And before it we have to figure out which equation is better. 
@jwesthues,
So, how I understood, the current two-cased implementation of the parallel equation simply has no any discontiunity and any way jump over?

jwesthues, 2016-10-13:
Three cases, actually, each of which fails at a different coordinate axis. We take whichever puts the failure farthest from the initial conditions. For an extra-hacky "fix", we could make the failing positions some arbitrary weird vectors, much less likely to hit by accident...

I suspect that it would be possible to readjust at each Newton step in some useful way, avoiding this problem. There's just a lot of potential for subtle misbehavior.

Evil-Spirit, 2016-10-13:
I've pushed workaroud for option 2.
https://github.com/Evil-Spirit/solvespace/commit/3bf2fe42c5912d664806498eebb731c023eccfbd

whitequark, 2016-10-14:
Merged in master and 2.x, thanks everyone!

@ruevs SolveSpace 2.2 will be as snappy as 2.0 (although at a cost of small visual degradation, which will be fixed in 3.0).

closed 2016-10-14 by whitequark

ruevs, 2016-10-16:
Nice job guys! I certainly could not have figured out the second one.