solvespace/solvespace #17: Use AppImage packages for Linux

Symbian9, 2016-07-17:
While DEB packages is not universal and not portable for many Linux distributives, propose replace them with AppImages for both 32bit and 64bit
* https://github.com/probonopd/AppImages/issues/75

References:
* http://solvespace.com/forum?action=viewthread&parent=1369
* http://appimage.org
* http://askubuntu.com/questions/774490/what-is-an-appimage-how-do-i-install-it

probonopd, 2016-07-17:
Is there interest from the solvespace team in an AppImage?

Symbian9, 2016-07-26:
@probonopd, we all interested ;-)

Can you make at least one AppImage of SolveSpace v.2.1 for 32bit for test it?

probonopd, 2016-07-26:
Can you provide either debs compiled on debian oldstable, Ubuntu 12.04 or older; or rpms compiled on CentOS 6? Then I could easily take it from there.

whitequark, 2016-07-26:
@probonopd  Not really. The current automated build infrastructure we use (Travis for *nix) only has 64-bit Ubuntu trusty, which is one of the reasons we have the problem.

probonopd, 2016-07-26:
Here is an AppImage I generated from your provided Ubuntu trusty deb files:
https://bintray.com/probono/AppImages/SolveSpace#files

And here is the recipe:
https://github.com/probonopd/AppImages/blob/master/recipes/solvespace/Recipe
Note that this merely repackages the existing binaries, so that they not only run on Ubuntu but also other distributions. But this also means that it will run on systems that are no older than Ubuntu 14.04, so most distributions from 2015 and later should be able to run it.

To make it run on older distributions, we would need to compile on an older build system; and to produce 32-bit AppImages, we would need to compile for 32-bit. Both is significantly more work.

whitequark, 2016-07-26:
> To make it run on older distributions, we would need to compile on an older build system; and to produce 32-bit AppImages, we would need to compile for 32-bit. Both is significantly more work.

Yeah. Which is exactly why it's still not done... It is actually not enough to build on an *older* system while distributing .debs, the debs I currently build don't install on Xenial.

> You could also incorporate the AppImage generation into your travis builds.

Possible, but probably doesn't make a lot of sense until I can generate 32-bit debs, which is as hard as any other part of this list, really.

probonopd, 2016-07-26:
Actually the latter is not so hard, as shown here:
https://github.com/travis-ci/travis-ci/issues/5770

Symbian9, 2016-07-26:
@probonopd, here is SolveSpace 2.1 for 32bit (i386)
* deb installer - https://www.cnc-club.ru/forum/download/file.php?id=82708
* binary - https://www.cnc-club.ru/forum/download/file.php?id=82688

So, try make AppImage for 32 using them (for testing)

Symbian9, 2016-08-21:
https://bintray.com/package/files/probono/AppImages/SolveSpace

whitequark, 2016-10-10:
So, I have looked at this again. Here are my thoughts:
  * AppImage is in essence "just another distribution" we have to build for. It has less ABI problems due to the way it packages libraries, but it still has ABI problems, e.g. in glibc, in GTK plugin loading, maybe somewhere else.
  * We do not have the resources to set up a Linux build farm. There are way too many combinations of (distro × version × architecture) in use. It is clearly not enough to provide just one or two of them, since we did that and people complained. With the proliferation of RPis and so on people will also complain there's no ARM version... etc.
  * Building on Travis is not enough. Travis only has 14.04 which according to @probonopd is too new (!). I am definitely not bothering with oldstable, or really anything without GCC 5+, since we don't support that. I also dislike the idea of using Docker, since our Travis is already plenty fragile, and so I would like to not add another component to slow it down and create breakage.
  * The .debs we used to build were useless if you didn't run 14.04 exactly. Both older and newer distributions couldn't use them well.
  * The distributions are picking up. Debian Stretch and Ubuntu Yakkety will have version 2.2; I think it's also at least in AUR, AltLinux.

And here's the conclusion: the SolveSpace project should not provide any Linux packages at all, AppImage or not. It is the job of distributions to properly package software for all releases and architectures they have, and I will leave it to them.

If you want the freshest code, it is also not at all hard to build from source. You could even just run `./.travis/install-debian.sh && ./.travis/build-debian.sh` on any Debian system, and that will make a working build for you. It has to, because our CI system depends on it.

closed 2016-10-10 by whitequark

probonopd, 2016-10-10:
@whitequark I respect your decision, but of course you can build AppImages on Travis CI with 14.04 - as long as you don't expect them to run on target systems older than 2014ish. I have good experiences with using 14.04-built binaries on newer systems.

whitequark, 2016-10-10:
That unfortunately still leaves out 32-bit binaries. I feel like a person external to SolveSpace but interested and familiar with packaging could produce much higher quality packages, such as AppImages, at much lower cost (since that wouldn't have to happen for every build, but only for releases, presumably, and wouldn't have to be maintained as much too).

whitequark, 2016-10-10:
Overall, probably the single most important part is that I *already* spend easily half of my time working on platform support. Adding Linux packaging there would just be too much.

probonopd, 2016-10-10:
So Linux stays a 2nd-class citizen compared to Windows and macOS, dependent on 3rd parties who may or may not fully understand your software :-/

whitequark, 2016-10-10:
Actually, so far the reverse has happened. The Debian Science packaging team made such a better job than me at it, that I have removed the in-tree `debian/` stuff entirely in favor of using their repository.

Symbian9, 2016-10-11:
> So Linux stays a 2nd-class citizen compared to Windows and macOS, dependent on 3rd parties who may or may not fully understand your software :-/

There is no more to say...

Symbian9, 2016-10-11:
@probonopd, think, if SolveSpace will ignore Linux we came to time, when it would be forked in independent project (like LibreCAD was forked from QCAD CE, when original author of QCAD try ignore Linux users).

And I won't this future for SolveSpace.

@whitequark, @jwesthues, @Evil-Spirit, <b>PLEASE, think twice before drop Linux support!</b>

whitequark, 2016-10-11:
@Symbian9 What? I didn't drop Linux support. I simply chose the same distribution model as *the majority* of software packages ever written for Linux or Unix.

Symbian9, 2016-10-11:
> I simply chose the same distribution model as the majority of software packages ever written for Linux or Unix

@whitequark, Ubuntu has standart way for distributing using Launchpad, but you already drop SolveSpace repo on Launchpad...
* https://launchpad.net/~whitequark/+archive/ubuntu/solvespace/+packages
* http://solvespace.com/forum.pl?action=viewthread&parent=574

Symbian9, 2016-10-15:
https://github.com/solvespace/solvespace/pull/3#issuecomment-184319506
> CI builds with gcc-4.6 succeed. What gcc version and what compilation problem do you have?

https://github.com/solvespace/solvespace/blob/master/CMakeLists.txt
> GCC 4.8/4.9 ship with broken but present <regex>. meh.

@whitequark, @Evil-Spirit, is it a way keep support for GCC 4.8.2 and Trusty (14.04)?

Or how run GCC 5.0 on Trusty if its not officially support?
* https://gist.github.com/beci/2a2091f282042ed20cda
* https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test/+packages

whitequark, 2016-10-15:
@Symbian9 GCC 5+ is a hard requriement. You can install gcc-5 from the ubuntu-toolchain-r repository that you have found, which is what I use for CI.

Symbian9, 2016-10-17:
@whitequark, where are Linux binary for SolveSpace 2.2? And where gone deb's from 2.1 release page?

whitequark, 2016-10-17:
@Symbian9 Provided by your distribution (the 2.2 binaries will be built for Debian and Ubuntu in a matter of days). You can also build them yourself.

I have removed the 2.1 debs from the download page because 2.1 is already included in Debian and Ubuntu, and the packages in these distributions are better.

Symbian9, 2016-10-17:
> I have removed the 2.1 debs from the download page because 2.1 is already included in Debian and Ubuntu, and the packages in these distributions are better.

But Ubuntu have not binary for 14.04 LTS and 16.04 LTS, that was presented on SolveSpace release page on Github.
* http://packages.ubuntu.com/search?keywords=solvespace

@whitequark, So, please, recover at least those 2.1 binary for 14.04.

Symbian9, 2016-10-25:
Dirty build for Trusty 14.04
https://launchpad.net/~alex-p/+archive/ubuntu/solvespace-trusty2

probonopd, 2016-10-26:
Thanks @Symbian9. Here is an AppImage I generated from your ppa:
https://bintray.com/probono/AppImages/SolveSpace#files

Code:
https://github.com/probonopd/AppImages/blob/master/recipes/meta/SolveSpace.yml

Symbian9, 2016-10-26:
THANK YOU, @probonopd !

probonopd, 2016-10-26:
@Symbian9 my packages are just examples; AppImage is designed for upstream packaging, i.e., AppImages made by the original software authors. It should be doable to compile the tools and run the recipe on a 32-bit system (even ARM/but unfortunately I don't have the time to do it.

Symbian9, 2016-10-26:
@probonopd, now I'm trying run your AppImage...
* https://dl.bintray.com/probono/AppImages/SolveSpace-2.2.0+git1000.glibc2.14-x86_64.AppImage

... under LinuxMint 17.3 Xfce (64bit).
* https://www.linuxmint.com/edition.php?id=213

It ask me to install it, but then give error

```
/home/mint/.cache/thumbnails/normal/cadf6d654aa10b104fd5856b2a809309.png is missing. Probably not running ./bin//solvespace.wrapper from within an AppImage.
Hence falling back to using .DirIcon
tmp/.mount_o2E6MD/usr/bin/solvespace: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /tmp/.mount/usr/bin/solvespace)
tmp/.mount_o2E6MD/usr/bin/solvespace: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /tmp/.mount/usr/bin/solvespace)
```

probonopd, 2016-10-26:
This means that the `libstdc++.so.6` is too old for this AppImage to run. To make it work on this system, we'd have to compile on an older system than the ppa above.

Symbian9, 2016-10-26:
@probonopd, Think, you must bundle in this AppImage also toolchain  dependency

> Dependencies: Toolchain test builds (included on 2016-10-16)
* https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test

probonopd, 2016-10-26:
@Symbian9 I don't understand that comment. AppImages are supposed not to duplicate core system libraries like glibc, libstdc++ etc. but use the ones from the OS.

Symbian9, 2016-10-26:
@probonopd, AppImage for SolveSpace 2.2 that you build, depend from Toolchain test builds (that not distributed in with Trusty 14.04LTS).

So, user cant detect is it installed or not before run this AppImage.

So, how user must understand how run it?

probonopd, 2016-10-27:
The toolchain is only needed at _build_ time and not at _runtime_, correct?

Symbian9, 2016-10-27:
@probonopd, did you try your binary SolveSpace 2.2 AppImage under Trusty 14.04 or LinuxMint 17.3?

Read this again!
* https://github.com/solvespace/solvespace/issues/17#issuecomment-256655277#issuecomment-256453946

Symbian9, 2017-07-25:
@probonopd, use OpenGL 1.x flag for build AppImage, please (because OpenGL|ES 3.x supported not on all PC, that in same time has good support for OpenGL 1.x)
* https://github.com/solvespace/solvespace/blob/master/CMakeLists.txt#L46

probonopd, 2017-07-25:
How do I do that?

Symbian9, 2017-07-25:
@probonopd, setup your recepie

* https://github.com/probonopd/AppImages/blob/master/recipes/meta/SolveSpace.yml

Or who is creator of those AppImages?

* https://bintray.com/package/files/probono/AppImages/SolveSpace

probonopd, 2017-07-25:
I mean, what do I have to change exactly to set the "OpenGL 1.x flag"?

whitequark, 2017-07-26:
I don't think this actually works with GTK 3 (SolveSpace master).

Symbian9, 2017-08-01:
> I mean, what do I have to change exactly to set the "OpenGL 1.x flag"?

@Evil-Spirit, need help with flags for OpenGL 1.x!

@probonopd, please, try make AppImage according releases version: 2.1, 2.2, **2.3**. Because currently your builds mostly look like 'nightly' builds for git1000 and git1078, that is not mean 'release'.

probonopd, 2017-08-01:
Hi @Symbian9 is this project interested in generating official AppImages? Mine are just examples.

Symbian9, 2017-08-01:
> is this project interested in generating official AppImages?

@probonopd, main developers mostly don't respond to Linux users

* https://github.com/solvespace/solvespace/issues/17#issuecomment-252766048
  > And here's the conclusion: the SolveSpace project should not provide any Linux packages at all, AppImage or not. It is the job of distributions to properly package software for all releases and architectures they have, and I will leave it to them.

Thanks to You there is most up to date builds, that work on xUbuntu 16.04 LTS.

> Mine are just examples

@probonopd, Please, do not drop Linux users of SolvSpace and give us more  up to date builds as AppImages!

jwesthues, 2017-08-01:
> @probonopd, main developers mostly don't respond to Linux users

@whitequark develops under Linux. He provides detailed build instructions under Linux. Most Linux users can type `apt-get install solvespace` or equivalent, and get the latest stable. If other people want to use his work to build packages in other forms, then we're in favor; but his own time is finite, and he decides how to spend it.

Only in free software have I seen people show such extraordinary disrespect and entitlement to the developers working for their (and everyone's) benefit.

probonopd, 2017-08-01:
@jwesthues it's not disrespect. I'm happy to help @whitequark if he is interested in providing AppImages to reach Linux users despite whatever distribution they might be running.

jwesthues, 2017-08-01:
@probonopd To be clear, I'm referring only to @Symbian9's "mostly don't respond to Linux users".

The AppImages are cool, and if you want to continue then we'll advise people to use them, just like we tell them to `apt-get install solvespace`. It seems natural enough to me that @whitequark prefers to focus his own time on developing, and leave packaging to the expert packagers.

Symbian9, 2017-08-01:
@jwesthues, to be clear, I'm not say desrispect, but as in fact that you build binary for Windows & macOS. Why users of Windows & macOS not build SolveSpace from sources? In this situation its look like descrimination for Linux users.

> Most Linux users can type
`apt-get install solvespace` or equivalent, and get the latest stable.

I use one of the most popular Linux distributive - xUbuntu 16.04 LTS. And there NO builds with latest SolveSpace version for it

* https://packages.ubuntu.com/search?keywords=solvespace

SolveSpace team made great work on improving since 2.0, but we, users, need binary. We make testing and feedback, that is also big pie of development process. I discovered many issues, give many ideas, that now implemented in SolveSpace 2.2-3.0. I, as simple user that help test and make you program better, not ask too much - only binary builds, that can test.

Please, understand - I want only help make it better l. For all us.

jwesthues, 2017-08-02:
2.3 is in zesty; so just a matter of time, or you could upgrade, or cherry-pick.

I am certain that in the time you've spent on this thread you could have figured out how to build from source. I think most users who should be running from development branches know how to do that, and it's a generally useful skill.

In any case, your request is very clear. It's not useful to repeat it, or to say that the present maintainer (who's responsible for the Linux port in the first place!) doesn't respond to Linux users.

Evil-Spirit, 2017-08-02:
@Symbian9, I am using Windows, sorry, bro. The flags is exist, but @whitequark says what this is not compatible with the new versions of GTK we are using now. I don't know what does it means, and can't help to solve this.

Evil-Spirit, 2017-08-02:
btw windows uses ANGLE and this is dead-slow, probably we should provide true-opengl build too.

whitequark, 2017-08-02:
> But if not even the original author is interested in providing and supporting an official author-provided, downloadable and supported version for Linux, then I am not, either. ;-)

To clarify: I think AppImages are certainly a solution to some problems, but they do not have the same degree of portability as our macOS or Windows applications, because they still have versioning issues with system ABIs such as glibc or GTK. This means that an AppImage is *not* a "build once, run anywhere on the same architecture at any point in the future" deal like our Windows and macOS binaries. This is the specific reason I opt not to provide them; I do not have the time to figure out the issues AppImages create.

In other words, I see AppImages as not better or worse than any other Linux distribution mechanism. Launchpad PPAs are fine, AppImages are fine, Flatpak is fine, Zero Install is fine, .rpm's for Red Hat are fine, but none of these things is *obviously superior to all others* or *supersedes them*. And of course even if I start providing AppImages, people will still want [insert their preferred Linux redistribution format here] regardless of that fact, and I cannot tell them "see, AppImages completely supersede your use case" because they don't!

I am 100% fine with @probonopd building and distributing 3rd party AppImages, just like I'm fine with the Debian Science packaging team, or whoever maintains any other packages. If the packages appear competently done to me, I will link to them from the website; they will be clearly marked as 3rd party (because I cannot control what people put there), and anyone not fine with that is, of course, free to build from source, which I specifically made as easy as possible and continue to improve when people report troubles. I will coordinate with @probonopd, or any other packager, to best serve the needs of our users.

What I cannot do, however, is to take onto myself the Sisyphean task of maintaining a dozen disparate Linux packages, none of which would be built quite right because I am not--cannot possibly--know every quirk of every distribution. Maintaining just *one*, Ubuntu 14.04 debs, has been a major sink of my time, which would have been much better spent doing QA of the 2.x release series.

Finally. @Symbian9, I've grown rather weary of your repeated demands for features that are out of scope, merely a matter of taste, do not agree with my roadmap for the project, or mathematically impossible. Everyone else over here, even people I had major ideological disagreements with, have been able to respect the simple facts that a) my time is finite and scarce and b) I do not owe it to anyone to spend it on whatever they think is most important. I have dropped the Ubuntu .debs *specifically* because of your behavior, and I would not have blamed a person less patient for dropping the Linux support altogether. I am thankful for your positive contributions, such as bug reports or translations, but that does not mean I will tolerate this behavior forever. Please consider this your final warning.

whitequark, 2017-08-02:
> btw windows uses ANGLE and this is dead-slow, probably we should provide true-opengl build too.

@Evil-Spirit let's discuss it in #278.

whitequark, 2017-08-02:
> The flags is exist, but @whitequark says what this is not compatible with the new versions of GTK we are using now. I don't know what does it means, and can't help to solve this.

GTK 3 requires the use of OpenGL ES 2 / OpenGL 3. There's no point in building our code to use OpenGL 1 calls because if the hardware does not support OpenGL 3, GTK itself would still have to fall back to software rendering, slowing us down into oblivion. (There are other implementation issues, but this is the conceptual one.)

probonopd, 2017-08-02:
> To clarify: I think AppImages are certainly a solution to some problems, but they do not have the same degree of portability as our macOS or Windows applications, because they still have versioning [...]

Thanks @whitequark for your opinion, which I highly respect. In fact, you are descibing exactly what I am trying to provide for application developers, a way to use desktop Linux as a platform lke macOS or Windows. I have put a lot of thinking into this over the last decade. I am sorry to hear that you think we are not on the same level yet, and I would like to understand what exactly you mean by this, so that we can eventually get there. 

The way platforms typically work is that you develop for version n for a platform, and then the platform guarantees you that the code will continue to run on platform version n ... n+x. Even on macOS, x is not infinite, as Apple does deprecate APIs from time to time, but with a due warning over many years before the APIs are eventually removed. This usually is not a problem, because developers have plenty of time to migrate to the new APIs in the meantime.
Also, on most platforms if you develop against version n, then your binaries will not run on platform version n-x. This is not only the case for desktop Linux but also for Windows and macOS. You cannot develop on macOS 10.12 and expect your application to run on macOS 10.9 (at least not if you don't specifically compile against the 10.9 SDK).

I think desktop Linux can be used in the very same way. That is, develop against the oldest still-supported version of desktop Linux that you still would like your application to run on, and then expect from the desktop Linux platform that you code will still run on the next x versions without breakage (if breakage does occur, then I consider it a bug of the platform that needs to be fixed). The only difference is that no one has clearly defined so far what we can actually expect from the "desktop Linux platform", because every distribution puts in someting slightly different. Luckily, 99% of the basic desktop Linux stack is the same on most major desktop Linux distributions, and the remaining 1% should be worked out. For this we need people like you to point out the remaining issues.

Now, of course you could say "just ignore whatever the distribution happens to ship and use our own, private copy of the desktop stack as the platform". This is exactly what Snappy and Flatpak do. Some may consider this as bloated, and it introduces an additional dependency onto something else than the target system (distribution), namely "runtimes" or "base snaps", which complicates the system since the user can no longer just download an application and run it without having to set up someting else (namely, the runtime) first. If you really want to "just ignore whatever the distribution happens to ship and use our own, private copy of the desktop stack as the platform", you can do that with AppImage, too - just bundle _everything_ down to `glibc` inside the AppImage. It will be larger, but it will work. I'd say this is working around, rather than solving, the destop Linux platform challenge though.

Here is what I think needs to happen, and I'd highly appreciate your thoughts.
https://github.com/AppImage/AppImageKit/wiki/Desktop-Linux-Platform-Issues

> What I cannot do, however, is to take onto myself the Sisyphean task of maintaining a dozen disparate Linux packages, none of which would be built quite right because I am not--cannot possibly--know [...]

Fully agree. This is exactly why AppImage exists. Your choice of 14.04 is a good one since it is reasonably old. In fact, this enabled me to convert your existing binaries into an AppImage with just a few lines of code.

@whitequark please do __not__ see this as me requesting anything. In fact, I am not really using SolveSpace at all. I am just here to offer my help for a problem that I see needs to be solved to make desktop Linux more attractive for developers like you.

whitequark, 2017-08-02:
> Also, on most platforms if you develop against version n, then your binaries will not run on platform version n-x. This is not only the case for desktop Linux but also for Windows and macOS. You [...]

The comparison to desktop Linux is not really valid. On both Windows and macOS, if you have the appropriate SDK installed and specify the corresponding flag during the build, then the resulting binary will run on the OS the SDK is for. Right now we target Windows Vista and newer, and OS X 10.7 and newer, which are outstandingly old. (If I was willing to spend time on that, we could still run on Windows XP, but it doesn't seem like a worthwhile goal.) This does not require me to waste huge amounts of time staring at the output of `objdump`, it just works.

> If you really want to "just ignore whatever the distribution happens to ship and use our own, private copy of the desktop stack as the platform", you can do that with AppImage, too - just bundle [...]

This doesn't work because both glibc and GTK attempt to load plugins from the base filesystem. In case of glibc that's the NSS plugins (which you can work around with musl, assuming GTK compiles under musl, OpenGL libraries still load, and someone is going to waste their time on this), and in case of GTK it's primarily the theme engine, but I think there were others too.

This is not a theoretical concern. For example, for a good while, Skype was unusable for me because it could not load the theme engine I had set up in my system, which meant it could not properly render on my HiDPI display, and the fonts were simply too tiny to see.

> Your choice of 14.04 is a good one since it is reasonably old. In fact, this enabled me to convert your existing binaries into an AppImage with just a few lines of code.

Last time I looked, GTK 3 from trusty was not compatible with GTK 3 from xenial because of ABI breakage. Also, that doesn't solve the issue of providing 32-bit binaries, because last time I looked, multilib did not work well enough on trusty for SolveSpace to build with `-m32`.

> I am just here to offer my help for a problem that I see needs to be solved to make desktop Linux more attractive for developers like you.

Really the only way to make desktop Linux attractive to me is to let me not think about deployment for a longer time than it's necessary to specify a flag or two, i.e. the same thing as Windows and macOS. Individual distributions packaging SolveSpace do this for me. AppImages *built by someone else* do this for me. Having to ever think about whether the latest GTK 3 ABI policy will result in breakage or not, doesn't.

probonopd, 2017-08-02:
> On both Windows and macOS, if you have the appropriate SDK installed and specify the corresponding flag during the build, then the resulting binary will run on the OS the SDK is for.

The equivalent for this would be, say, a CentOS 6 chroot, correct?

> This does not require me to waste huge amounts of time staring at the output of objdump, it just works.

Luckily I've never had to stare at the output of objdump when making AppImages ;-)

>  Bboth glibc and GTK attempt to load plugins from the base filesystem. In case of glibc that's the NSS plugins

_If_ you choose to bundle your private glibc, then you need to bundle _all_ of the glibc package including the matching NSS plugins. But again, I think the better way is to compile against an "old enough" glibc and not bundle it privately.

> in case of GTK it's primarily the theme engine

> Last time I looked, GTK 3 from trusty was not compatible with GTK 3 from xenial because of ABI breakage.

Can you point me to details? I would like to learn more about this.

whitequark, 2017-08-02:
> The equivalent for this would be, say, a CentOS 6 chroot, correct?

There's no equivalent. GTK 3 does not have ABI stability guarantees; you cannot target "GTK 3". You can target "GTK 3.20", "GTK 3.22", etc. If I built SolveSpace against libgtk 3.2 then those binaries would have long stopped working on modern Debian.

Using gtkmm makes this an order of magnitude more complicated because libstdc++'s ABI depends on the set of flags used to build it.

> If you choose to bundle your private glibc, then you need to bundle all of the glibc package including the matching NSS plugins. But again, I think the better way is to compile against an "old enough" glibc and not bundle it privately.

That's not how it works. The point of having loadable NSS plugins is so that 3rd party software (e.g. samba) can provide their own NSS plugins, and so provide new resolvers, etc. The same way with theme engines.

traverseda, 2017-08-02:
Perhaps removing the dependency on native-widgets entirely would be a good thing? SDL is a pretty reasonable cross-platform target, and it certainly looks like that would simplify a lot of your build stuff. Especially as the project moves on to android and emscripten.

It seems like a pretty complicated artiface to support "text editors, menu bars, tooltips and file pickers".

[This](https://github.com/mlabbe/nativefiledialog.git) implements native file-dialogs with "No third party dependencies for building or linking" for use under SDL. It only support desktop-os's though.

whitequark, 2017-08-02:
For reasons outlined many times before, no. Also, that still depends on GTK 3 on Linux, so it doesn't even solve the problem.

probonopd, 2017-08-02:
> There's no equivalent. GTK 3 does not have ABI stability guarantees; you cannot target "GTK 3". You can target "GTK 3.20", "GTK 3.22", etc. 

That is because GTK+ 3 still in development, and as such, apparently still a "moving target". If you target GTK+ 2 which is stable by now then you should not have these hassles. (In my personal view, GTK+ 3 should have been named 2.99.x until the ABI was stable, which was apparently GTK+ 3.22, released 21 September 2016 https://www.reddit.com/r/linux/comments/6ad7v0/is_gtk3_stable_now/).
We should expect from a project like GTK+ to release stable versions at some point, with a major version number if they follow [semver](semver.org), and then keep that ABI working throughout the major version.

[22 more replies]