solvespace/solvespace #18: *** Error in `solvespace': double free or corruption (fasttop): 0x0000000002993380 *** opening DWG file

jasonblewis, 2016-07-18:
When I try and open a DWG file, I get the following error

```
*** Error in `solvespace': double free or corruption (fasttop): 0x0000000002993380 ***
```
SolveSpace version 2.1~dce90646 on debian jessie

jasonblewis, 2016-07-18:
DWG file can be downloaded from here: http://www.seanmichaelragan.com/html/[2008-04-18]_MC_Escher_lizard_vector_art.shtml

[MC_Escher_single_lizard_tile.dwg.zip](https://github.com/solvespace/solvespace/files/368330/MC_Escher_single_lizard_tile.dwg.zip)

attached for reference

whitequark, 2016-07-19:
@Evil-Spirit Please take a look. This is a double free in libdxfrw:
```
==4544== Invalid free() / delete / delete[] / realloc()
==4544==    at 0x4C2D2CB: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4544==    by 0x5AC806: DRW_Spline::~DRW_Spline() (drw_entities.h:909)
==4544==    by 0x6B1618: DRW_Spline::~DRW_Spline() (drw_entities.h:908)
==4544==    by 0x6B22B4: DRW_HatchLoop::~DRW_HatchLoop() (drw_entities.h:964)
==4544==    by 0x6B16D3: DRW_Hatch::~DRW_Hatch() (drw_entities.h:1000)
==4544==    by 0x733102: dwgReader::readDwgEntity(dwgBuffer*, objHandle&, DRW_Interface&) (dwgreader.cpp:1114)
==4544==    by 0x731069: dwgReader::readDwgEntities(DRW_Interface&, dwgBuffer*) (dwgreader.cpp:943)
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==  Address 0x101e9c90 is 0 bytes inside a block of size 24 free'd
==4544==    at 0x4C2D2CB: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4544==    by 0x5AC806: DRW_Spline::~DRW_Spline() (drw_entities.h:909)
==4544==    by 0x6B1618: DRW_Spline::~DRW_Spline() (drw_entities.h:908)
==4544==    by 0x6B22B4: DRW_HatchLoop::~DRW_HatchLoop() (drw_entities.h:964)
==4544==    by 0x6B16D3: DRW_Hatch::~DRW_Hatch() (drw_entities.h:1000)
==4544==    by 0x733102: dwgReader::readDwgEntity(dwgBuffer*, objHandle&, DRW_Interface&) (dwgreader.cpp:1114)
==4544==    by 0x731069: dwgReader::readDwgEntities(DRW_Interface&, dwgBuffer*) (dwgreader.cpp:943)
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==  Block was alloc'd at
==4544==    at 0x4C2C16F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4544==    by 0x69BCD9: DRW_Hatch::parseDwg(DRW::Version, dwgBuffer*, unsigned int) (drw_entities.cpp:1964)
==4544==    by 0x733060: dwgReader::readDwgEntity(dwgBuffer*, objHandle&, DRW_Interface&) (dwgreader.cpp:1112)
==4544==    by 0x731069: dwgReader::readDwgEntities(DRW_Interface&, dwgBuffer*) (dwgreader.cpp:943)
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==    by 0x55E815: SolveSpace::MainMenuItem<Gtk::MenuItem>::on_activate() (gtkmain.cpp:947)
==4544==    by 0x5CB086F: Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) (in /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1.1.0)
==4544==    by 0x79171D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1)
```

whitequark, 2016-07-19:
Oh and while you're at it, these also should be fixed:
```
==4544== Conditional jump or move depends on uninitialised value(s)
==4544==    at 0x7020A3: dwgBuffer::getBitDouble() (dwgbuffer.cpp:353)
==4544==    by 0x7021AD: dwgBuffer::get3BitDouble() (dwgbuffer.cpp:374)
==4544==    by 0x69BD03: DRW_Hatch::parseDwg(DRW::Version, dwgBuffer*, unsigned int) (drw_entities.cpp:1964)
==4544==    by 0x733060: dwgReader::readDwgEntity(dwgBuffer*, objHandle&, DRW_Interface&) (dwgreader.cpp:1112)
==4544==    by 0x731069: dwgReader::readDwgEntities(DRW_Interface&, dwgBuffer*) (dwgreader.cpp:943)
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==    by 0x55E815: SolveSpace::MainMenuItem<Gtk::MenuItem>::on_activate() (gtkmain.cpp:947)
==4544==    by 0x5CB086F: Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) (in /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1.1.0)
```
```
==4544== Conditional jump or move depends on uninitialised value(s)
==4544==    at 0x701F63: dwgBuffer::getBitLong() (dwgbuffer.cpp:331)
==4544==    by 0x69C26B: DRW_Hatch::parseDwg(DRW::Version, dwgBuffer*, unsigned int) (drw_entities.cpp:1999)
==4544==    by 0x733060: dwgReader::readDwgEntity(dwgBuffer*, objHandle&, DRW_Interface&) (dwgreader.cpp:1112)
==4544==    by 0x731069: dwgReader::readDwgEntities(DRW_Interface&, dwgBuffer*) (dwgreader.cpp:943)
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==    by 0x55E815: SolveSpace::MainMenuItem<Gtk::MenuItem>::on_activate() (gtkmain.cpp:947)
==4544==    by 0x5CB086F: Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) (in /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1.1.0)
==4544==    by 0x79171D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1)
```
```
==4544== Conditional jump or move depends on uninitialised value(s)
==4544==    at 0x701CB3: dwgBuffer::getBitShort() (dwgbuffer.cpp:305)
==4544==    by 0x69C375: DRW_Hatch::parseDwg(DRW::Version, dwgBuffer*, unsigned int) (drw_entities.cpp:2003)
==4544==    by 0x733060: dwgReader::readDwgEntity(dwgBuffer*, objHandle&, DRW_Interface&) (dwgreader.cpp:1112)
==4544==    by 0x731069: dwgReader::readDwgEntities(DRW_Interface&, dwgBuffer*) (dwgreader.cpp:943)
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==    by 0x55E815: SolveSpace::MainMenuItem<Gtk::MenuItem>::on_activate() (gtkmain.cpp:947)
==4544==    by 0x5CB086F: Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) (in /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1.1.0)
==4544==    by 0x79171D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1)
```
```
==4544== Conditional jump or move depends on uninitialised value(s)
==4544==    at 0x69CF9A: DRW_Hatch::parseDwg(DRW::Version, dwgBuffer*, unsigned int) (drw_entities.cpp:2036)
==4544==    by 0x733060: dwgReader::readDwgEntity(dwgBuffer*, objHandle&, DRW_Interface&) (dwgreader.cpp:1112)
==4544==    by 0x731069: dwgReader::readDwgEntities(DRW_Interface&, dwgBuffer*) (dwgreader.cpp:943)
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==    by 0x55E815: SolveSpace::MainMenuItem<Gtk::MenuItem>::on_activate() (gtkmain.cpp:947)
==4544==    by 0x5CB086F: Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) (in /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1.1.0)
==4544==    by 0x79171D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1)
==4544==    by 0x79319A5: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1)
```
```
==4544== Conditional jump or move depends on uninitialised value(s)
==4544==    at 0x7027DB: dwgBuffer::getHandle() (dwgbuffer.cpp:539)
==4544==    by 0x702886: dwgBuffer::getOffsetHandle(unsigned int) (dwgbuffer.cpp:546)
==4544==    by 0x68B634: DRW_Entity::parseDwgEntHandle(DRW::Version, dwgBuffer*) (drw_entities.cpp:432)
[... cut ...]
==4544==    by 0x709E2D: dwgReader15::readDwgEntities(DRW_Interface&) (dwgreader15.h:35)
==4544==    by 0x6F69E7: dwgR::processDwg() (libdwgr.cpp:301)
==4544==    by 0x6F61FB: dwgR::read(DRW_Interface*, bool) (libdwgr.cpp:158)
==4544==    by 0x5DA463: SolveSpace::ImportDwg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (importdxf.cpp:926)
==4544==    by 0x649A3C: SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) (solvespace.cpp:550)
==4544==    by 0x55E815: SolveSpace::MainMenuItem<Gtk::MenuItem>::on_activate() (gtkmain.cpp:947)
```

whitequark, 2016-08-07:
@Evil-Spirit This also needs to go to 2.2.

Evil-Spirit, 2016-08-10:
fixed https://github.com/Evil-Spirit/libdxfrw/commit/3eb29b92b208b83f48e514961dc1aae2fa1f4fe7

whitequark, 2016-08-10:
@Evil-Spirit But you didn't fix the uses of uninitialized values.

Evil-Spirit, 2016-08-11:
@whitequark, I have no idea why it happens. I need all the valgrind log, especially all about `duint8 dwgBuffer::get2Bits()`. PVS Studio reports nothing about it.

whitequark, 2016-08-11:
PVS Studio is a static analyzer, it has no insight into this kind of problem... I'll get the complete log, ok.

whitequark, 2016-08-13:
@Evil-Spirit Your fix doesn't work. Have you tried importing the file? It fails.

whitequark, 2016-08-13:
@Evil-Spirit Nevermind, that's a problem with libdxfrw. Your fix is fine.

whitequark, 2016-08-13:
Crash fixed in master and 2.x.

@jasonblewis Please note that your DWG file will still not fully import. This is a deficiency in the DWG parsing library we use. I have no current plans to address this deficiency.

closed 2016-08-13 by whitequark

Evil-Spirit, 2016-08-13:
@whitequark, with this fix file is loaded for me. but there is some problems with coordinates and hatches.

whitequark, 2016-08-13:
@Evil-Spirit It errors out with a message for me, which is not "loading fine".

Evil-Spirit, 2016-08-13:
Oh, yes, as for me.