solvespace/solvespace #21: bug: parallel constraint on linked solid

wolfv6, 2016-08-01:
The attached example has one constraint, and it fails:

    SOLVE FAILED! redundant constraints

IMHO the constraint can not be redundant because there is only one constraint:

    constraints in group (0 DOF)
       c00b-parallel

Removing the c00b-parallel constraint gains 6 DOF:

    constraints in group (6 DOF)
     (none)

I am running SolveSpace version 2.1~dce90646 on Fedora 24.
[demo_parallel_constraint.zip](https://github.com/solvespace/solvespace/files/394894/demo_parallel_constraint.zip)

whitequark, 2016-08-02:
@Evil-Spirit Why does this happen?

Evil-Spirit, 2016-08-02:
@whitequark, Constraint is created in workplane, somehow. I've tried to create POINTS_COINCIDENT and get strange behaviour.

whitequark, 2016-08-02:
Ahhh! We need a better error message for this. Why does it say redundant and not unsolvable?

As to why it's created in workplane, this is because a workplane is active.

whitequark, 2016-10-09:
@Evil-Spirit Can you explain why the rank test fails in this case, and maybe how we can present a better message?

Evil-Spirit, 2016-11-07:
@whitequark, I think we can't distinguish here without additional checking. This case is the same as for two parallel segments and pt-pt-coincident of points of the same segment. In this case we have the same derivatives for different points (since this is the same point for solver after subsitution). This results in linear dependency inside Jacobian matrix. As solution, we just can perform some tests before solving and report more informative messages. This case already tested for 2d ("zero-length-edge"),

whitequark, 2016-11-18:
@Evil-Spirit please implement this check