solvespace/solvespace #23: Pattern fill (in line style)

Symbian9, 2016-08-07:
Please, add option to select what type of pattern would be showed in closed sketch
!(image)[http://community.coreldraw.com/resized-image/__size/680x1000/__key/CommunityServer.Discussions.Components.Files/323/7532.DESX5_5F00_hatch_2D00_presets.png]

whitequark, 2016-10-10:
So how is this actually useful for filled paths? It would definitely be useful if specified in a difference group, to produce technical drawings with the traditional section views and such, but I don't see any use in 2d.

Evil-Spirit, 2016-10-10:
@whitequark, it is useful for 2d only drawings, probably.

Symbian9, 2017-01-13:
> it is useful for 2d only drawings, probably.

@Evil-Spirit, not only for 2D drawings!
* http://help.solidworks.com/2010/English/SolidWorks/Sldworks/LegacyHelp/Sldworks/Detailing/Area_Hatch.htm
[image]
[image]

* AutoCAD 3D Hatch Tutorial
  * Part 1 - https://m.youtube.com/watch?v=ApGbQtMzDEE
  * Part 2 - https://www.youtube.com/watch?v=Lnuat6TxKRM

It could be used in "Export 2D view/section" and in "Export 3D mesh" (as vector texture for cutted faces or as material texture)

whitequark, 2017-01-13:
@Symbian9 What you are describing is not a pattern "for filled paths", it is a pattern for faces of solid objects. The implementation and UI is completely different.