solvespace/solvespace #25: Export of HTML files with Dimensions shown.

mmiscool, 2016-08-07:
I would like to be able to export html files with dimensions shown.

whitequark, 2016-08-07:
I'm unconvinced in the utility of this.

mmiscool, 2016-08-07:
Would be extremely beneficial. My current work environment uses PMI views to define our parts that can be consumed in to 3d work packages for use on the shop floor. The ability to spin a part in the web browser and view dimensions for the part would allow for disclosure of information to suppliers/venders.

Also it works very nicely on mobile devices like android tablets for use in the work environment.

whitequark, 2016-08-07:
Okay, I will look into it. There are some vague plans for an Android port of SolveSpace as well.

Incidentally, if you feel that SolveSpace would be useful in your company but the [existing roadmap](https://github.com/whitequark/solvespace/issues?q=is%3Aopen+is%3Aissue+milestone%3A3.0) does not bring enough attention to the features that you would like to see, [M-Labs](http://m-labs.hk) provides commercial support for SolveSpace, including development of new features. If this is something you're interested in, please contact us at [solvespace@m-labs.hk](mailto:solvespace@m-labs.hk). We will be glad to help.

Symbian9, 2016-08-08:
Also look in "Doc Scenes" add-on for Blender and its output!
* https://wiki.blender.org/index.php/Extensions:2.6/Py/Scripts/Import-Export/Doc_Scenes
* https://github.com/Antonioya/blender/tree/master/doc_scenes

whitequark, 2016-10-11:
@mmiscool BTW, maybe @cr1901 can help you here, he wrote the original three.js export.

cr1901, 2016-10-13:
I'm thinking about how to add this... exactly what dimensions would you need? Height/width/length of the minimum bounding rectangle, or something more "intricate" like circumference or radius of parts?

whitequark, 2016-10-13:
I think just the usual WYSIWYG thing. I.e. export whatever constraints are in the active group, as lines.

mmiscool, 2016-10-16:
What would truly be use full is the ability to have a list of views stored in the part.

Each view would need to store a camera position in 3d space and a zoom level. That way you could select a view and it would display it.
Each of these views could have reference dimensions in them or other annotations such as a call out for the name of a part in an assembly with a leader line. Possibly weld symbols or GD&T feature frames.  It would be good if dimensions and other annotations were specific to the particular view.

whitequark, 2016-10-16:
@mmiscool Right, so my thoughts here are: if this feature required generalized constraint rendering in 3.js, it would be a non-starter. But it seems like it can be implemented by dumping the edges corresponding to constraints from the PMI views, which is *much* simpler and is actually not much more work than just those views at all.

Closing this in favor of #26 since this is basically the same feature.

closed 2016-10-16 by whitequark