solvespace/solvespace #26: Create views with in  a part or assembly that are zoomed and contain separate dimensions.

mmiscool, 2016-08-07:
I would like to be able to define views with in a part each with there own zoom and orientation captured. Also would like to be able to create reference dimensions against each of these views that would not appear in the other views. 

Would allow for 3d disclosure of information contained with in the file.

whitequark, 2016-08-07:
Basically you want to export technical drawings. This is a planned feature, but don't expect it soon.

mmiscool, 2016-08-07:
I like the idea of PMI and defining PMI views. Not so much the paper end of it however being able to produce a "Drawing" would be beneficial. 
http://www.cadalyst.com/files/cadalyst/nodes/2006/10794/1PMI.jpg

Would much prefer to be able to produce one of those fancy html outputs that would contain a list of views for the down stream (Shop floor, customers, ect).

Could be produced in such a way that that a batch script on a web server could produce a nice navigate able package for an assemblies work order. Or put in a product catalog automatically for use in a sales web site.

whitequark, 2016-08-07:
What's PMI?

> Would much prefer to be able to produce one of those fancy html outputs that would contain a list of views for the down stream (Shop floor, customers, ect).

Hm, yes, that's possible. Should the view parameters be defined in the file itself, or should it be passed to the CLI tool that generates it (when such a CLI tool is used)?

> Could be produced in such a way that that a batch script on a web server could produce a nice navigate able package for an assemblies work order. Or put in a product catalog automatically for use in a sales web site.

Yeah, most of the infrastructure for that is in 3.0 already.

mmiscool, 2016-08-07:
If you are free and would like to have a bit of a conversation I am in the US, have skype, google hangouts, cell phone, ect. Could have a discussion. I started playing around building from source and have started poking at things. Would like to get involved.

whitequark, 2016-08-07:
Do you have Jabber/XMPP? I could try setting up Skype if not...

mmiscool, 2016-08-07:
I can set up what ever you are use to. What would be a good jabber service.
otherwise admin*no ***** spam@smbisoft.com will work for google hang outs.
Sype name is bobobob10

whitequark, 2016-08-07:
jabber.org is fine. My Jabber ID is whitequark@jabber.ru.