solvespace/solvespace #27: Enhancement : Allow change of default of not allowing redundant constraints.

mmiscool, 2016-08-07:
I would like to be able to change a setting that makes it so that the check box for allowing redundant constraints can be turned on by default and turned off at user discretion.

whitequark, 2016-08-07:
Sorry, no, this is a bad idea. The solver does not handle redundant constraints well, and it is unlikely that you will be able to do anything complicated with the redundant constraints allowed. The mode exists solely in order to constrain assemblies, where positioning them in 3d without overconstaining is very awkward.

whitequark, 2016-08-07:
(On top of that you will find that reducing the amount of constraints to the bare minimum lets you produce sketches that are much easier to modify.)

closed 2016-08-07 by whitequark