solvespace/solvespace #29: Enhancement: Generalized constraint creation.

mmiscool, 2016-08-08:
Would be useful to display a list of valid constraint types for the selected objects.

Desired workflow for creating a constraint.

Click "Constraint" command --> Select items to be constrained --> Select from a list in the "property browser window" of possible valid constraints. 

As more items get selected the list would get smaller. For example I could select 2 lines. It would show all the constraints I could do with 2 lines. As soon as I click on a point. It shows things I could do with 2 lines and a point.

mmiscool, 2016-08-08:
Would also be good to show a list of invalid constraints and what is missing to make it valid. 

For example.

phkahler, 2022-07-05:
No. Selecting a general constraint command, then selecting entities, then a specific constraint? That first command is almost completely redundant.

As it is, you just select entities and apply a specific constraint.

@ruevs I think we should close this.

ruevs, 2022-07-05:
@phkahler the part where, once one selects some entities a list of the possible constraints that can be applied to them is shown, could be useful for a complete novice. We could do it in this "beautiful" function - it will just get slightly longer ;-) https://github.com/solvespace/solvespace/blob/b5333608e9d4826fcb6dc497a9e94e162267a84c/src/describescreen.cpp#L64

Anyway, just like you I find this superfluous.

closed 2023-08-09 by phkahler