solvespace/solvespace #30: Export mesh to AC3D (for assembled models)

Symbian9, 2016-08-08:
Allow export triangle mesh to *.AC (AC3D) file format specially for assembled models.

AC3D format very simple and very powerfull especially for models with parented elements. It will be usefull for export assembled models ready for animating in Blender.

REFERENCES:
* http://www.inivis.com/ac3d/man/ac3dfileformat.html
* http://paulbourke.net/dataformats/ac3d/
* https://github.com/NikolaiVChr/Blender-AC3D
* http://forum.freecadweb.org/viewtopic.php?t=11951
* http://away3d.com/livedocs/2.5.2_lib/away3d/loaders/AC3D.html
* http://developer.x-plane.com/?article=creating-animated-objects-with-ac3d-6-1

Symbian9, 2016-08-08:
Related thing is groups in Three.JS
* http://stackoverflow.com/questions/7985805/three-js-mesh-group-example-three-object3d-advanced

whitequark, 2016-08-08:
As I've said before, I have no plans of implementing AC3D export. You are free to implement it yourself or use a converter from any other supported format.

closed 2016-08-08 by whitequark

whitequark, 2016-08-08:
On top of that, the full model hierarchy is not currently accessible for export anyway, so even if it was implemented today, that would not help you to animate anything.

Symbian9, 2016-08-08:
C source code for an AC3D file loader/viewer. Requires Opengl and Glut. Windows exe, VC++ project files and unix makefile included.
* * http://www.inivis.com/ac3d/download/AC3Dloader113.zip

Think it can be easy rewrited to AC3D exporter.

whitequark, 2016-08-08:
No, this is a loader. Completely irrelevant.

Symbian9, 2016-08-08:
Here is Blender exporter that well documented
https://github.com/NikolaiVChr/Blender-AC3D

Anyway, STL and OBJ not so helpfull as AC3D for store multiparts mesh objects.

Symbian9, 2016-08-08:
SolveSpace already has multipart models - assemble ('asm')

whitequark, 2016-08-08:
Models. But there is always only one mesh, joined together from all constituent parts, and it cannot be exported separately.

Evil-Spirit, 2016-08-08:
For my own I recently updated some obj export to support materials. I can push it if needed. Obj is enough for mesh output.

Symbian9, 2016-08-08:
OBJ or STL good for single (as one part) element, but if need mechanism mesh its better AC3D, because its save parts inside mesh according each reference points (local coordinates).

Symbian9, 2016-08-20:
One more AC3D related sourcecode
* https://github.com/creaktive/ACFTools