solvespace/solvespace #31: Show standart data point in "g001-#references"

Symbian9, 2016-08-08:
Currently standart point (at 0.0, 0.0, 0.0) not listed in group "g001-#references", but when select this point directly then in property browser can see string <code>IN GROUP g001-#references</code>

Please, fix it and show this point as other points listed in next groups.

jwesthues, 2016-08-08:
The list in the browser is of requests (like a workplane created with Sketch -> Workplane), not entities (like the workplane's automatically-generated normal and origin point).

The point that you've selected is the origin of a workplane. The workplanes are listed in `g001`'s browser screen. The reference group behavior is consistent with that of other groups.

closed 2016-08-08 by jwesthues

Symbian9, 2016-08-08:
Then 'origin' must be listed before "g001-#references" as "g000-#origin".

Also axis must be presented in too ('X axis', 'Y axis', 'Z axis'). Now if select in general window one of three main axis its showed as 'NORMAL', but this three 'normals' in some way special.

jwesthues, 2016-08-08:
The references group is special, because it gets created automatically and can't be edited. The origin point and normals aren't: they function identically to any other point or normals in an earlier group.

Symbian9, 2016-08-08:
Axis are special type of normals at least because they have own style.

If i can select orgin point in main window, then I must have a way select it using only property browser too.

Or orgin must be visualized in the center of axis navigator too (left-down corner of main window).

jwesthues, 2016-08-08:
The x-y-z normals are indeed drawn in red-green-blue. That doesn't change their function.

You can't select e.g. the endpoints of a line segment, or a stepped-and-repeated circle, from the browser either. It selects requests, not entities.

The axes are also drawn translated because a vector means the same thing regardless of where you draw its tail. To draw the origin someplace other than the origin makes no geometric sense.

Symbian9, 2016-08-08:
Main reason for showing it is situation when two or more points have same placement but not constrained, then its to hurd select one needed point in main window by mouse. For this situation selecting directly in property browser is a way how solve it.

jwesthues, 2016-08-08:
A better UI for that case might be nice, for reasons not limited to the origin (like to show a context menu of all entities close to the mouse pointer or whatever). For now, you can always get to the reference origin by showing only the reference group.