solvespace/solvespace #32: Lathe difference broken

jockbert, 2016-08-09:
Using **Group rotating on active sketch** (Shift+L) together with using solid model as **difference** seem to be broken.

Way to reproduce:
1. Extrude a rectangle to a box.
2. Sketch e.g. a circle on one of the box sides.
3. Group rotate circle sketch (Shift+L) and use difference.

I'm using SolveSpace version 2.1~dce90646 on Linux Mint 17.3. 

[image]
[failed lathe.slvs.zip](https://github.com/solvespace/solvespace/files/408873/failed.lathe.slvs.zip)

_BTW: Thanks for the effort to provide a free and usable 3D CAD program for Linux (too). The best alternative, OpenSCAD, only works for some use cases. Truly appreciated._

whitequark, 2016-08-09:
This is a known limitation with the SolveSpace NURBS backend. There are no plans to improve the backend for now. Check the "Force NURBS surfaces to triangle mesh" option in the group properties every time this issue arises, as a workaround.

whitequark, 2016-08-09:
Generally, you can achieve some success with getting the NURBS operation to proceed by making sure the NURBS surfaces do not coincide with each other. I.e. in this case, try making your torus's center slighly off the edge of the rectangle. This is, of course, inconvenient to do.

closed 2016-08-09 by whitequark

ruevs, 2020-06-23:
@whitequark would you please label this 'NURBS' so I can find it easier some day in the future.

whitequark, 2020-06-23:
done! feel free to send me any other issues to tag, I appreciate this

phkahler, 2020-10-15:
This is actually fixed now.

ruevs, 2020-10-15:
Fixed on 2020-10-09 22:10:22 in commit d72eba8039c52f5e8d3ab36ec2d8a7598dfd578b
Pull request https://github.com/solvespace/solvespace/pull/733

I am adding this comment just so that things are cross-referenced.