solvespace/solvespace #37: Crash when using symmetric contraint

hean01, 2016-08-28:
Easily reproduced by as:

 - Create two circles
 - Select the center points
 - Setup symmetric constraint

Here follows a backtrace for this crash:

```
Program received signal SIGSEGV, Segmentation fault.
0x00000000005b5aa9 in SolveSpace::Constraint::MenuConstrain (id=<incomplete type>)
    at /home/henrik/Development/upstream/solvespace/src/constraint.cpp:373
373	                c.entityA = gs.entity[0];
Missing separate debuginfos, use: dnf debuginfo-install atk-2.18.0-1.fc23.x86_64 atkmm-2.24.1-1.fc23.x86_64 bzip2-libs-1.0.6-19.fc23.x86_64 cairo-1.14.2-2.fc23.x86_64 cairomm-1.12.0-1.fc23.x86_64 expat-2.1.1-2.fc23.x86_64 fontconfig-2.11.94-5.fc23.x86_64 freetype-2.6.0-3.fc23.x86_64 gdk-pixbuf2-2.32.3-1.fc23.x86_64 glib2-2.46.2-2.fc23.x86_64 glibmm24-2.46.2-1.fc23.x86_64 graphite2-1.3.6-1.fc23.x86_64 gtk2-2.24.30-1.fc23.x86_64 gtkmm24-2.24.4-7.fc23.x86_64 gtk-murrine-engine-0.98.2-10.fc23.x86_64 harfbuzz-1.0.6-1.fc23.x86_64 json-c-0.12-6.fc23.x86_64 libdatrie-0.2.8-6.fc23.x86_64 libffi-3.1-8.fc23.x86_64 libgcc-5.3.1-6.fc23.x86_64 libpng-1.6.23-1.fc23.x86_64 libselinux-2.4-4.fc23.x86_64 libsigc++20-2.6.2-1.fc23.x86_64 libstdc++-5.3.1-6.fc23.x86_64 libthai-0.1.21-2.fc23.x86_64 libX11-1.6.3-2.fc23.x86_64 libXau-1.0.8-5.fc23.x86_64 libxcb-1.11.1-1.fc23.x86_64 libXcomposite-0.4.4-7.fc23.x86_64 libXcursor-1.1.14-4.fc23.x86_64 libXdamage-1.1.4-7.fc23.x86_64 libXext-1.3.3-3.fc23.x86_64 libXfixes-5.0.1-5.fc23.x86_64 libXi-1.7.6-1.fc23.x86_64 libXinerama-1.1.3-5.fc23.x86_64 libXrandr-1.5.0-2.fc23.x86_64 libXrender-0.9.9-2.fc23.x86_64 mesa-libGLU-9.0.0-9.fc23.x86_64 pango-1.38.1-1.fc23.x86_64 pangomm-2.38.1-1.fc23.x86_64 pcre-8.39-2.fc23.x86_64 pixman-0.34.0-1.fc23.x86_64 zlib-1.2.8-9.fc23.x86_64
(gdb) bt
#0  0x00000000005b5aa9 in SolveSpace::Constraint::MenuConstrain(SolveSpace::Command) (id=<incomplete type>)
    at /home/henrik/Development/upstream/solvespace/src/constraint.cpp:373
#1  0x00000000005a7e89 in SolveSpace::MainMenuItem<Gtk::MenuItem>::on_activate() (this=0xfcc0b0)
    at /home/henrik/Development/upstream/solvespace/src/platform/gtkmain.cpp:935
#2  0x00007ffff753c910 in Gtk::MenuItem_Class::activate_callback(_GtkMenuItem*) () at /lib64/libgtkmm-2.4.so.1
#3  0x00007ffff54a69d4 in _g_closure_invoke_va () at /lib64/libgobject-2.0.so.0
#4  0x00007ffff54c12bd in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
#5  0x00007ffff54c18ff in g_signal_emit () at /lib64/libgobject-2.0.so.0
#6  0x00007ffff6c38bdc in closure_accel_activate () at /lib64/libgtk-x11-2.0.so.0
#7  0x00007ffff54a67a5 in g_closure_invoke () at /lib64/libgobject-2.0.so.0
#8  0x00007ffff54b8851 in signal_emit_unlocked_R () at /lib64/libgobject-2.0.so.0
#9  0x00007ffff54c1122 in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
#10 0x00007ffff54c18ff in g_signal_emit () at /lib64/libgobject-2.0.so.0
#11 0x00007ffff6a328af in gtk_accel_group_activate () at /lib64/libgtk-x11-2.0.so.0
#12 0x00007ffff6a33d0d in gtk_accel_groups_activate () at /lib64/libgtk-x11-2.0.so.0
#13 0x00007ffff6c52ce1 in gtk_window_activate_key () at /lib64/libgtk-x11-2.0.so.0
#14 0x00007ffff6c52d78 in gtk_window_key_press_event () at /lib64/libgtk-x11-2.0.so.0
#15 0x00007ffff759e904 in Gtk::Widget::on_key_press_event(_GdkEventKey*) () at /lib64/libgtkmm-2.4.so.1
#16 0x000000000059cc49 in SolveSpace::GraphicsWindowGtk::on_key_press_event(_GdkEventKey*) (this=0xe59300, event=0x1013d90)
    at /home/henrik/Development/upstream/solvespace/src/platform/gtkmain.cpp:698
#17 0x00007ffff75a0924 in Gtk::Widget_Class::key_press_event_callback(_GtkWidget*, _GdkEventKey*) () at /lib64/libgtkmm-2.4.so.1
#18 0x00007ffff6b0a75d in _gtk_marshal_BOOLEAN__BOXED () at /lib64/libgtk-x11-2.0.so.0
#19 0x00007ffff54a67a5 in g_closure_invoke () at /lib64/libgobject-2.0.so.0
#20 0x00007ffff54b8dfe in signal_emit_unlocked_R () at /lib64/libgobject-2.0.so.0
#21 0x00007ffff54c1122 in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
#22 0x00007ffff54c18ff in g_signal_emit () at /lib64/libgobject-2.0.so.0
#23 0x00007ffff6c3a79c in gtk_widget_event_internal () at /lib64/libgtk-x11-2.0.so.0
#24 0x00007ffff6b08b27 in gtk_propagate_event () at /lib64/libgtk-x11-2.0.so.0
#25 0x00007ffff6b08e0b in gtk_main_do_event () at /lib64/libgtk-x11-2.0.so.0
#26 0x00007ffff651186c in gdk_event_dispatch () at /lib64/libgdk-x11-2.0.so.0
#27 0x00007ffff1f83e5a in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#28 0x00007ffff1f841f0 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#29 0x00007ffff1f84512 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#30 0x00007ffff6b07e57 in gtk_main () at /lib64/libgtk-x11-2.0.so.0
#31 0x00007ffff753a9cf in Gtk::Main::run(Gtk::Window&) () at /lib64/libgtkmm-2.4.so.1
#32 0x000000000059a7d6 in main(int, char**) (argc=1, argv=0x7fffffffdcf8)
    at /home/henrik/Development/upstream/solvespace/src/platform/gtkmain.cpp:1668
(gdb) 

```

whitequark, 2016-08-28:
Which git commit?

hean01, 2016-08-28:
This was tip of master, commit b9d375652c9adfe6b410ea4f4042644c87f102cc

whitequark, 2016-08-28:
Thanks for the report. Fixed in ffd9c62.

closed 2016-08-28 by whitequark