solvespace/solvespace #41: DXF export does not work if filename has spaces in path

whitequark, 2016-09-24:
http://solvespace.com/forum.pl?action=viewthread&parent=1549; should not be hard to fix.

@Evil-Spirit I told you that libdxfrw's file save routine should not be used... now you see why.

Evil-Spirit, 2016-09-25:
@whitequark, ok I will fix it.

whitequark, 2016-09-25:
@Evil-Spirit I think that just needs another constructor that lets us pass FILE* and then ssfopen

Symbian9, 2016-09-29:
As for me, better way do not use spaces in file and folder naming and replace them with "_" symbol never.

Evil-Spirit, 2016-10-05:
Probably fixed (I don't have such problem with spaces)
But for unicode file names it works better.
https://github.com/Evil-Spirit/libdxfrw/commits/wide-file-names
https://github.com/Evil-Spirit/solvespace/commits/fix-dxf-path

whitequark, 2016-10-06:
@Evil-Spirit No, this is not a proper fix. *Never* use std::wstring except on Windows when interacting with WinAPI functions. See http://utf8everywhere.org/ for explanation.

Evil-Spirit, 2016-10-06:
@whitequark,
Is there any way to make work ifstream with utf8?

whitequark, 2016-10-06:
@Evil-Spirit No. Make it a string stream or something, I guess. That would anyway be handy for if we don't have a filesystem.

Evil-Spirit, 2016-10-08:
fixed again
https://github.com/Evil-Spirit/libdxfrw/commits/wide-file-names
https://github.com/Evil-Spirit/solvespace/commits/fix-dxf-path

whitequark, 2016-10-09:
Wow, std::fstream is a portability disaster. No portable (i.e. that works with MinGW) way at all to open a fstream with a wide path...

Fixed in master and 2.x.

closed 2016-10-09 by whitequark