solvespace/solvespace #42: Can't see files with uppercase extension names

Symbian9, 2016-09-26:
Currently SolveSpace not see supported file formats in filedialogs if their extension name include uppercase letters. For example, if in one folder there are "draft1.slvs",  "draft2.SLVS", "draft3.SlvS", "draft.SLVs" program can see only those that are with ".slvs" (lowercase) extension name.
Same problem for all other supported formats in all file choosing dialogs.

P.S.: As I know on Linux its possible hold in one folder files like "drawing.DXF" and "drawing.dxf", but on Windows its impossible. So, maybe its only Linux-related bug. 

My platform: LinuxMint 17.3 XFCE 
Program version: 2.1 (release)

whitequark, 2016-10-09:
Fixed in master.

whitequark, 2016-10-09:
Note that files with mixed-case extensions will not be shown (e.g. only `slvs` and `SLVS`, not `SLVs`). Fixing that is not worth the effort.

closed 2016-10-09 by whitequark

Symbian9, 2016-10-09:
Think, this issue can be marked as 2.2 milestone
* https://github.com/solvespace/solvespace/milestone/1

whitequark, 2016-10-09:
I've backported the fix to 2.x.