solvespace/solvespace #43: Add a better explanation on how to use the three.js html export

reportingsjr, 2016-09-28:
Currently you have to either scroll through the 46k line html file and notice that there are multiple js libraries, or read someone else's embedded code to know how to embed just the exported js in a page.

Some options that I think would help explain this:
* A comment at the top of the html file either stating that there are three libraries that need to be included if one wishes to directly embed the js (three.js, hammer.js, and SolveSpaceControls.js), or explain that it should be embedded as an iframe
* An archived export that contains the necessary javascript files separately + a much smaller html file
* A page on the website that explains how to use the different three.js exports (how to directly embed the  .js only export, what to do with the html export, etc)

reportingsjr, 2017-09-20:
I was thinking an okay solution for this would be to provide another export that is a .tar.gz of the three libraries and the  js mesh. If this is acceptable to you I'll make a pull request soon-ish.

whitequark, 2017-09-20:
How would you create the .tar.gz archive? How should Windows users read it? Should be a zip if anything.

BPLRFE, 2017-09-22:
7-Zip is often used on Windows Systems.
It can handle GZIP-Archives.
http://www.7-zip.org/

reportingsjr, 2017-09-22:
tar is a pretty simple format so I was just going to roll that by hand. Zlib is already included to inflate the javascript libraries, so there won't be any extra dependencies for that.

I didn't think about windows users, but is a program that can handle tarballs unexpected for someone who wants to embed this in a webpage?

cr1901, 2017-09-22:
I keep forgetting that the js library versions are pinned (and for good reason; `SolvespaceControls` doesn't work with the current version of THREE.js- on my todo list):

While I think separate files for a full viewer (model + HTML) is a good idea, perhaps as a third option (in addition to "model + HTML one file" and "model only"), I don't personally see the point of generating a tarball/zip for the following reasons:
* Although multiple outputs will be generated from a file dialog, you only really have control of the filename of the single html file and .js model file; the latter filename can be inferred from the former. This isn't much different from using your browser file dialog to save a "Complete Webpage".
* Tar/zip may be simple, but it's still a bit of work to get right. Better to use the system libraries if compressing to a single file is important.

rpavlik, 2022-07-08:
So I think we do have the two "whole thing" and "just data" exports now. I have tested this fairly recently too, when working on the Debian package, since there we use the separately packaged threejs. I also fixed the compatibility issue some time ago so we're using a more modern version. Documentation is still lacking, but it actually works now both in source builds and the Debian package, which is more than I can say about the first time I tried years ago 😁