solvespace/solvespace #44: Add properties in exported PDF file

Symbian9, 2016-09-29:
I propose add small addition to PDF exporter, that can add some extra info in PDF file properties.
For example, it would be usefull add in 'Keywords' property info about exported scale like <code>Scale 1:10</code> or <code>M 1:72</code>.
Also it would be usefull add info about chord tolerance or some other

whitequark, 2016-09-29:
Don't expect me to implement this. The PDF file format is very annoying, and the metadata section of the specification is especially bad.

Symbian9, 2016-09-29:
@whitequark, there are many other developers that can help do that. I not say that this task must be completed by you ;-)

whitequark, 2016-10-10:
Ref: http://blog.crossref.org/2007/08/metadata_in_pdf_1_strategies.html

whitequark, 2016-11-20:
I looked into this again. Unfortunately I do not see how to add metadata in a way that will be universally readable by PDF readers. There are two methods of adding metadata:

  1. "Document Information Dictionary", which defines a very small amount of fields, and where unknown fields are ignored;
  2. "Metadata Stream", which requires the use of a predefined RDF schema, and I believe there is no such schema that is suitable for SolveSpace.

Therefore I don't see how is it possible to implement this.

closed 2016-11-20 by whitequark