solvespace/solvespace #45: Need only one "force NURBS surfaces to triangle mesh" checkpoint

Symbian9, 2016-10-01:
As this option force all elements in SLVS, there no need add this checkbox in every group.
Think it would be better add only one checkbox at the lower part of main screen of "Property Browser" (under string <code>line styles / view / configuration</code>

whitequark, 2016-10-01:
> As this option force all elements in SLVS, there no need add this checkbox in every group.

That's not the case. It only forces the group in which it is checked as well as subsequent groups. Previous groups are not affected.

closed 2016-10-01 by whitequark

Symbian9, 2016-10-01:
@whitequark, I test it now and you are wrong!

<b>When I activate this function in group "g005-...", then all groups above (g001 - g004) and under (g006 ...) it also affected!</b>

Program: SolveSpace version 2.1~dce90646
Platform: Linux Mint 17.3 XFCE (64bit)

whitequark, 2016-10-01:
See https://github.com/solvespace/solvespace/blob/53482d3c5943be2fbe4541dd87b31a2477483857/src/groupmesh.cpp#L316-L355.

Symbian9, 2016-10-01:
@whitequark, I don't look in code, I test SolveSpace in real life and give you feedback.

Currently "force NURBS..." affected to all groups inside SLVS and don't depend from group order when I test this function by myself. 

So, think, its a bug of SolveSpace. Are you really try reproduce it?

whitequark, 2016-10-01:
That's not the case. E.g. activate an earlier group and then export NURBS surfaces as STEP.

jwesthues, 2016-10-01:
Maybe @Symbian9 is expecting the final model to contain a mixture of mesh and NURBS surfaces, vs. switching to mesh at the first group where the flag is set and thereafter? But that's not how the Booleans work.

Symbian9, 2016-10-01:
@jwesthues, @whitequark

Steps to reproduce:
1) run SolveSpace
2) create a circle in plane (group "g002-sketch-in-plane")
3) extrude it (you will get group "g003-extrude")
3) create new plane in any place (you will get group "g004-sketch-in-plane") and create one more circle in this new plane
4) extrude this circle (you will get group "g005-extrude").
5) now you will get two cylindres; try make them without crossing.
6) now activate "force NURBS..." in group "g004-extrude", and you will see that its also affected to "g003-extrude" (both cylindres will look like N-corners prisms).

jwesthues, 2016-10-01:
Like I said above, the setting is working as designed, just not as you expect. The setting is per-group so that you can do as much as possible with NURBS Booleans (which give better mesh quality, and permit STEP export, but fail with some complex surfaces) before switching to mesh Booleans. The model will never contain a mix of mesh and NURBS.

ruevs, 2016-10-01:
This is due to the "chord tolerance" setting, it has nothing to do with force NURBS.

Symbian9, 2016-10-01:
@ruevs, yes. And I'm talking now about that if any group has "force NURBS...", ALL groups displayed according "chord tolerance". And visually its impossible detect what group has "force NURBS..." function ON, and what has it OFF.

whitequark, 2016-10-01:
> And visually its impossible detect what group has "force NURBS..." function ON, and what has it OFF.

It is possible. Activate the group and it will go back to smooth view if that particular group wasn't forced to NURBS.

Symbian9, 2016-10-01:
Why not show "smooth" and "flat" models at one moment?

jwesthues, 2016-10-01:
@Symbian9, The Booleans do NURBS against NURBS, or mesh against mesh. They don't do a mix.

Symbian9, 2016-10-01:
A mean about what I see in main window, not what I would get in export.

I expect, as user, see at one moment what solid a forced and what are not.