solvespace/solvespace #48: Allow export flat closed sketches as flat faces in 3D mesh

Symbian9, 2016-10-05:
Please, allow export "sketch-in-plane" as flat double sided faces in mesh format.

whitequark, 2016-10-05:
What's the use case?

Symbian9, 2016-10-06:
It would be usefull for sheeting details, were extruding no needed

Like glass in windows.
I will post some screenshot soon.

whitequark, 2016-10-06:
Note that such zero-volume meshes are not valid, and are likely to cause all sorts of breakage in other software.

Symbian9, 2016-10-09:
As for me, target application for editing exported mesh is Blender.
And Blender has no any issues with meshes, that include flat faces.

Also, i don't know any 3D editor that has issues with flat planes.

whitequark, 2016-10-09:
I thought about this and I don't see a need for a dedicated feature. You can constrain the top face to the bottom face (e.g. select any point from the bottom face, top face, and constrain coincident) and then export the result.

whitequark, 2016-10-09:
Note that this will, as I've said earlier, result in a mesh with naked edges. I don't think it's a good idea to have a feature dedicated to exporting invalid meshes.

closed 2016-10-09 by whitequark