solvespace/solvespace #49: OSX build fails in a fresh install

seclorum, 2016-10-06:
Freshly cloned the repo, precisely followed the instructions for OSX build in the README.md, got this far:

`[ 85%] Copying resource threejs/hammer-2.0.8.js.gz
[ 85%] Copying resource threejs/SolveSpaceControls.js
[ 85%] Built target resources
src/CMakeFiles/solvespace.dir/build.make:286: *** missing separator.  Stop.
make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
make: *** [all] Error 2`

[build.make](https://github.com/solvespace/solvespace/files/513174/build.make.txt)

seclorum, 2016-10-06:
MacOS Sierra 10.12, latest homebrew as of today (updated, etc.)

whitequark, 2016-10-06:
Unable to reproduce, but I'm using an older macOS (10.8.5); since Yosemite it became much more annoying to install it in a VM.

Maybe your make isn't right? Try gmake.

trimailov, 2016-10-10:
I have the same problem

```
↪ make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
```

On my machine (macOS sierra 10.12) default `make` is the `gmake`. 

version of `cmake`:
```
↪ cmake --version
cmake version 3.6.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

whitequark, 2016-10-10:
@trimailov Can you post the offending file (presumably `src/CMakeFiles/solvespace.dir/build.make`) to pastebin?

trimailov, 2016-10-10:
https://gist.github.com/trimailov/9fa75b352f4b77e8375d62c753a3e8c7

whitequark, 2016-10-10:
Ah I see. Can you post the output of `otool -XD /usr/lib/libz.dylib` ?

trimailov, 2016-10-10:
```
/usr/lib/libz.dylib:
/usr/lib/libz.1.dylib
```

whitequark, 2016-10-10:
Can you apply this diff and try again?
```
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 482c308..48e76e6 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -213,7 +213,7 @@ if(APPLE)
         get_filename_component(name ${lib} NAME)
         set(target ${CMAKE_CURRENT_BINARY_DIR}/solvespace.app/Contents/MacOS/${name})
 
-        execute_process(COMMAND otool -XD ${lib}
+        execute_process(COMMAND sh -c "otool -D ${lib} | tail -n1"
             OUTPUT_VARIABLE canonical_lib OUTPUT_STRIP_TRAILING_WHITESPACE)
         add_custom_command(TARGET solvespace POST_BUILD
             COMMAND ${CMAKE_COMMAND} -E copy ${lib} ${target}
```

trimailov, 2016-10-10:
It worked. Thanks!

It builds and `build/src/solvespace.app` runs.

trimailov, 2016-10-10:
@whitequark should I make a pull request, or you will commit this yourself?

whitequark, 2016-10-10:
Fixed in master and 2.x.

closed 2016-10-10 by whitequark