solvespace/solvespace #50: Add Retina support

virtualritz, 2016-10-06:
The OS X version's main window looks pixelated on Retina MacBooks.

whitequark, 2016-10-06:
To fully fix this someone would have to redraw all the toolbar icons at 2x resolution. As I don't know anyone willing to do it, there's no ETA for that. If someone provides the icons I will implement the rest (which is simple after some recent changes).

virtualritz, 2016-10-06:
I am happy to do that. I will do SVG versions of the icons. The PNGs can be rendered from those on the fly at any desired resolution, as part of the build.

virtualritz, 2016-10-06:
Also note that this is not a requirement. It is better to have the icons scaled and the viewport at retina res than keeping it scaled just because the icons are not there. That's the approach most apps take that have icon dependencies. E.g. Qt Creator was retina capable long before they updated the icons.

whitequark, 2016-10-06:
> It is better to have the icons scaled and the viewport at retina res than keeping it scaled just because the icons are not there.

Correct. However, until recently, the code was structured such that all icon placement, text rendering, and so on was done in absolute coordinates, and not in a way that would have easily let me use two sets of them. Now it's better, and also I've recently added support for asset loading.

> I am happy to do that. I will do SVG versions of the icons. The PNGs can be rendered from those on the fly at any desired resolution, as part of the build.

Thanks! That works for me. We can probably just check .png's into the source to avoid multiplying external dependencies that almost no one will need (e.g. imagemagick is a nightmare on Windows).

The context is that I have recently got rid of all generation during build to simplify cross-compilation, mostly by making loading the in-tree format much faster and then moving the loading into runtime. This is unfortunately not an option for SVGs without fairly heavy dependencies and runtime penalty.

Evil-Spirit, 2016-10-06:
The icons can be rather simple. So, it can be drawn with solvespace itself)

Evil-Spirit, 2016-10-06:
[image]

whitequark, 2016-10-06:
The icons need to be aligned to the pixel grid during PNG export, which isn't quite that simple.

virtualritz, 2016-10-07:
I'm using Inkscape for the icons. I'm new to solvespace so doing it in that app (while appealing) would eat too much time for me now. I can't even figure out how to change the radius of a circle after creating it. :)
Aligning to a pixel grid is easy in Inkscape too.

virtualritz, 2016-10-07:
Here are some samples. Let me know if this is going in the right direction. I am trying to align everything on a pixel grid so it looks as crisp as possible at 24 pixel size and any even multiplier of that.

[image]
[image]

[image]
[image]

virtualritz, 2016-10-07:
Colors are named, I'll extract them to a separate definitions SVG that the individual icon SVGs will reference. Changing colors and stroke widths in that file will then allow changing the look of all icons at once.

whitequark, 2016-10-07:
> Let me know if this is going in the right direction.

I would aim for exactly representing the current items at 24x24. It doesn't have to be pixel-perfect but the dots inside rectangular points are, I think, excessive.

virtualritz, 2016-10-07:
The dots can be removed by just setting handle_fill equal to handle_stroke. I'll do that for now.

myfreeweb, 2016-10-11:
It's not just macOS... The situation on Windows with a HiDPI screen isn't perfect too. It's possible to turn off blurry scaling (right click → properties → compatibility), but then the UI text becomes a bit too small.

Blender has the best UI scaling ever, just a DPI slider:

[image]
[image]

Adjusting at least the text size in solvespace would be great.

whitequark, 2016-10-11:
> Adjusting at least the text size in solvespace would be great.

The problem is that Unifont, which we use, is available in 16x16 and that's it. I can scale it 2x I guess.

whitequark, 2016-10-11:
You can adjust the font used for constraints and edit boxes though, that works.

whitequark, 2016-10-11:
@myfreeweb Anyway, yeah, I know about Windows HiDPI issues. Do you have any idea how to test this with Windows in a VM?

virtualritz, 2016-10-12:
What's the problem with switching to another font? Particularly a non-bitmap font which would be nice. :)

whitequark, 2016-10-12:
First, we would need:
  * a TrueType font
  * that can be distributed combined with a GPLv3 program
  * with an excellent Unicode coverage, including CJK (Unifont has nearly 100% coverage of Unicode)

After that it's a matter of rewriting the text window rendering code, which is kind of gnarly.

virtualritz, 2016-10-12:
https://github.com/FortAwesome/Font-Awesome/issues/1124

whitequark, 2016-10-12:
> Font Awesome is a full suite of 634 pictographic icons for easy scalable vector graphics on websites, created and maintained by Dave Gandy

How is this relevant in any way?

virtualritz, 2016-10-12:
The discussion is about compatibility of OFL and GPL. They seem to be compatible. If that were true, we could simply pick any font under the OFL.

whitequark, 2016-10-12:
I see. The problem is that there are very few free monospace fonts with CJK support. In fact I could only find [Noto](https://www.google.com/get/noto/help/cjk/). On top of that the all-in-one Noto package is ~89MB archived, as opposed to the current Unifont situation, which is 700KB archived.

whitequark, 2016-10-12:
For now, I feel that the following HiDPI support would be sufficient:
  * All entities in the graphics window would be smoothly scaled according to the device scale factor.
  * Icons in the graphics window, and the text window, would remain at 1x if device scale factor is less than 1.5, and at 2x if the device scale factor is more than 1.5. (Could be 1.75 here, I think whatever number Windows uses is the right one.)

Smoothly scaling SVGs at small icon sizes tends to produce ugliness; it is much better to pre-render them at known sizes (and perhaps manually hint). Smoothly scaling the (what replaces) bitmap font is perhaps more desirable, but is a lot of work, and is the bitmap font, at a readable size, really all that ugly?

virtualritz, 2016-10-12:
Two thoughts:

1. If I need CJK support, I probably won't mind downloading a larger app. And if I don't need it, I can download a 'light' version of solvespace. The build could download the font, so it doesn't have to live in the repo. 
2. Doesn't google even give you APIs to use a font w/o having to download it (assuming you have a network connection)? Could be an option too?

whitequark, 2016-10-12:
All of what you describe is strictly worse than the current behavior. In particular, not requiring any network connection to function is a feature, as is not depending on third-party infrastructure.

jwesthues, 2016-10-12:
@virtualritz One version, without network dependencies.

@whitequark I agree, though to say "entities would be smoothly scaled" isn't a great expression of what's needed. Basically, I think you'd want to double the scale factor in the gl matrices (so that all constraints, etc. with lengths hard-coded in pixels get drawn twice as big), and double all line widths expressed in pixels.

whitequark, 2016-10-12:
@jwesthues The thing is that there are devices with DPI that call for intermediate scale factors. E.g. the laptop I'm typing this at has DPI of 170. In a browser this calls for a 150% zoom to be usable. In fact this is  how I've configured SolveSpace here as well, to the extent that it permits me to.

As the result, both major non-Apple vendors of scalable UIs (which is Microsoft and whoever is maintaining Qt these days) have adopted the algorithm that I describe: assets that only look good when aligned to pixel grid come in discrete increments, the rest is scaled smoothly. Apple, of course, avoids this by shipping hardware in only two DPI choices, corresponding to 1x and 2x.

jwesthues, 2016-10-12:
@whitequark Agreed. I was referring above to the specific implementation details to scale the graphics window, and didn't mean to imply that you had to snap the "vector scale factor" to 1 or 2; though if you don't, then you'll also have to adjust for the change in relative font size to make the constraints look right.

whitequark, 2016-10-12:
> though if you don't, then you'll also have to adjust for the change in relative font size to make the constraints look right.

Yeah. Fortunately the vector font is already fully scalable and e.g. constraint label offsets were rewritten in terms of the font size, so that's a simple matter of adding a factor.

jwesthues, 2016-10-12:
>Yeah. Fortunately the vector font is already fully scalable and e.g. constraint label offsets were rewritten in terms of the font size, so that's a simple matter of adding a factor.

Wait, isn't what I wrote above wrong? Since the only font used in the graphics window is the vector font, you should be able to scale that entire UI together continuously just by multiplying a scale onto the gl matrices, and lying to the zoom-to-fit and stuff about the window dimensions in physical pixels. That looks pretty easy.

whitequark, 2016-10-12:
> Since the only font used in the graphics window is the vector font

There are the icons, the bitmap font used in tooltips as well as the in-progress message (removed for now, since the implementation doesn't work with offscreen rendering used on non-Windows platforms).

> should be able to scale that entire UI together continuously just by multiplying a scale onto the gl matrices

To get a crisp rendering at low DPIs, the vector font uses a simple form of hinting (at character boundaries only), and many entities are also aligned to the pixel grid, so it's not quite as simple as that. But it doesn't require any radical changes either, just a matter of adding a conversion factor in a few places, I think.

virtualritz, 2016-10-12:
I don't want anything scaled. The sizes of fonts, dimensions and icons in SolveSpace are perfectly fine on my Retina MacBook. They are just not crisply rendered. And no, the scale factor is not 2. There are different scale factors, depending on what you choose in the Display Preferences of OS X.

jwesthues, 2016-10-12:
>There are the icons, the bitmap font used in tooltips as well as the regeneration-in-progress message (removed for now, since the implementation doesn't work with offscreen rendering used on non-Windows platforms).

But those are basically just overlays. I don't think anything bad happens if those render at slightly different scale from the rest of the graphics window content.

> To get a crisp rendering at low DPIs, the vector font uses a simple form of hinting (at character boundaries only), and many entities are also aligned to the pixel grid, so it's not quite as simple [...]

But that hinting doesn't require you to coarsely quantize the vector font size, does it? So it's only the code that renders the font that needs to know the true physical resolution, not e.g. code that just needs to know the bounding box of the text.

whitequark, 2016-10-12:
@virtualritz That's because SolveSpace [already scales](https://github.com/solvespace/solvespace/blob/master/src/platform/cocoamain.mm#L189-L202) the UI on Retina displays. However, it does that by simply taking the offscreen rendering buffer and stretching it 2x, which is what causes the lack of crispness.

whitequark, 2016-10-12:
> I don't think anything bad happens if those render at slightly different scale from the rest of the graphics window content.

Sure, it's just a distinction that has to be made during rendering.

> So it's only the code that renders the font that needs to know the true physical resolution, not e.g. code that just needs to know the bounding box of the text.

The difference in horizontal width introduced by hinting can be considerable. E.g. (approximate numbers) in a 100 Latin character string it can be maybe 3 or 5 `a` characters. This actually has interesting effects because looking at text at any angle except truly perpendicular causes it to 'jump' as hinting gets turned off; I originally thought this would make such an implementation unviable but it is almost completely unnoticeable.

myfreeweb, 2016-10-13:
> The problem is that Unifont, which we use, is available in 16x16 and that's it. I can scale it 2x I guess.

> the all-in-one Noto package is ~89MB archived, as opposed to the current Unifont situation, which is 700KB archived.

uh. Just looked up unifont. If it's [this one](http://www.unifoundry.com/unifont.html), the first thing here is a 12 MB truetype font.

(also why not use system fonts and/or let the user choose any font)

> Do you have any idea how to test this with Windows in a VM?

I guess you could just set the display scale above 100%? (I have 150% on my 4K monitor)
[image]

whitequark, 2016-10-13:
> uh. Just looked up unifont. If it's this one, the first thing here is a 12 MB truetype font.

It's this one, but we are using the .hex.gz version, which is almost as compact as an actual bitmap. TrueType fonts, as you have noticed, are pretty bad for representing bitmaps.

whitequark, 2017-01-11:
@virtualritz Any update on the assets? Meanwhile, I've added preliminary Hi-DPI support to the GTK port as well.

Symbian9, 2017-01-17:
Maybe it would be usefull make some simple "my-own-theme.cfg" template, that would be stored in config folder, that should define how represent UI (icon set, positions of UI elements in main window, etc).

And every user can make own theme.

P.S.: it could be as "load custom theme file" option in "Property Browser"

whitequark, 2017-01-17:
No, this would be a complete waste of time.

Evil-Spirit, 2017-01-17:
@virtualritz , @whitequark,
We can render unifont in bigger size from .ttf. There is no problem to build distance-map for glyphs and perform alpha-clipping to get perfecly-resizable font in just 16x16 pixels or may be a little bit bigger like 32x32 for precision.

Evil-Spirit, 2017-01-17:
@virtualritz, @whitequark 
http://wiki.polycount.com/wiki/Transparency_map

whitequark, 2017-01-17:
@Evil-Spirit But Unifont only has 16x16 glyphs...

Evil-Spirit, 2017-01-17:
@whitequark, we can render any resolution and technique we want if it have TTF version.

Evil-Spirit, 2017-01-17:
@whitequark, It will be raster font with vector quality.

Evil-Spirit, 2017-01-17:
[image]

whitequark, 2017-01-17:
This doesn't actually work for fonts that have been specifically designed to be displayed with pixels

whitequark, 2017-02-19:
Haha wow SolveSpace is currently VERY broken on Linux HiDPI systems. Since one is now my main laptop... guess I'll have to address this.

NikiSchlifke, 2017-07-09:
Hey, is there anybody working on that right now. I use solvespace on a 276dpi laptop and it's really barely usable.
Regarding aligning svg images on a pixel grid and such, yeah that's nice but actually if you are using a HiDPI display you likely on't care. Because at above around 200dpi you can't even see the difference between naive grayscale anti-aliasing, sub-pixel anti-aliasing, font hinting or no font hinting.

So, can anybody point me on how to implement use svg icons and larger fonts on my own fork, in case anybody is into it I would then be submitting a pull request for the main fork.
Is there svg loading and rendering implemented in any libraries that are already there?

whitequark, 2017-07-09:
Just use the master branch, that already has HiDPI support.

NikiSchlifke, 2017-07-09:
Just compiling and running it won't result in any changes for me. Perhaps I'm blind I can't see any option in the ui or code that is supposed to do the scaling for that matter.
Can you point me towards the code that is supposed to do the scaling?
All I can find is 
https://github.com/solvespace/solvespace/blob/33b6e5173724ebb207c4d8450a561448b2e07e5d/src/view.cpp#L12-L19
which seems to be regarding the scale of the drawing on the screen, not scaling icons, fonts, etc...

I'm running Ubuntu 17.04 everything stock btw.

whitequark, 2017-07-10:
HiDPI support on GTK was added in commit 96476ca2e5af872c4fc0399be553951c6052254a. This doesn't actually bring you larger icons or larger text in the browser. Although I still find it quite usable, a quick and dirty way to increase icon and text size would be to scale the viewport 2x when creating `UiCanvas`.

myfreeweb, 2018-03-24:
[image]

it's really smol!! but that's better than blurry :D

myfreeweb, 2018-04-03:
@DougBeney is it just me or does it look kinda blurry? Anyway, while gnome-shell is capable of running Xwayland apps without blur, that's not true for most other compositors.

NikiSchlifke, 2018-04-03:
@myfreeweb That is to be expected since these kind of hacks can't change the actual rendering code is still the same. That is still a great help, for usability, if you need it.

There also seems to be some confusion regarding what HiDPi on the desktop actually means, so far only apple has done it right. With a 200+ dpi screen you aren't concerned about "blurriness" when scaling pixel graphics (like the solvespace icons & fonts). Normal desktop gui toolkits are designed for the "96dpi" display "standard" (actual size may differ) HiDPI screens for double of that. Everything that is not drawn as vector graphic should be scaled by 200% by "default" on these screens.

DougBeney, 2018-04-04:
Yup. It is a bit blurry.

Another thing you could try is running KDE with Wayland (Not xorg). Just learned about it today, but keep in mind KDE w/ Wayland is not perfectly stable.

The benefits are that there is no lag like there is with the method I described previously. Also, I'm pretty sure there isn't any blur. Maybe a bit with the icons, but probably not anywhere else.

whitequark, 2018-07-31:
SolveSpace now has native HiDPI support on all platforms. Icon scaling will be added once someone contributes hi-res icons.

closed 2018-07-31 by whitequark

DougBeney, 2018-07-31:
[image]

Confirmed! Nice work. Luckily, the icons don't look that bad so it's not a major problem.

Symbian9, 2018-08-05:
> Icon scaling will be added once someone contributes hi-res icons

I could try create hires icons. Can you give details, what size it should be?

virtualritz, 2018-08-05:
The icons should be created as vector graphics with pixel-grid snapped lines (pre-hinted). I started doing this last year but couldn't donate more time to this.
I may have more time soon but I'm happy to hand out what I did to someone else.

[5 more replies]