solvespace/solvespace #51: solvespace fails to build on OS X 10.12

virtualritz, 2016-10-06:
```
$ cmake .. -DENABLE_TESTS=OFF
-- The C compiler identification is AppleClang 7.0.2.7000181
-- The CXX compiler identification is AppleClang 7.0.2.7000181
-- Check for working C compiler: /Applications/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework
-- Using in-tree libdxfrw
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.8")
-- Found PNG: /opt/local/lib/libpng.dylib (found version "1.6.24")
-- Found Freetype: /opt/local/lib/libfreetype.dylib (found version "2.6.5")
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/moeller/code/solvespace/build

```
```
$ make
[ 13%] Built target dxfrw
[ 43%] Built target solvespace_cad
[ 85%] Built target resources
[ 85%] Building CXX object src/CMakeFiles/solvespace.dir/platform/cocoamain.mm.o
In file included from /Users/moeller/code/solvespace/src/platform/cocoamain.mm:11:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSArray.h:5:
/System/Library/Frameworks/Foundation.framework/Headers/NSObject.h:44:12: error: unknown property attribute 'class'
@property (class, readonly) BOOL supportsSecureCoding;
           ^
In file included from /Users/moeller/code/solvespace/src/platform/cocoamain.mm:11:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:6:
/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:262:12: error: unknown property attribute 'class'
@property (class, readonly) const NSStringEncoding *availableStringEncodings;
           ^
/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:269:12: error: unknown property attribute 'class'
@property (class, readonly) NSStringEncoding defaultCStringEncoding;    // Should be rarely used
           ^
In file included from /Users/moeller/code/solvespace/src/platform/cocoamain.mm:11:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:11:
/System/Library/Frameworks/Foundation.framework/Headers/NSNotification.h:45:12: error: unknown property attribute 'class'
@property (class, readonly, strong) NSNotificationCenter *defaultCenter;
           ^
In file included from /Users/moeller/code/solvespace/src/platform/cocoamain.mm:11:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:33:12: error: unknown property attribute 'class'
@property (class, readonly, strong) NSBundle *mainBundle;
           ^
/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:46:12: error: unknown property attribute 'class'
@property (class, readonly, copy) NSArray<NSBundle *> *allBundles;
           ^
/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:47:12: error: unknown property attribute 'class'
@property (class, readonly, copy) NSArray<NSBundle *> *allFrameworks;
           ^
In file included from /Users/moeller/code/solvespace/src/platform/cocoamain.mm:11:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:14:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSCalendar.h:7:
/System/Library/Frameworks/Foundation.framework/Headers/NSDate.h:46:12: error: unknown property attribute 'class'
@property (class, readonly) NSTimeInterval timeIntervalSinceReferenceDate;
           ^
/System/Library/Frameworks/Foundation.framework/Headers/NSDate.h:60:12: error: unknown property attribute 'class'
@property (class, readonly, copy) NSDate *distantFuture;
           ^
/System/Library/Frameworks/Foundation.framework/Headers/NSDate.h:61:12: error: unknown property attribute 'class'
@property (class, readonly, copy) NSDate *distantPast;
           ^
In file included from /Users/moeller/code/solvespace/src/platform/cocoamain.mm:11:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:14:
/System/Library/Frameworks/Foundation.framework/Headers/NSCalendar.h:108:12: error: unknown property attribute 'class'
@property (class, readonly, copy) NSCalendar *currentCalendar;                                  // user's preferred calendar
           ^
/System/Library/Frameworks/Foundation.framework/Headers/NSCalendar.h:109:12: error: unknown property attribute 'class'
@property (class, readonly, strong) NSCalendar *autoupdatingCurrentCalendar NS_AVAILABLE(10_5, 2_0); // tracks changes to user's preferr...
           ^
In file included from /Users/moeller/code/solvespace/src/platform/cocoamain.mm:11:
In file included from /System/Library/Frameworks/AppKit.framework/Headers/AppKit.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:15:
/System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:21:22: error: unknown property attribute 'class'
@property (readonly, class, copy) NSCharacterSet *controlCharacterSet;
                     ^
/System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:22:22: error: unknown property attribute 'class'
@property (readonly, class, copy) NSCharacterSet *whitespaceCharacterSet;
                     ^
/System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:23:22: error: unknown property attribute 'class'
@property (readonly, class, copy) NSCharacterSet *whitespaceAndNewlineCharacterSet;
                     ^
/System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:24:22: error: unknown property attribute 'class'
@property (readonly, class, copy) NSCharacterSet *decimalDigitCharacterSet;
                     ^
/System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:25:22: error: unknown property attribute 'class'
@property (readonly, class, copy) NSCharacterSet *letterCharacterSet;
                     ^
/System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:26:22: error: unknown property attribute 'class'
@property (readonly, class, copy) NSCharacterSet *lowercaseLetterCharacterSet;
                     ^
/System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:27:22: error: unknown property attribute 'class'
@property (readonly, class, copy) NSCharacterSet *uppercaseLetterCharacterSet;
                     ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [src/CMakeFiles/solvespace.dir/platform/cocoamain.mm.o] Error 1
make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
make: *** [all] Error 2

```

whitequark, 2016-10-06:
I believe the problem is that macOS Sierra requires XCode 8.

virtualritz, 2016-10-06:
I have the latest version of the command line tools installed. They were updated automatically via the app store after I installed Sierra. I'm installing XCode too now, just to be sure. I'll get back in a bit on if that improves matters.

whitequark, 2016-10-06:
As can be seen from your CMake log:

> -- The C compiler identification is AppleClang 7.0.2.7000181

I'm not sure how to make CMake use a specific version of XCode tools, but from what I see it is using the default toolchain:

> -- Check for working C compiler: /Applications/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc

I'm not sure how to switch that but I suppose it can be switched somewhere in the GUI. (I don't use XCode, as it is probably evident.)

virtualritz, 2016-10-07:
Apparently, the beta of XCode is not updated automagically. After deleting the beta and installing the latest XCode, the build goes through fine. So this can be closed.

closed 2016-10-07 by virtualritz