solvespace/solvespace #52: Debian Jessie fails to build solvespace

albertocb, 2016-10-06:
Any idea? (thanks by the way!)

$ make
[ 86%] Built target dxfrw
[ 86%] Built target solvespace_cad
[ 86%] Built target resources
[ 86%] Linking CXX executable solvespace
CMakeFiles/solvespace.dir/platform/gtkmain.cpp.o: In function `SolveSpace::SetCurrentFilename(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
gtkmain.cpp:(.text+0x1026): undefined reference to `Glib::ustring::ustring(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/solvespace.dir/platform/gtkmain.cpp.o: In function `SolveSpace::RefreshRecentMenu(SolveSpace::Command, SolveSpace::Command)':
gtkmain.cpp:(.text+0x1ace): undefined reference to `Glib::ustring::ustring(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/solvespace.dir/platform/gtkmain.cpp.o: In function `SolveSpace::ConvertFilters(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, SolveSpace::FileFilter const*, Gtk::FileChooser*)':
gtkmain.cpp:(.text+0x1d4a): undefined reference to `Glib::ustring::ustring(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
gtkmain.cpp:(.text+0x1ec6): undefined reference to `Glib::ustring::ustring(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/solvespace.dir/platform/gtkmain.cpp.o: In function `SolveSpace::GetOpenFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, SolveSpace::FileFilter const*)':
gtkmain.cpp:(.text+0x21ba): undefined reference to `Glib::ustring::ustring(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/solvespace.dir/platform/gtkmain.cpp.o:gtkmain.cpp:(.text+0x22f5): more undefined references to `Glib::ustring::ustring(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' follow
collect2: error: ld returned 1 exit status
src/CMakeFiles/solvespace.dir/build.make:283: recipe for target 'src/solvespace' failed
make[2]: *** [src/solvespace] Error 1
CMakeFiles/Makefile2:230: recipe for target 'src/CMakeFiles/solvespace.dir/all' failed
make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

whitequark, 2016-10-07:
Yes. This is caused by what is known in the industry as 'the libstdc++ ABI fiasco', see [this](https://gcc.gnu.org/onlinedocs/libstdc++/manual/using_dual_abi.html).

Try reconfiguring with `cmake -DCMAKE_CXX_FLAGS="-D_GLIBCXX_USE_CXX11_ABI=0"`.

whitequark, 2016-10-13:
Please reply in this issue if the solution above doesn't work for you.

closed 2016-10-13 by whitequark