solvespace/solvespace #53: LibreOffice cant open generated PDF

Symbian9, 2016-10-09:
LibreOffice has possibility to open and edit PDF format.

But when I try open any PDF's, generated by SolveSpace, LO fail on import.

whitequark, 2016-10-09:
How are you importing it exactly?

Symbian9, 2016-10-09:
1) run LibreOffice
2) "File -> Open -> PDF-Portable Document Format (Draw)"
3) select PDF generated by SolveSpace and press OK

RESULT:
> General Error
General input/output error.

whitequark, 2016-10-09:
Which libreoffice version? Mine doesn't have PDF there..

Symbian9, 2016-10-09:
LibreOffice 5.0.3.2 and LibreOffice 5.2.1.2
Result same for both.

whitequark, 2016-10-09:
I have 5.2.2.2.0+ and there is no PDF in the list of supported formats. Perhaps it was removed because it was buggy?

Symbian9, 2016-10-09:
Look deeply! There are no release notes about  remove of PDF import support, so think, 5.2.2.2 also include it.
Anyway, you can try "File -> Open -> All files -> (select pdf from SolveSpace)"

Also I can open other PDF's using installed LibreOffice.

LibreOffice 5.0.3.2 and LibreOffice 5.2.1.2
Result same for both.

Maybe proble in SolveSpace output - PDF-1.1 format

Why you not use PDF-1.4 or PDF/A-1a specifications? Both they opened in LibreOffice.

whitequark, 2016-10-09:
> Anyway, you can try "File -> Open -> All files -> (select pdf from SolveSpace)"

That just opens the PDF as text.

> Maybe problem in SolveSpace output - PDF-1.1 format?!

Maybe. But everything else I've tried so far can open these. Maybe @bofh453 knows what's wrong with SolveSpace's PDFs...

Symbian9, 2016-10-09:
http://www.technorms.com/26572/edit-save-pdf-libreoffice-openoffice

whitequark, 2016-10-09:
Yeah, I don't need guidance to navigate a menu. Pretty sure this got removed.

whitequark, 2016-10-09:
Ah, confusingly, this is an extension--I had to install `libreoffice-pdfimport` and now I can reproduce this.

Symbian9, 2016-10-09:
So can you detect is it problem of SolveSpace PDF export or LibreOffice PDF import extension now?

whitequark, 2016-10-09:
I have no idea because LibreOffice PDF import doesn't have any debug output that would indicate what the message is. I suggest reporting this against the LibreOffice PDF import plugin, and if they come back explaining exactly how our PDF export is broken, I'll fix it.

Meanwhile, just use SVG. I don't know why you bothered with importing PDF in the first place.

closed 2016-10-09 by whitequark