solvespace/solvespace #57: Segfault on Fedora 24

scott-maddox, 2016-10-09:
I managed to compile and install on Fedora 24 with the following commands:

    sudo dnf install gcc gcc-c++ libpng zlib json-c-devel fontconfig freetype \
        gtkmm24-devel pangomm-devel mesa-libGL-devel mesa-libGLU-devel cmake
    git clone https://github.com/solvespace/solvespace
    cd solvespace
    git submodule update --init extlib/libdxfrw
    mkdir build
    cd build
    cmake .. -DENABLE_TESTS=OFF
    make
    sudo make install

However, now when I run `solvespace`, all I get is the following:

    realpath(/usr/local/bin/../res): No such file or directory
    json_object_from_file: error opening file /home/smaddox/.config/solvespace/settings.json: No such file or directory
    SolveSpace!

    Segmentation fault (core dumped)

If there's any more information I can provide, please let me know.

whitequark, 2016-10-09:
Yes. Please provide a backtrace. It can be obtained by running `gdb solvespace` and then, once it crashes, `bt` in the gdb prompt.

scott-maddox, 2016-10-11:
```
[smaddox@localhost src]$ gdb solvespace
GNU gdb (GDB) Fedora 7.11.1-86.fc24
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from solvespace...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/local/bin/solvespace 
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.23.1-10.fc24.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
realpath(/usr/local/bin/../res): No such file or directory
json_object_from_file: error opening file /home/smaddox/.config/solvespace/settings.json: No such file or directory
SolveSpace!

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff3f6836a in __strcmp_sse2_unaligned () from /lib64/libc.so.6
Missing separate debuginfos, use: dnf debuginfo-install PackageKit-gtk3-module-1.1.3-2.fc24.x86_64 adwaita-gtk2-theme-3.20.2-1.fc24.x86_64 atk-2.20.0-1.fc24.x86_64 atkmm-2.24.2-2.fc24.x86_64 bzip2-libs-1.0.6-20.fc24.x86_64 cairo-1.14.6-2.fc24.x86_64 cairomm-1.12.0-2.fc24.x86_64 expat-2.1.1-2.fc24.x86_64 fontconfig-2.11.94-7.fc24.x86_64 freetype-2.6.3-2.fc24.x86_64 gdk-pixbuf2-2.34.0-1.fc24.x86_64 glib2-2.48.2-1.fc24.x86_64 glibmm24-2.48.1-1.fc24.x86_64 graphite2-1.3.6-1.fc24.x86_64 gtk2-2.24.31-1.fc24.x86_64 gtkmm24-2.24.4-9.fc24.x86_64 harfbuzz-1.2.7-1.fc24.x86_64 json-c-0.12-7.fc24.x86_64 libX11-1.6.3-3.fc24.x86_64 libXau-1.0.8-6.fc24.x86_64 libXcomposite-0.4.4-8.fc24.x86_64 libXcursor-1.1.14-6.fc24.x86_64 libXdamage-1.1.4-8.fc24.x86_64 libXext-1.3.3-4.fc24.x86_64 libXfixes-5.0.3-1.fc24.x86_64 libXi-1.7.7-1.fc24.x86_64 libXinerama-1.1.3-6.fc24.x86_64 libXrandr-1.5.1-1.fc24.x86_64 libXrender-0.9.10-1.fc24.x86_64 libcanberra-0.30-11.fc24.x86_64 libcanberra-gtk2-0.30-11.fc24.x86_64 libdatrie-0.2.9-3.fc24.x86_64 libffi-3.1-9.fc24.x86_64 libgcc-6.2.1-2.fc24.x86_64 libglvnd-0.1.1-1.fc24.x86_64 libogg-1.3.2-5.fc24.x86_64 libpng-1.6.23-1.fc24.x86_64 libselinux-2.5-9.fc24.x86_64 libsigc++20-2.8.0-1.fc24.x86_64 libstdc++-6.1.1-2.fc24.x86_64 libtdb-1.3.9-1.fc24.x86_64 libthai-0.1.24-1.fc24.x86_64 libtool-ltdl-2.4.6-12.fc24.x86_64 libvorbis-1.3.4-6.fc24.x86_64 libxcb-1.11.1-2.fc24.x86_64 mesa-libGLU-9.0.0-10.fc24.x86_64 nss-softokn-freebl-3.27.0-1.0.fc24.x86_64 pango-1.40.3-1.fc24.x86_64 pangomm-2.40.0-1.fc24.x86_64 pcre-8.39-3.fc24.x86_64 pixman-0.34.0-2.fc24.x86_64 zlib-1.2.8-10.fc24.x86_64
(gdb) bt
#0  0x00007ffff3f6836a in __strcmp_sse2_unaligned () at /lib64/libc.so.6
#1  0x000000000058f06e in SolveSpace::NeedsGl1V1Workaround() ()
#2  0x0000000000591a11 in SolveSpace::OpenGl1Renderer::DrawPixmap(std::shared_ptr<SolveSpace::Pixmap const>, SolveSpace::Vector const&, SolveSpace::Vector const&, SolveSpace::Vector const&, SolveSpace::Point2d const&, SolveSpace::Point2d const&, SolveSpace::Canvas::hFill) ()
#3  0x0000000000665bfd in SolveSpace::UiCanvas::DrawBitmapChar(char32_t, int, int, SolveSpace::RgbaColor) ()
#4  0x00000000005790e1 in SolveSpace::TextWindow::Paint() ()
#5  0x0000000000582e5d in SolveSpace::TextWidget::on_gl_draw() ()
#6  0x000000000058730c in sigc::bound_mem_functor0<void, SolveSpace::GlWidget>::operator()() const ()
#7  0x000000000058581b in std::_Function_handler<void (), sigc::bound_mem_functor0<void, SolveSpace::GlWidget> >::_M_invoke(std::_Any_data const&) ()
#8  0x000000000058e05e in std::function<void ()>::operator()() const ()
#9  0x000000000058de91 in SolveSpace::GlOffscreen::Render(int, int, std::function<void ()>) ()
#10 0x0000000000581333 in SolveSpace::GlWidget::on_expose_event(_GdkEventExpose*) ()
#11 0x00007ffff75bc017 in Gtk::Widget_Class::expose_event_callback(_GtkWidget*, _GdkEventExpose*) () at /lib64/libgtkmm-2.4.so.1
#12 0x00007ffff6b35ace in _gtk_marshal_BOOLEAN__BOXED ()
    at /lib64/libgtk-x11-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#13 0x00007ffff54d83e5 in g_closure_invoke () at /lib64/libgobject-2.0.so.0
#14 0x00007ffff54ea82d in signal_emit_unlocked_R ()
    at /lib64/libgobject-2.0.so.0
#15 0x00007ffff54f2b8f in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
#16 0x00007ffff54f343f in g_signal_emit () at /lib64/libgobject-2.0.so.0
#17 0x00007ffff6c6608c in gtk_widget_event_internal ()
    at /lib64/libgtk-x11-2.0.so.0
#18 0x00007ffff6b343d2 in gtk_main_do_event () at /lib64/libgtk-x11-2.0.so.0
#19 0x00007ffff65288df in _gdk_window_process_updates_recurse ()
    at /lib64/libgdk-x11-2.0.so.0
#20 0x00007ffff6528885 in _gdk_window_process_updates_recurse ()
    at /lib64/libgdk-x11-2.0.so.0
#21 0x00007ffff6525161 in gdk_window_process_updates_internal ()
    at /lib64/libgdk-x11-2.0.so.0
#22 0x00007ffff6525ab8 in gdk_window_process_all_updates ()
    at /lib64/libgdk-x11-2.0.so.0
#23 0x00007ffff6aadd51 in gtk_container_idle_sizer ()
    at /lib64/libgtk-x11-2.0.so.0
#24 0x00007ffff6503717 in gdk_threads_dispatch () at /lib64/libgdk-x11-2.0.so.0
#25 0x00007ffff1bb76ba in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#26 0x00007ffff1bb7a70 in g_main_context_iterate.isra ()
    at /lib64/libglib-2.0.so.0
#27 0x00007ffff1bb7d92 in g_main_loop_run () at /lib64/libglib-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#28 0x00007ffff6b33197 in gtk_main () at /lib64/libgtk-x11-2.0.so.0
#29 0x00007ffff7558aaf in Gtk::Main::run(Gtk::Window&) ()
    at /lib64/libgtkmm-2.4.so.1
#30 0x0000000000580597 in main ()
(gdb)
```

whitequark, 2016-10-11:
Already fixed in master, please pull.

closed 2016-10-11 by whitequark

scott-maddox, 2016-10-11:
Thanks, works like a charm now!