solvespace/solvespace #59: Comments inside "sketch-in-plane" not locked in plane

Symbian9, 2016-10-09:
Currently comments added in "sketch-in-plane" only parallel to this plane, but not locked in it, so can move them in 3D.

This is mistake/bug here, because if user work with "sketch-in-plane" all elements must be only in this plane.

Symbian9, 2016-10-09:
Currently comments added in "sketch-in-plane" only parallel to this plane, but not locked in it, so can move them in 3D (they has 3 coordinates instead of all other elements in plane that has only 2 coordinates).

This is mistake/bug here, because if user work with "sketch-in-plane" all elements must be only in this plane.

whitequark, 2016-10-09:
Yes, many constraints come in two kinds: free in 3d, and locked in workplane. Many others, including comments, don't... e.g. "point on face".

Symbian9, 2016-10-09:
what a reason to make comment from "sketch-in-plane" with additional 3rd coordinate???

whitequark, 2016-10-09:
There are no "comments in plane" constraints just like there are no "point in face" constraints. These constraints are exactly the same whether you are placing them while a workplane is active or not.

Symbian9, 2016-10-09:
Then why comments from 3D and comments from plane are different?

whitequark, 2016-10-09:
Hm, you are right.

whitequark, 2016-10-10:
@Evil-Spirit Can you look at this? I made this patch
```diff
--- a/src/mouse.cpp
+++ b/src/mouse.cpp
@@ -273,7 +273,20 @@ void GraphicsWindow::MouseMoved(double x, double y, bool leftDown,
     switch(pending.operation) {
         case Pending::DRAGGING_CONSTRAINT: {
             Constraint *c = SK.constraint.FindById(pending.constraint);
-            UpdateDraggedNum(&(c->disp.offset), x, y);
+            if(c->type == Constraint::Type::COMMENT &&
+               c->workplane.v != Entity::FREE_IN_3D.v) {
+                EntityBase *wrkpl = SK.GetEntity(c->workplane),
+                           *norm  = wrkpl->Normal();
+                Vector wo = SK.GetEntity(wrkpl->point[0])->PointGetNum(),
+                       wu = norm->NormalU(),
+                       wv = norm->NormalV(),
+                       wn = norm->NormalN();
+                Vector pp = (UnProjectPoint(mp).Minus(wo)).DotInToCsys(wu, wv, wn);
+                pp.z = 0.0;
+                c->disp.offset = pp.ScaleOutOfCsys(wu, wv, wn).Plus(wo);
+            } else {
+                UpdateDraggedNum(&(c->disp.offset), x, y);
+            }
             orig.mouse = mp;
             InvalidateGraphics();
             return;
```
however this makes dragging something across a workplane at an angle very awkward. This is actually not just a problem with my patch, because there is the same issue with points in workplane being dragged.

jwesthues, 2016-10-10:
The plane determines the plane in which the comment text is drawn, with "free in 3d" always drawn facing the viewer. It seems reasonable to want both kinds of text anywhere in (x, y, z), and dimension labels are usually free in three axes. Why special-case this?

whitequark, 2016-10-10:
> It seems reasonable to want both kinds of text anywhere in (x, y, z)

Hm, this is true. And also this would be a breaking change, too, although only visible when attempting to change a file.

I have also found out that "snap to grid" will put the comment into the workplane where it belongs, and this should suffice for just "tidying up", helpful as it is by aligning everything to grid. So anyone facing a problem of constraints out in 3d could just select them all and snap to grid.

jwesthues, 2016-10-10:
And @whitequark, are you referring to the fact that when the workplane isn't parallel to the display, mouse motion gets attenuated?  That's indeed annoying, but I judged the other obvious choice to be worse, for the potential to gain up mouse motion to infinity. I guess some heuristic, with a maximum gain of ~5x vs. parallel?

whitequark, 2016-10-10:
Yes. That seems like a workable compromise; though I'm not sure how to implement it. But @Evil-Spirit will probably know.

jwesthues, 2016-10-10:
I think it should be obvious to him what I mean here, and a max gain of ~5 is probably reasonable, with some kind of soft knee as you switch from "exactly follow user dragging in-plane DOF" to "ignore user accidentally dragging normal-to-plane DOF". Let me know if not.

Evil-Spirit, 2016-10-10:
@jwesthues, yes, I understand what you mean about dragging. I thinking about it many times and about what we can do with. When we have almost parrallel workplane we can just move something out of our view range (into "infinity"). This is obviously bad, but it is case where we really don't want to operate. We can just calculate precise dragging when we have view direction 0-80 degrees with normal of plane, and switch off ability to drag or just switch to current logic when we have view direction almost parallel to plane. We also need to figure out how the others CAD systems handle this case. May be there is better solution.

Evil-Spirit, 2016-10-10:
@whitequark, @jwesthues,  do we want to have constraint labels located out of workplane? I mean, what we can drag all the constraint labels out of plane, and I don't see the real needs in this. This is just breaks visualization and make it hard to implement(fix visualization), hard to read and understand.

jwesthues, 2016-10-10:
To force all constraint labels into the workplane would (a) be a breaking change, and (b) make it harder to arrange the constraint labels when viewing a sketch in non-parallel orientations, which is pretty common.

On the dragging, a soft transition would be more consistent with the rest of the program generally, like how dragging a constrained point behaves, and perhaps less surprising than a hard transition. I don't think many users would expect to lose and gain the ability to drag points around that arbitrary 20 degree threshold.

Evil-Spirit, 2016-10-10:
@jwesthues , can you imagine the case where user have to adjust constraint labels in almost parallel workplane? Obvously, the workplane is for plane alignment, is for "working in plane" and it is not intuitive to operate it and make shop drawings in such projections.

jwesthues, 2016-10-10:
A picky user would keep the viewing plane exactly parallel to the workplane, and an unpicky user won't care if the labels are displaced slightly out of plane. Both should be able to make themselves happy.

Beyond that, you can apply a constraint, projected into workplane A, on two entities in workplane B. Into which plane would you force the label?

Evil-Spirit, 2016-10-10:
@jwesthues, I am picky user. If I moved label out of plane by accident I don't have ability to bring it back exactly :(

jwesthues, 2016-10-10:
Select and snap to grid, no?

Evil-Spirit, 2016-10-10:
@jwesthues , yes, but this is not obvious solution :)

whitequark, 2016-10-10:
@Evil-Spirit Seems quite straightforward to me, and if you want very tidy sketches (I do!) then you want snap to grid anyway, or constraints will be messy.

whitequark, 2016-10-12:
Revisiting this, it seems that Snap to Grid is sufficient, so closing.

closed 2016-10-12 by whitequark