solvespace/solvespace #60: Cant place point from "sketch-in-3D" directly on circle from "sketch-in-plane"

Symbian9, 2016-10-09:
1) create circle in sketch-in-plane
2) create new group in 3D
3) sketch datum point somewhere
4) select this point and circle from sketch-in-plane and try constraint it

RESULT:
Point still in 3D and not locked in same plane as circle, so it need one more additional constraint point in plane.

But if do same with line and point, this point no need one additional constraint in plane

whitequark, 2016-10-09:
@jwesthues What was the reason that constraining a point to a circle constrains it to the cylinder? Did you just never get around to adding an in-plane branch like for the rest, or something deeper?

jwesthues, 2016-10-10:
@whitequark I'd state that in the opposite direction: the point-on-circle constraint works _only_ projected into a plane, except that it's always the plane of the circle. The general case would be elegant only if SolveSpace elegantly and generally supported ellipses, which it doesn't.

Since SolveSpace doesn't have point-on-cylindrical-surface constraints, and doesn't automatically generate curves from surfaces when you sketch in planes that cut them, the present behavior seemed most likely to be helpful.

closed 2016-10-10 by jwesthues

whitequark, 2016-10-10:
This makes sense; and the workaround isn't too complex either.

Symbian9, 2016-10-10:
Think, for "point-on-cylindrical-surface" better make as Shift+O, but current point-on-circle must give what is claimed and snap point directly on this circle in same workplane!

jwesthues, 2016-10-10:
An ellipse entity and a point-on-ellipse constraint could work like @Symbian9 hopes, and might be a nice enhancement. I never found a solver representation for ellipses that I loved (foci plus "length of string"?), and exact ellipses don't arise much outside optics and stuff.

Evil-Spirit, 2016-10-10:
@jwesthues what the problem with ellipse representation like normal(in plane, just for angle z).
pt-on-ellipse
```
x = cos(a) * r1
y = sin(a) * r2
[a = const]
```
"a" is the angle on ellipse where point is located and we can lock it by additional equation (which can be solved like substitution)
but this is not case of this discussion, can you explain about what the problems with ellipses appears when we operate only with circles?

jwesthues, 2016-10-10:
@Evil-Spirit You mean like by introducing a new point type just for ellipses, parametrized in angle? That's possible but not necessary (as it would be e.g. for the arbitrary rational polynomial case).

I mean that the analogy with lines fails for circles (since the projection of a circle into another plane isn't generally a circle), but succeeds for ellipses.

Evil-Spirit, 2016-10-10:
@jwesthues , No, I mean introducing new param for pt-on-line, pt-on-circle, pt-on-ellipse, pt-on-spline, pt-on-something... But even not introducing, just turning valA into param and use it for distances and so on. With this, we can turn hacky Satisfy function into honest through solving real equations with fixing of coordinates but rexaxing valA param... but this is again, just solver imrovement not for subject of this issue.