solvespace/solvespace #61: New plane throw point need 3 normals

Symbian9, 2016-10-10:
Currently if you make new plane throw point it generates plane with this point, that is parallel to display.

But it give only one normal throw this point, that is perpendicular to display.

Add two additional normals, where one of them oriented vertically up, and other oriented horizontally from left to right.

Every plane in SolveSpace must include at least one point with three local  normals, that are perpendicular each to other.

whitequark, 2016-10-10:
I have no idea what a "throw point" is.

Symbian9, 2016-10-10:
select only one point and click "New group in new workplane (thru given entities)"

As this new plane already has rectangle border, then its two additional normals could be calculated easy.

P.S.: I will improve my English. Soon.

whitequark, 2016-10-10:
Um, that's not how normals work. Right now every plane has exactly one normal, which is, well, normal (as in perpendicular) to the plane. The normals of the three planes from the reference groups form the axes, but there are still one per group.

If you want to get vectors (one or two) that are parallel to the axes of the plane's coordinate system, simply use the "horizontal" or "vertical" constraint with a construction segment. I don't see a reason to add this special case, even more so that it will change the file format.

closed 2016-10-10 by whitequark

jwesthues, 2016-10-10:
A "normal" in SolveSpace is a 3x3 rotation matrix. It already contains this information, and many constraints use it. Take a look, for example, at how same-orientation snaps. If you select a normal, then you'll see all three vectors (n, u, v) in the browser.

@Symbian9 I'd suggest that for high-level questions of this type, it would be more helpful to discuss on the forum than to submit issues.