solvespace/solvespace #62: Latest master fails to build

virtualritz, 2016-10-10:
Maybe some of the latest commits was missing some new header/definitions.
This is building master at 4537444db59d07110b5c3fd1c2283f736e837077.

```
$ make -j32 install
-- The C compiler identification is AppleClang 8.0.0.8000038
-- The CXX compiler identification is AppleClang 8.0.0.8000038
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using in-tree libdxfrw
-- Found PNG: /opt/local/lib/libpng.dylib (found version "1.6.25")
-- Found Freetype: /opt/local/lib/libfreetype.dylib (found version "2.7.0")
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/moeller/code/solvespace/build
[...]
[ 13%] Built target dxfrw
[ 13%] Building CXX object src/CMakeFiles/solvespace_cad.dir/exportvector.cpp.o
/Users/moeller/code/solvespace/src/exportvector.cpp:699:11: error: no matching constructor for initialization of 'dxfRW'
    dxfRW dxf;
          ^
```

whitequark, 2016-10-10:
Did you do `git submodule update` after pulling?

virtualritz, 2016-10-10:
Doh! I forgot solvespace has submodules. Never mind the noise, please.

closed 2016-10-10 by virtualritz

whitequark, 2016-10-10:
There's a reason we have the CI systems build our commits :) https://github.com/solvespace/solvespace/commits/master