solvespace/solvespace #63: Crash on OS X with latest master

virtualritz, 2016-10-10:
Just moving the windows around a bit and resizing them, I can get solvespace to crash. The location is always the same (textwin.cpp:1045):
```
Process:               solvespace [54260]
Path:                  /Applications/solvespace.app/Contents/MacOS/solvespace
Identifier:            solvespace
Version:               3.0 (3.0~4537444d)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           solvespace [54260]
User ID:               502

Date/Time:             2016-10-10 12:54:14.567 +0200
OS Version:            Mac OS X 10.12 (16A323)
Report Version:        12
Anonymous UUID:        508FCD10-DA9E-A0A1-60AD-3B8AA7D1E46C

Sleep/Wake UUID:       CF86623B-C25F-4FC3-983B-7E382B9A56EA

Time Awake Since Boot: 870000 seconds
Time Since Wake:       240000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000002000000020
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x2000000020:
    MALLOC_LARGE           0000000119879000-000000011a234000 [ 9964K] rw-/rwx SM=PRV  
--> 
    MALLOC_NANO            0000600000000000-0000600000800000 [ 8192K] rw-/rwx SM=PRV  

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	0x0000002000000020 0 + 137438953504
1   solvespace                    	0x00000001030434d0 SolveSpace::TextWindow::MouseEvent(bool, bool, double, double) + 512 (textwin.cpp:1045)
2   solvespace                    	0x00000001030491c4 -[TextWindowView mouseMoved:] + 100 (cocoamain.mm:922)
3   com.apple.AppKit              	0x00007fff7b4f12d5 -[NSTrackingArea mouseMoved:] + 150
4   com.apple.AppKit              	0x00007fff7bb301c2 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2508
5   com.apple.AppKit              	0x00007fff7bb2f42a -[NSWindow(NSEventRouting) sendEvent:] + 541
6   com.apple.AppKit              	0x00007fff7b9d02df -[NSApplication(NSEvent) sendEvent:] + 2915
7   com.apple.AppKit              	0x00007fff7b2b6009 -[NSApplication run] + 1002
8   solvespace                    	0x000000010304af5c main + 700 (cocoamain.mm:1346)
9   libdyld.dylib                 	0x00007fff92669255 start + 1
```

whitequark, 2016-10-10:
Ohhh excellent, this is the same crash as https://github.com/solvespace/solvespace/issues/5! I've been trying to reproduce it for *months*. For some reason it never happens on my dev VM.

whitequark, 2016-10-10:
Can you try applying this patch?
```
diff --git a/src/textwin.cpp b/src/textwin.cpp
index 2bdcf6e..ffab474 100644
--- a/src/textwin.cpp
+++ b/src/textwin.cpp
@@ -1023,7 +1023,7 @@ void TextWindow::MouseEvent(bool leftClick, bool leftDown, double x, double y) {
             break;
         }
     }
-    if(r >= rows) {
+    if(r >= rows || c >= MAX_COLS) {
         SetMousePointerToHand(false);
         goto done;
     }
```

virtualritz, 2016-10-10:
That seems to fix it. No crashes after moving and resizing windows frantically for 2 minutes. :)

whitequark, 2016-10-10:
Fixed in master and 2.x.

closed 2016-10-10 by whitequark

thetreknerd, 2019-08-01:
Crash on launch.  OSX.  I'll come back and try again in a few years.