solvespace/solvespace #67: Modal error dialog box should respond to OK as well as escape

d235j, 2016-10-11:
The modal error dialog box that appears when attempting to set up an invalid constraint should respond to OK (with the button highlighted) as well as escape.

Longer term maybe it would be better to eliminate modal dialog boxes completely, but rather refuse to perform the operation and provide visual feedback that lasts a few seconds.

whitequark, 2016-10-12:
This doesn't appear to be possible; focusing the button explicitly lets me dismiss it with spacebar but not Return. How about switching the key equivalent from Escape to Return? Doesn't seem like I can do much here without reimplementing half of NSAlert.

whitequark, 2016-10-12:
Fixed in master and 2.x as described above.

closed 2016-10-12 by whitequark