solvespace/solvespace #68: Support trackpad gestures

d235j, 2016-10-11:
macOS has scroll and zoom gestures which are essential for usability on a trackpad (as right-dragging is tricky and there's no center button). This should be possible on Windows and Linux as well, at least on computers that have multitouch trackpads.

Wishlist: support new Apple/Mac pressure-sensitive Force Touch trackpad.

whitequark, 2016-10-11:
OS X docs: https://developer.apple.com/library/content/documentation/Cocoa/Conceptual/EventOverview/HandlingTouchEvents/HandlingTouchEvents.html
Windows docs: https://blogs.msdn.microsoft.com/jennifer/2010/12/14/multitouch-part-2-support-for-gestures-in-windows-7/
GTK docs: lol

The Windows API is more or less the same as OS X one.

tpimh, 2017-05-13:
I think this is the GTK+ doc that you are missing: https://developer.gnome.org/gtk3/stable/GtkGesture.html

whitequark, 2017-05-13:
@tpimh It is a really bad doc.

ss0git, 2017-07-14:
Since my issue was closed I'll copy my suggestion here. It is more specific than what is proposed in this support ticket.

UX improvement - Switch to using Shift + "Two finger drag" (no click) on trackpads to Orbit around

Proposed change to switch from Shift + "Two finger click and drag" to Shift + "Two finger drag" (no click) on trackpads to Orbit around.

### System information

SolveSpace version: 2.3

Operating system: OS X Sierra

### Expected behavior

Orbiting around using Shift + "Two fingers drag on trackpad" (no click) should work better. Pushing down to click is very cumbersome. You want to spin around the model like spinning an earth globe in school, effortless.

### Actual behavior

Shift + "Two fingers click and drag on trackpad" orbits around but not pushing down hard enough to click causes the app to Zoom.

To better see the change i made this table:

Current behaviour

_ | Pan | Zoom | Orbit
------------ | ------------- | ------------- | -------------
Flick fingers | _ | X  | _
Push down + Flick fingers | X | _  | X
Shift | _ | _  | X

Proposed behaviour

_ | Pan | Zoom | Orbit
------------ | ------------- | ------------- | -------------
Flick fingers | _ | X  | X
Push down + Flick fingers | X | _  | _
Shift | _ | _  | X

### Possible incompatibility with 

"Two fingers drag on trackpad" (mainly up and down) are used for Zooming (scrolling) now which is fine, but it might lead to a possible incompatibility.

To fix this we could use Shift as a separator. "Flick your fingers" to Zoom(scroll) (current functionality) and press Shift +  "Flick your fingers" to Orbit.

### Testing

I ran a couple of tests, for other reasons I have a setup that can produce or reverse the effect of any gestures and clicks, and the user experience will improve sensibly.

whitequark, 2017-07-14:
I would actually prefer use 2-finger "pinch" gesture for zoom, 3-finger "pan" gesture for translation, and 1-finger double-click plus drag gesture for rotation. Objections?

ss0git, 2017-07-15:
I'll do some testing here, with your setup, to see how it feels. My proposition was made with "mess with as little as possible" in mind. I tried using the app to model something as part of a real world project and it proved quite cumbersome to navigate after a while. Most projects are 75% moving around to get a better view.

d235j, 2017-07-15:
@whitequark why not stick with the Apple standards:
2-finger "pinch" for zoom
2-finger "pan" for translation
2-finger "rotate" for rotation
and some way to change axes.

whitequark, 2017-07-15:
@d235g Sure that works

ss0git, 2017-07-15:
There might be some use getting this down in some menu for user preferences or something. Have each user select the one they like.

That table I build for Pan, Zoom, Orbit could be pivoted and used for this purpose.

I'm a long time CAD user and there are many possibilities depending on what you want to achieve. Some people transition from other software packages, others appreciate uniformity with os but productivity is the main factor, spend 2-3 hours daily in a cad package and "ease of use" won't keep you company for long.

A user preferences menu would be great for everyone!

whitequark, 2017-07-15:
So far the majority of people have been satisfied with the default bindings. I think I'll first implement the OS-native gestures, and see how it goes from there.

damian0815, 2017-10-03:
As a Mac user I'm not at all a fan on the pinch to zoom gesture, it always feels awkward.

My preference would be for the following:

"Two finger flick" (system scroll wheel gesture) = zoom (as it is now)
Cmd-drag = pan
Alt-drag = orbit

d235j, 2017-10-03:
@damian0815 @whitequark as it seems some people prefer OS-native gestures and others prefer gestures/keyboard shortcuts that are in line with CAD software on other platforms, this will probably end up having to be a preference option.

ruevs, 2022-01-24:
https://github.com/solvespace/solvespace/pull/1107
https://github.com/solvespace/solvespace/pull/1093
https://github.com/solvespace/solvespace/pull/825

@tomsci @vespakoen @phkahler @d235j considering the above changes, what do you think, can we close this? I do not have a real Mac with a trackpad to test on.

vespakoen, 2022-01-24:
Trackpad support kindof broke in with the last changes done to support the magic mouse, I want to take a look at this for a while already but was busy with other stuff.

This issue should stay open for now at least

Thovthe, 2022-04-15:
You don't have to rely on gnome docs for touchpads anymore.  libinput [to the rescue](https://wayland.freedesktop.org/libinput/doc/latest/gestures.html)!

Also the [maintainer](http://who-t.blogspot.com/2021/09/whats-new-in-xi-24-touchpad-gestures.html) is an entertaining speaker (/watch?v=HllUoT_WE7Y).

ruevs, 2022-07-05:
Hey guys should we close this? @tomsci @vespakoen @phkahler ?
https://github.com/solvespace/solvespace/pull/1093

phkahler, 2022-07-05:
Last I checked on my old MB Air... in version 2.3 you could rotate with 2-finger drag and no click or keypress. Now you need a modifier key. So IMHO a slight step backward.

Nowhere near as bad as having to click-drag though.