solvespace/solvespace #70: Orientation arrows in lower left corner would be more useful as part of a viewcube

d235j, 2016-10-11:
The orientation arrows in the lower left corner currently seem to only be useful to provide an indication of the view. It would be nice if they could be extended to make them usable for navigation, as part of a viewcube allowing dragging around and selection of faces/edges.

whitequark, 2016-10-11:
@jwesthues thoughts on this?

jwesthues, 2016-10-12:
I guess @d235j doesn't have a center mouse button? I could see the utility if not, and there's some vague analogy with e.g. dragging imported part normals to rotate the part.

Symbian9, 2016-10-12:
Think, @d235j expect something like Blender manipulator, yes?
[image]

* https://www.blender.org/manual/ru/editors/3dview/transform/transform_control/manipulators.html

d235j, 2016-10-12:
Yeah, exactly like the Blender manipulator. There are a couple of reasons for this, one being ease of use on devices without a center mouse button, but more importantly it would make it much easier to jump to a specific view without having to hunt for menu options.

Symbian9, 2016-10-12:
@d235j, I only can't understand what you mean in this:
> and selection of faces/edges.

Can you give some pictures?

whitequark, 2016-10-12:
> I guess @d235j doesn't have a center mouse button?

Center *and* right mouse button; I have a laptop with only left and right buttons, and Shift+RMB is fairly usable (though awkward at first), but I imagine it's really painful on an Apple trackpad. Gestures (https://github.com/solvespace/solvespace/issues/68) would also help here.

Evil-Spirit, 2016-10-12:
Viewcube is for selecing projections/views. We click on vertices/edges/faces to align view.
[image]

jwesthues, 2016-10-12:
The x-y-z normals in the corner can also be constrained against, so the viewcube alone would lose functionality. I guess you could do a modified viewcube that also kept that. For iso/ortho views, it's faster than the menus, but the keyboard shortcuts are faster still. SolveSpace's interface would require considerable changes beyond the viewcube to be well-suited to touchscreen devices.

> Center and right mouse button; I have a laptop with only left and right buttons, and Shift+RMB is fairly usable (though awkward at first)

I guess we could add more keyboard modifiers, to permit all three operations (select, translate, rotate) with left.

d235j, 2016-10-12:
I think a modified viewcube would work well.
Keyboard shortcuts, while great for an experienced user, mean an increased learning curve and are not very intuitive for someone starting out.

Symbian9, 2016-10-12:
> I guess we could add more keyboard modifiers, to permit all three operations (select, translate, rotate) with left.

@jwesthues, it would be better make fully configurated keyboard shortcuts that any user can set by himself or set default.