solvespace/solvespace #71: Update documentation to mention macOS keybindings

whitequark, 2016-10-12:
E.g.:
  * mention the use of the cloverleaf key instead of Control
  * the key that's Backspace on PC keyboards acts as Delete since there's no separate Delete key

virtualritz, 2016-10-14:
- The cloverleaf key is called 'Command' on the Mac. If you replace 'Control' with 'C' in the docs, this is solved. I.e. if you write press 'C-V' to paste this is usually understood by Linux/Win and Mac users alike as to mean 'Control-V' on Linux/Win and 'Command-V' on OS X.

- The backspace key on PC keyboards is actually labeled 'delete' on a Mac keyboard so neither docs nor menus need to be changed for that platform.

Regarding docs: I ported the entire reference docs to [Sphinx](http://www.sphinx-doc.org/).

I've been working on this the entire week in my spare time (evenings). It's almost ready. I have a Sphinx CMake wrapper in place so docs are spit out as part of a regular build in HTML and ePub format (or explicitly via `make htmldoc` or `make epubdoc` targets). For the HTML version I used the theme from [ReadTheDocs](http://www.readthedocs.com/).

I think once this is on github, we can also auto-generate/auto-publish docs from this @ ReadTheDocs.

Of course it is also easy to generate PDF or a myriad of other formats via Sphinx (through the LaTex route).

virtualritz, 2016-10-14:
Regarding docs: I ported the entire reference docs to [Sphinx](http://www.sphinx-doc.org/).

I've been working on this the entire week in my spare time (evenings). It's almost ready. I have a Sphinx CMake wrapper in place so docs are spit out as part of a regular build in HTML and ePub format (or explicitly via `make htmldoc` or `make epubdoc` targets). For the HTML version I used the theme from [ReadTheDocs](http://www.readthedocs.com/).

I think once this is on github, we can also auto-generate/auto-publish docs from this @ ReadTheDocs.

Of course it is also easy to generate PDF or a myriad of other formats via Sphinx (through the LaTex route).

P.S.: In my version of the docs 'Control' is already replaced by 'C'.

whitequark, 2016-10-14:
@virtualritz You do know that the website is already on GitHub? https://github.com/solvespace/website

virtualritz, 2016-10-15:
@whitequark: no, I didn't.
The docs can be on github raw, as ReST. I.e. ReadTheDocs will generate HTML documentation automagically from the repo.