solvespace/solvespace #76: Chirality-preserving constraints

whitequark, 2016-10-14:
For many constraints, to ensure the sketch doesn't flip-flop between multiple stable states (e.g. pentagon vs five-pointed star), we need such constraints to preserve chirality. We will also need to infer chirality from 2.x sketches when loading old files.

The list of constraints is:
* PROJ_PT_DISTANCE
* PT_LINE_DISTANCE
* PT_PLANE_DISTANCE
* PT_FACE_DISTANCE
* SYMMETRIC
* SYMMETRIC_HORIZ
* SYMMETRIC_VERT
* SYMMETRIC_LINE
* HORIZONTAL
* VERTICAL
* PERPENDICULAR
* ANGLE
* ARC_LINE_TANGENT
* CUBIC_LINE_TANGENT
* CURVE_CURVE_TANGENT
* PARALLEL

Depends on #75 as I would like to avoid adding new options to the old file format.

ruevs, 2020-10-23:
Duplicates:
https://github.com/solvespace/solvespace/issues/440
https://github.com/solvespace/solvespace/issues/676
https://github.com/solvespace/solvespace/issues/770

ruevs, 2023-04-05:
For cross reference https://solvespace.com/forum.pl?action=viewthread&parent=211