solvespace/solvespace #77: Parametric sketches

whitequark, 2016-10-14:
There should be a way, through the text window, to input a table of parameters. These parameters could then be used in expressions around the sketch, and so the entire sketch could be reconfigured by changing a single parameter value.

Depends on #75 (since we need to save this table), and #76 (since if changing parameters would break chirality, it's useless).

MJLHThomassen, 2017-11-08:
What is the current status of this? This is basically the only feature I'm currently missing.

whitequark, 2017-11-08:
This is a major feature that depends on updating the savefile format, unfortunately.

Evil-Spirit, 2018-02-25:
@whitequark 
I've made a prototype https://youtu.be/r85jSktKrWw

whitequark, 2018-02-26:
The hard part here is changing the savefile format and adding the UI, unfortunately.

Evil-Spirit, 2018-02-26:
@whitequark no problem here. 
1. just save one more textfield.
2. just another text screen with list of parameters and ability to change names.

Symbian9, 2018-02-28:
> The hard part here is changing the savefile format and adding the UI

1. One more text field is good solution; For export list of variables/parameters use CSV format.

2. Current line styles UI could be used as template for variables/parameters UI.

whitequark, 2018-02-28:
@Symbian9 Anyone can say "could be used as a template". How about writing the code for it? I'm not interested in your opinions, I have mine.

baryluk, 2020-01-16:
Is the intention to only support explicit constraints of the form `x = express/function of other variables`, or implicit constraints could also be in the scope of this bug.

Something like this:

[image]

where the constraint definition (and its text) is not attached directly to any entity, and there is no way to solve for one specific variable. Should be possible to incorporate this into existing solver using Lagrange multipliers maybe?

baryluk, 2020-01-16:
@Evil-Spirit That is a very cool and impressive demo. Had you had chance to do more work on it recently?

Evil-Spirit, 2020-01-23:
@baryluk Hello, unfortunately, I have no time for doing that. But if someone wants integrate this, I can help in advice. Ofc we can intoduce such kind of constraint like arbitrary equation, so It will be awesome! At the moment I need such thing inside my CAD project NoteCAD.

Evil-Spirit, 2020-01-27:
Done! So if someone wanted such thing in SolveSpace, I can point where and how it can be implemented. 
[image]

phkahler, 2020-06-13:
@whitequark what if we just use @Evil-Spirit implementation which seems to identify variables in expressions and create a parameter for them. Then add a new constraint type "equation" where you can type things like:   x=5   or   a*a + b*b = c*c    the equation constraints would need a text string that would need to be saved in the file, much like the parameters would need name saved.

The equation constraints could be placed at the top of the constraint list in the text window where they could be edited. This would be a bit limited, but fairly simple and useful.

whitequark, 2020-06-13:
Let's discuss this once 3.0 is out. I don't think we need any more features to delay the release.

phkahler, 2020-06-13:
Oh I completely agree with that!

phkahler, 2020-09-22:
@Evil-Spirit  While there is a lot to do for this to be a feature, I took some time to rebase it on master before it bit-rots too much. You can try it on my "parameters" branch here:  https://github.com/phkahler/solvespace/tree/parameters

Commit: https://github.com/phkahler/solvespace/commit/d6f9730a33906a56371f2d0df232529707b84b1c
or Patch: https://github.com/phkahler/solvespace/commit/d6f9730a33906a56371f2d0df232529707b84b1c.patch

I saw the video months ago and wanted to try it myself. Very nice!

Evil-Spirit, 2020-09-22:
It seems, it needs to introduce one more list to represent parameters in browser.

ruevs, 2020-09-23:
@phkahler two links to be aware of:
https://github.com/ryannining/solvespace
https://youtu.be/UPgOUa1Q2Ns

Sil68, 2021-10-05:
> @Evil-Spirit While there is a lot to do for this to be a feature, I took some time to rebase it on master before it bit-rots too much. You can try it on my "parameters" branch here: https://github.com/phkahler/solvespace/tree/parameters
> 
> Commit: [phkahler@d6f9730](https://github.com/phkahler/solvespace/commit/d6f9730a33906a56371f2d0df232529707b84b1c) or Patch: https://github.com/phkahler/solvespace/commit/d6f9730a33906a56371f2d0df232529707b84b1c.patch
> 
> I saw the video months ago and wanted to try it myself. Very nice!

Sound really compelling! I am absolutely looking forward to this feature! Currently I am using FreeCAD and it is feeling clunky & sluggish. How I have experienced SolveSpace so far, I am almost ready to migrate!

Using named parameters is one of my relevant requirements, and I really would like to try out this patch of yours. I've already successfully built your branch on my Mac, but now I'm struggling to find the option of defining a named parameter.

drozdziak1, 2022-03-28:
Hey folks, I am almost ready to convert to SolveSpace fully, if not for the variables/expressions feature. What could be the remaining blockers for this feature?

phkahler, 2022-03-28:
> What could be the remaining blockers for this feature?

The demo that Evil-Spirit did is not a complete implementation. It is very interesting and his changes would probably be the starting point for this feature. I have some questions about his changes which I can get answers to by testing it more, but then the there are some things that have to be worked out before adding it to an official Solvespace release. First, we'd need a good way to define constant parameters like "width = 5". Then there is the issue of scope - defining when a variable (named parameter) can be used. Only in the same group it's defined? Only in that and subsequent groups? What if a later group also defines a value for the same name? It really goes along with the whole hierarchical sketches thing, which is really a long way off if ever.

If we had a way to define constants, and the scope was limited to just one group, I might be OK with adding it without a solution to the scope issues since any future change should be compatible.

Thovthe, 2022-04-15:
I like the idea that it propagate to child groups but there should be a top level where global variables can be defined.  Maybe something like a spreadsheet.

phkahler, 2022-04-15:
> I like the idea that it propagate to child groups but there should be a top level where global variables can be defined.

yeah, that's why it's closely related to issues #78 and #80.

HarryPehkonen, 2022-08-08:
Hi!

My apologies if this has already been suggested.  I went through comments in this issue, but didn't see it mentioned.

One *possible* approach is to have a global sketch (the first sketch) available where you can specify values.  Naming / labelling might be an add-on feature eventually also.  This is not a "spreadsheet" solution that everybody wants, but it might be enough.

Here is how you could use it:

First, you create this special sketch, and you add examples of all the line segments (construction lines so there's no panic with non-closed loops?) that you may want, and their dimensions.  You can specify lengths, radii, and angles (whatever works with the Q shortcut key).  Next, whenever you are working on another sketch, select something (a line segment, for example), go to the special sketch, select the appropriate line segment (or whatever), and hit the Q key to specify equivalency (normal shortcut key -- nothing unusual here).

You can almost do this already.  In the first sketch, just add some construction lines with their dimensions.  You can also use the "Sketch curves from text" tool to add labels.  What would be nice is if you could now create a new sketch on the same workplane (or whatever) that is actually your first real sketch.  When you are showing off your finished design, just un-check the visibility of that first sketch (with the dimensions).

I realize this is a method that can very quickly get out of hand  :(

ruevs, 2022-08-09:
> First, you create this special sketch, and you add examples of all the line segments (construction lines so there's no panic with non-closed loops?) that you may want, and their dimensions.

Your suggestion is perfectly reasonable and most experienced users already do this in order to parametrise their designs with what SolveSpace can already do.

The real topic of this issue is to allow named constraints and saved equations between them ([equations when entering dimensions are already supported but they support only numbers and are evaluated on the spot and discarded](https://solvespace.com/ref.pl#Dimension)).

For example (imagining assembling a laser cut acrylic box):
'thickness = 10'
'chamfer = thickness / 4'
'joint_tolerance = 0.2'
'joint_width = thickness + joint_tolerance'
'joint_length = 2 * joint_width'
etc. Imajine these equations entered for dimensions and remembered - so that later on when you edit the constraint they are displayed. Of course everything is dynamically recalculated on the fly - just like everything else already is.

uberintj, 2022-12-23:
As I mentioned my intention in the forum this summer (https://solvespace.com/forum.pl?action=viewthread&parent=4637), I have started tackling this problem here: https://github.com/uberintj/solvespace-vars/tree/unconstrained-vars 

That branch is a simple merge of the Evil-Spirit prototype with the current (DEC 2022) master.

I have replicated (closely) his demo: https://projects.absolutepowerandcontrol.com/attachments/download/43/slvs-unconstrained-vars-demo.mp4 (8:30 min, about 18 MB).

I did this to get myself up to speed with gitub, get a working environment for editing, compiling, and debugging the SolveSpace source tree (get back into C++ after many (many) years, in Linux no less), and prove that those changes were still compatible with the current master.  They seem to be.
 
I'm next looking to understand and selectively integrate the changes from here: https://github.com/ryannining/solvespace/commit/8fa6818092b46789a4b68299315a0d5a710e441e (comment based variables).  I think his fork has several commits related to this.

At that point I'll be much more comfortable with things and start finalizing this as a feature (add needed UI, save files, control precedence between linked models, etc).  I anticipate this will take me all winter, maybe longer, due to other responsibilities, but I fully intend to work it through to some meaningful completed state (ie: not just a prototype).  In the mean time, I have some questions about previous changes that were made, though not completed, and are likely to affect this effort.  Where is the best place to discuss those things?

ruevs, 2022-12-23:
@uberintj I'm glad to see you came back to this. This issue is the best place to discuss things. You can @ mention specific people if you want to attract their attention.

uberintj, 2023-01-05:
I'm not sure about how I feel about the details of the actual implementation yet.  After all, it WAS a prototype, and I truly only merged in Evil-Spirit's changes.  I didn't alter them at all.  But I do like the results so far.  I do agree in general with the comment about string comparisons.

Regarding reusing the _comment_ field and having a specific value in the Type enum for variable expressions, I think that's more in keeping with the current design, though 'comment' seems like a poor name for that value (I understand these things happen as concepts evolve over time).  I think maybe the other attempt to work this out by https://github.com/ryannining/solvespace/commit/8fa6818092b46789a4b68299315a0d5a710e441e did something like that, and it's one of the things I'll be looking at - merging those two things together. 

One of the questions I have is related to another change made several years ago in this commit (https://github.com/uberintj/solvespace-vars/commit/61f248899a4f3c4ce06e63759669fe6c3e727417) wherein the Expr::Op::VARIABLE was added, but not completely implemented.  This comes into play when ExpParser::Lex encounters an upper case input (create a constraint using an upper case letter) and it sets the token up to have the Expr::Op value of VARIABLE.  Then, in other parts of the code (eg: expr.cpp:342) there are assertions that fail based on the Op type, resulting in the app aborting. 

This feels like an early attempt to add variable constraints that was really only 'stubbed in'.  At any rate, it **seems like** the two concepts are attempting to address the same issue in different ways.  Or maybe I'm misreading the intent and just ran into some dead-end code that interferes with these changes when coincidentally using upper case letters.  Any ideas there?

FWIW, I haven't looked at this in depth yet, but intend to shortly.  I'm currently in the middle of finalizing a major release at work, which will most likely drag through the end of the month, so until then my time is limited.  But expect some movement and possibly more questions about it after that, maybe sooner if the end of this release goes very well (not likely, LOL).

caseybasichis, 2023-01-20:
Are Evil-Spirits changes reachable through the python interface?

On a project that needs constraints within a range.

Maybe without all the ux. file structure etc his design is ok to work with as is?

uberintj, 2023-01-31:
@caseybasichis

Sorry, I'm very new to this code base and not familiar with what is or isn't provided by the python bindings, or even how they are exposed.  Maybe @https://github.com/phkahler would know.

---- On Fri, 20 Jan 2023 18:16:47 -0500 Casey Basichis ***@***.***> wrote ---

Are Evil-Spirits changes reachable through the python interface?

On a project that needs constraints within a range.

Maybe without all the ux. file structure etc his design is ok to work with as is?

phkahler, 2023-01-31:
@caseybasichis  I don't know how much is behind the solver interface. Since a lot of those changes were to the expression parser and the algebraic equation solver, I would think most of his changes could be included in the solver library. That doesn't mean those features would be accessible via python. Either way I think we need to wait for acceptance into the Solvespace master branch before any of that becomes relevant.

caseybasichis, 2023-01-31:
Thanks for that info.

To clarify: a Python binding would be unlikely/difficult or is it that it won't inherently show up once those features are merged in?

The immediate need is for <= >= on distance and angle.
Probably a big bite, trying to hack that constraint into a fork of the main code base?

phkahler, 2023-08-03:
I've merged the old Evil-Sprit changes into a "parameters" branch on both my fork and the main repo here.
Problems I see immediately:

1) Make a rectangle, make the width "width" then extrude and make the extrusion length "width". There appears to be a single parameter (not sure, does "save" work?) but you can change it in either group by dragging. Dragging the extrusion length does not update the previous sketch, but dragging the underlying sketch width updates both. Ideally there should be one "width" parameter in the first group and it should be held constant in the later group.

2) Create two parameters "width" and another "Width". It will crash when you press enter.

3) I don't think file I/O is supported yet.

Someone else had done additional work based on these patches. I'd like to round it all up and complete the feature for version 3.3, including comment-constraints or similar to write "width = 5" or similar on the drawing.

phkahler, 2023-08-09:
@jwesthues  I was going to try to fix the above scope issue by treating named parameters from previous groups as fixed during solving, but I'm not sure how to do that.  In expr.cpp it's not clear what list of parameters is even being used. I thought there was a global parameter list, or is it per group? The Parameter class doesn't have any reference to a group. I figured when writing the equations for solving I'd convert parameters from previous groups to a constant but not sure how to tell what group they belong to. If there is a list of parameters per group, I don't understand how names can conflict (see above comment and the "parameters" branch).

phkahler, 2023-08-14:
@jwesthues  The old implementation creates named parameters at the time a dimension is evaluated. I'm not sure when those are evaluated. I also thought those got evaluated to a constant, but in this case expressions are saved and displayed, and can be edited. Any text string in the expression that is not a built-in and not already a parameter name is used to create a new named parameter. So questions about that - not necessarily for @jwesthues 

1) When in a dimension expression evaluated?  (does editiing one cause a regeneration?)
2) How is it storing the string in the constraint?
3) Do we have the group info at the time of evaluation to get it in the named parameter handle?

I wanted to either create a new constraint (equation) or just allow equations in the existing comments. Probably a new constraint type. This way we can just type "X=5" as a constraint equation on the diagram. Or even general algebraic expressions with more than one variable. Suddenly SS becomes an algebra tool as a side effect. This should require very little new code.

@uberintj  if you still want to tackle this, it's all right here right now ;-)  It seems like a good intro to some core parts of the code.

uberintj, 2023-08-14:
@phkahler, I'm trying to get back into the swing of things. Got knocked out of my normal orbit early this spring and am just now getting back to being able to handle my normal work load.  Things slow down in the winter, so that's the most likely time I'll be able to dedicate.  

But I'll chime in here anyway for the things I can answer.

1. I'm pretty sure when a dimension is edited the model is regenerated.
2. A new member is added to the Constraint.  It is called 'expression'.  See ConstraintBase in sketch.h
3. I'm not sure about that

I agree with creating a new constraint type to specify a parameter's value (my terminology is currently 'unconstrained variable'), because this lets us give definitive constraint.  I think we need some precedence rules for these, so an assembly being linked into another can have its parameter(s) overridden to make them apply to the linking assembly (I guess this can be generalized down to the GROUPS that source the parameters since linked assemblies end up in groups?).  Anyway, this is what I was digging on when I got derailed.

And other one...A couple weeks ago you reported 
> Create two parameters "width" and another "Width". It will crash when you press enter." 

Back [here](https://github.com/solvespace/solvespace/issues/77#issuecomment-1372498868) I wrote:

> ...This comes into play when ExpParser::Lex encounters an upper case input (create a constraint using an upper case letter)

With that issue I think you're running into that incomplete VARIABLE op type 'feature'.  I'm still not sure what that is all about.

At any rate, I still have this on my radar, but can't make any promises as to time frames.  My feelings won't be hurt if someone beats me to it, but it **will** be in the back of my mind whether I can actually work on it or not.

phkahler, 2023-08-15:
> With that issue I think you're running into that incomplete VARIABLE op type 'feature'. I'm still not sure what that is all about.

Yes. I tried removing that and it fixed the upper-case issue, but broke the test suite (wouldn't build) so I modified the test to look for the new Op::PARAM instead. Then it built but failed the test. Didn't debug it but that is the issue with that. It seems VARIABLE was an idea to implement a similar feature using some other type of "variable" that never got implemented. Evil-Spirits approach of just allowing existing parameters to have names seems like it may be better, and the ability to implicitly create them seems kind of nice.

IMHO using a new constraint type Constraint->Equation is the way to go. You can verify the text is an equation by finding the "=" and then get two expressions from either side of that. Then take their difference as a constraint expression. Constraints down in the solver are just expressions that are implied to equal zero.

I'd rather leave it to others so I can get back to geometry and NURBS stuff. but I have also been rather busy.

ruevs, 2023-08-15:
> IMHO using a new constraint type Constraint->Equation is the way to go.

In my opinion the parametric sketches ahouls introduce NO (or close to none) user interface.

Here is a short description of the idea I've formed in my mind of how this should work over the years.

1. In any place, where one can currently type a constant expression (and quoted with ` in comments) allow the user to type any valid equation, that has a single valid variable name to the right of the `=` or it does not contain `=`. If the expression is constant evaluate it immediately and store only the result - just like currently. If the equation contains variables - "store" it for future editing. E.g. `y=3*sin(angle)+x`, `5*thickness` etc.
2. As a particular case of the above - any vali variable name (in e.g. `C`) is added as and "unkown" to the system of equations. E.g. `X`, `thickness' etc.
3. All unknown (unconstrained) variables are displayed as is, e.g. `X`.
4. All equations that have no `=` but are fully solvable (constrained) are displayed with the calculated value and "eq" e.g. `5 eq`. If they are not constrained: `? eq`.
5. Equations that have an = are displayed as `X=7 eq` or `X=? eq`
6. Single variables that are constrained are displayed as `X=6.35"

That is (roughly) the UI - in other words almost no new UI - just equations that work. Of course we will show some "good stuff" in the text window when an constraint that is an equation is selected.

As for storing this stuff I was thinking of introducing an entirely new "structure" to the file format - let's call it "user equations list" - that holds all equation in a file. Then in the constraints where the equations are written by the user store a handle pointing to the equation in the list. But this idea for stoig them is fron 4+ years ago and it was fuzzy at the time ;-)

phkahler, 2023-08-16:
> I think we need some precedence rules for these, so an assembly being linked into another can have its parameter(s) overridden to make them apply to the linking assembly

@uberintj  Not at this time. That requires #78 (self contained sketches) and #80 (Hierarchical sketches). Currently linked sketches are static with no regeneration of groups happening within them, so there is no way to modify any part of them. What I'm hoping for short term is to get the named parameter stuff merged in a way that is useful within the current limitations. I would like equations that create named parameters to do so when linked into another sketch. This would allow a simple sketch containing definitions "N=5", "width=15" to be linked into a sketch, and you could link different version of it with different values and these would be used in later groups of the present sketch. I other words you could link in a parameter table and swap it out later. 

Fully hierarchical sketches is probably going to break file compatibility anyway, so I'd like to put it off as long as possible. We can build a lot towards it prior to then while maintaining compatibility and adding new features. Having named parameters implemented in some form should make less work at the time of change, and will allow people to experiment and figure out what it would ideally look like.

@ruevs  I disagree with part of your proposal.  I don't think it's appropriate to type an equation into a dimension. An expression is fine (just like now), but an equation doesn't make sense. Also, as soon as we allow named parameters in those expressions, they have to be re-evaluated during group regeneration/solving. So the expression has to be stored rather than evaluated and saved as a single value. There is some question about displaying the value or expression on the sketch. The current patch displays the expression and the old video shows adding an additional reference dimension to display the value. I'm not a fan of that, but it seems like a detail to work out after a more complete implementation is done.

Adding equations in constraints could be done by looking for valid equations in comment constraints but I think it may be better to have an actual equation constraint type that is largely like a comment. This is a very minimal UI change. The existing patch has essentially zero new UI while adding a good bit of capability - but not quite enough.

uberintj, 2023-08-22:
@phkahler That explanation makes sense and gives me a better idea of the scope of changes that fit in with the history of this project.  I was kind of conflating all of those things into this ticket to get a full parametric sketches feature complete.  I'm good with the incremental steps required to get there.  And I *did* mean that the linking file should be able to override parameters of incoming linked files - to me, this would be the epitome of a parameterized design where it's defined in the place where it is used.

I also agree about equations being a different beast and making a specific constraint type and UI facilities for them, not that my experience here bears much weight.  Just from a high level, it seems to make more sense to have the assignment of parameter values be a specific 'thing' with it's own use cases and flow, however simplistic those requirements may be.

phkahler, 2023-08-22:
> And I _did_ mean that the linking file should be able to override parameters of incoming linked files - to me, this would be the epitome of a parameterized design where it's defined in the place where it is used.

@uberintj   That's a tricky one. I don't think overriding existing parameters is a great idea. I think the linked sketch should have it's own "local variables" if there is a name conflict. If you've ever used Simulink (block diagrams) I like their concept of a "mask" which can be used to define parameters on a per-instance basis. So maybe a sketch contains a parameter table with defaults, and when linked you can override those with different values per instance. These would be stored in the parent sketch. This gets back to the idea of a parameter table of some sort rather than just named parameters in the solver.

If we do go with some kind of parameter table, would there still be interest in the idea of general algebraic equations typed on the sketch? I still think that's also a cool interim way to define fixed values ;-)   A parameter table probably belongs in the text window and that makes more sense when it's per instance for linked sketches later on.

uberintj, 2023-08-22:
@phkahler I think we may be having a vocabulary issue here, and it's probably my fault for not being more versed in this problem domain.  The per-instance basis solves the use case I mainly think of with a parameterized model, which is: I can define something, and reuse it in different contexts to get the most use out of that design.  Think nested end tables - they all look the same but are of progressively smaller dimensions.  Or even screws - they are mostly the same except for sizes and pitches.  Pulleys with different diameters and different sized holes for the shafting.  Things like that.  So your idea of a parameter table puts words to what I envisioned at least.  A place in the text window where you could define default values for them and some way to specify what those values should be when you reuse the model.

As for the expressions/equations in the model I think that's too powerful to let go.  It seems like removing them would require specifying a distinct parameter for each of those constraints instead.

dgramop, 2023-12-07:
@phkahler I'm currently working in the parameters branch. When I was writing some code to let users edit parameterized constraints (with their original expressions, instead of by-computed-value), I ran into a problem: units.

The way I see it, we have a few options:
Suggested: **Tack-on a scale factor **
Let c_1 ... c_n be constants and p_1 ... p_n be parameters for an expression inside of a constraint...
User, in mm mode, types in:
c_1 * p_1 + c_2 + p_2

In inches mode, the formula becomes:
(c_1 * p_1 + c_2 + p_2)/25.4

Benefits:
- Low-complexity
- User-predictable behavior

Drawback:
- Files will lots of parametric expressions will just have scale factors tacked on everywhere if a user looks at the sketch in different units

Alternative: **Dimensional Analysis**
Use constraints to perform dimensional analysis on the parameters, to guess which constants/variables need to be rescaled (based on the units selected by the user). 

Example: Suppose a user, in mm mode, types in:
Let c_1 ... c_n be constants and p_1 ... p_n be parameters for an expression inside of a constraint...
c_1 * p_1 + c_2 + p_2
Since [c_1 * p_1 + c_2 + p_2] = mm,  we need to only scale p_1, p_2, and c_2 by the appropriate unit conversion, and can call c_1 unitless. 

User in inches mode sees c_1 * p_2 + c_2 + p_2, except the parameters p_1 and p_2's current values are divided by 25.4, and c_2 is divided by 25.4

Possible edge case/confusing behavior:
Now suppose the user does something like this:
p_1*p_2

Which parameter, p_1 or p_2, is expressed in mm^2, and which parameter in 1/mm? If we had more constraints, we could probably stick all the constraints into a matrix and do the dimensional analysis to figure out what's what. Should one of the parameters be dimensionless and the other parameter be rescaled? 

Benefits:
- Will meaningfully convert formulas when changing units

Drawbacks:
- Possible non-user-predictable dimension guessing on the part of solvespace
- High-complexity

We could totally pick a certain convention and stick to it. I think the goal should be to do something that the user would consider predictable and correct. I'm worried that dimensional analysis may lead to the system making incorrect assumption and re-writing people's formulas when they convert from inches to mm in ways that they may not predict or approve of.

ruevs, 2023-12-08:
I for one, do not see a problem with units. SolveSpace is in mm internally - in RAM and in the .slvs files on disk.
The option:
[image]
only changes how the internal values in mm are converted for display.

We should just store all constants inside user entered formulas in mm and convert them back and forth for display, No more, no less. No lists of parameters needed.

ruevs, 2023-12-08:
> In inches mode, the formula becomes:
> (c_1 * p_1 + c_2 + p_2)/25.4

To be more specific what I propose is that in inches mode the formula becomes:

`(c_1/25.4) * p_1 + (c_2/25.4) + p_2`

where all expressions in the brackets are displayed as numbers - already calculated.

e.g. in mm:
`25.4 * p_1 + 50.8 + p_2`
in inches:
`1 * p_1 + 2 + p_2`

That's all.

phkahler, 2023-12-08:
> no lists of parameters needed.

@ruevs  Regardless of how we deal with units,  I'm very much in favor of adding names to solver parameters rather than having separate storage of "variables" as suggested by the existing stub and test case.

What if arbitrary constraint equations are allowed? How should units be handled for this:
A = 2
B = 1
X + Y = A
X * Y = B
Clearly X = 1 and Y = 1
But what happens if units are inches?
A = 50.8    <- these are converted behind the scenes.
B = 25.4
The equation X+Y = A  we expect to get X and Y both 25.4, but that will not satisfy the last equation.
I think we run into a problem where parameters need to know if they're just a number of if they are a dimension. My example fails because the last equation doesn't balance units. Another issue is people might use a variable for an angle, and that should not get unit converted at all. I'm afraid determining what to do automatically is similar to type inference in programming languages.

jwesthues, 2023-12-08:
> @jwesthues What do you think of 1) the parameters branch with the @Evil-Spirit changes and 2) The problem mentioned above about trying to deal with units in general expressions? Or should we settle for just having named parameters defined in the text window and do unit conversion there?

My first impulse might be to split the work into three stages:

1. Define a syntax to enter constants with units (`(3 in)*(2 in) + 4 in^2`), moving the problem of which constants should get the unit conversion to the user.
2. Apply dimensional analysis when an equation is first entered, and ask the user if they want to infer the current length units for any inconsistent operations (add length plus dimensionless, cos of length, etc.). Display the equation with the inferred units, so the user can see what happened and correct that if they don't like it.
3. Apply dimensional analysis before the complete system is solved, and warn on any inconsistency.

Each of those stages is useful in isolation, and I don't think implementing each next stage breaks backwards compatibility. I haven't thought about this much recently, though.

ruevs, 2023-12-08:
This is more complicated than I initially thought - I apologise for my last comment.

I was thinking only of units (mm, incjes etc.). But the problem comes from dimensionality mm², mm³ etc. and from degrees.

I will read all comments from the last few days and think about it carefully.

Like @phkahler, I still really like allowing "arbitrary algebra" but the dimensions need to be take into account somehow...

dgramop, 2023-12-09:
> @phkahler ... just take the expression and multiply the entire thing by the scale factor ... Would this cover all or most of the bases? 

This would cover all the bases, and have good predictable behavior, and add no burden to the user. We could do it this way and see if people complain. If they do, we can see what the users are clamoring for and cater more directly to that.

jwesthues, 2023-12-09:
> just take the expression and multiply the entire thing by the scale factor ... Would this cover all or most of the bases?

@phkahler I'm not sure I understand the desired behavior here. Like if I define `A = 2*3` and set a line length equal to A, initially in millimeter mode, then I pretty unambiguously want the line to be 6 mm long. But if I change to inches, what happens?

- Should the line length change? I'm assuming no, since that would be pretty weird.
- Should the displayed `2*3` change?
    - If it doesn't, then the behavior is "expressions are always in mm, regardless of units mode". I guess that's the default lazy option, initially confusing but simple.
    - If it does, then we don't know whether they meant `2*0.118` or `0.078*3` without some additional cue.

dgramop, 2023-12-09:
> @phkahler I'm not sure I understand the desired behavior here. Like if I define `A = 2*3` and set a line length equal to A, initially in millimeter mode, then I pretty unambiguously want the line to [...]

I think for this solution we could just tack on the scale factor without simplifying, unless we can find the exact inverse of the scale factor in the formula (in which case we simplify to 1). That way we don't have to pick what to scale (ambiguous behavior) and the user knows exactly what happened (recognizable constant multiplying their formula)

jwesthues, 2023-12-09:
@dgramop Okay, I think I get @phkahler's proposal now. Basically when we change display units, we multiply a scale factor onto the expression as displayed to and editable by the user, and then silently divide by the same scale factor internally when we evaluate it. So the geometry doesn't change, but the displayed quantity is in the right units. So in my example, we still show `A=2*3`, but the dimension now shows `A/25.4`, and gets evaluated to `(A/25.4)*25.4 = A`.

That's relatively low effort, and results in display in correct units. The displayed scale factors are ugly, but a user who always works in the same units will never see them. I don't think that breaks backwards compatibility for any of the fancier options later. So I don't love that, but probably prefer it to "always in mm".

ruevs, 2024-08-21:
> places where variables don't seem to be accepted yet. Like e.g. for the number of repeats in a linear array.

Ha!  This is an interesting thought. If we make the number of repeats in a step translating/rotating group an equation with parameters - like all other constraints - that may be a unique fearure among all CAD software.

Would it be? 

Of course we will have to round or truncate the number of repeats resulting from the equation.

dgramop, 2024-08-21:
We have the solver powering everything, let's go nuts! 

> Would it be?
I'm sure _somebody's_ cad supports variable input for repeats, but parameters are always a hacked-in afterthought for mainstream cad modelers. thanks to the nice underlying solver and all the other design decisions made long ago, i feel like solvespace readily takes these kinds of features. I doubt any other cad so richly integrates parameters with constraints... then again I mostly use FOSS cads and sparingly free versions of professional software.

>  Like e.g. for the number of repeats in a linear array
This would be a very useful feature. I for one would like to use it in many of my models. For situations where I need something that parametric, I must turn to OpenSCAD. I think round-down truncation is a sensible. We can also introduce explicit `ceil` and `floor` operators

@NothanUmber Do you have time to carry the torch for this excellent feature idea you suggest? I am unlikely to get around to it for a few months, with some other pressing projects using up all my cycles. I think it's a good entry point to the `parameters` branch and solvespace.

NothanUmber, 2024-08-26:
The variable pattern feature would indeed add wings to Solvespace!
But I think I would still need a rocket to get me to my current target, the moon. Or, let's say at least a very strong catapult that supports joint sketches (so the linear pattern can be part of a sketch that can then be e.g. extruded), And lofts. And optimally fillets (so I would not have to switch programs for that step).
But will keep an eye on Solvespace for other, more fitting projects!

ruevs, 2024-08-27:
> the linear pattern can be part of a sketch that can then be e.g. extruded

That "just works™". Here is an example with step rotating https://github.com/solvespace/solvespace/issues/841#issuecomment-742142047

Everybody wants chamfer/fillet and many people want loft ;-)

NothanUmber, 2024-08-27:
Thanks, looks promising!
Imho loft would be much more important than fillets. Because adding fillets is the last "sugar on the cake" step that might have to be redone anyways in case of bigger geometry changes. So doing this in another program isn't so bad.
But creating complex bodies by lofting between different sketches is often very hard to substitute.
Can often be tricky which points to connect though. Some programs fully rely on heuristics (with more or less desirable results) or manual point assignment (which can also lead to problems when the geometry changes a lot).
So it is a sound approach to have a CAD system that focusses on 100% "stable" features. Which works very reliable in the supported cases. And leaves the other tasks to the many heuristics based CADs.

I still like the CadQuery loft approach the most. It is very simple but also pretty predictable. As I understand it: Connect the first point in one sketch to the first point in the second and so on. So one has to specifically make these sketches for this purpose. But as all sketches are generated programmatically there it doesn't hurt.

phkahler, 2024-08-27:
>> I still like the CadQuery loft approach the most. It is very simple but also pretty predictable. As I understand it: Connect the first point in one sketch to the first point in the second and so on.

I've had this thought for Solvespace. When we extrude, copies of all the sketch entities are made at the end of the extrusion (2 copies for double sided) and these are simple transforms of the original sketch entities. It might be possible to make copies that are embedded in a plane but otherwise free to be moved and constrained (perhaps copy the constraints too but allow deletions?) the extrusion surfaces would still connect entity to entity - point to point - but the new ones would be movable and constrainable.

Of course then people will come along with all sorts of new options they want - can the new plane be non-parallel to the first one? Oh now we need to allow manipulation of the surface curvature between the planes. And on and on.

ruevs, 2024-08-27:
Hey, [loft has it's own issue](https://github.com/solvespace/solvespace/issues/439) - don't "scatter" the discussion ;-)

dgramop, 2024-09-06:
https://github.com/user-attachments/assets/974ba12a-0a6e-4b24-98b0-8fb3bfedda3f

@NothanUmber this may be of interest to you

just hacked together proof-of-concept. really an experiment for adding expression support to groups with inputs in the text window. was this approach/effect/result jive with what you were thinking?

NothanUmber, 2024-09-08:
Hi! Just to understand what you did - you added x+3 as a driven dimension constraint to the width of the rectangle and then used that "x" as expression to drive the amount of copies in the linear pattern? If yes then this is indeed one incredent of the things I had in mind!
I want to pattern two sketches on parallel construction planes, both N times. And then select both patterns and loft those the closed wires in those pair-wise (first wire-loop in pattern on one plane with first pattern on parallel plane etc, second wire-loop on plane one with second wire-loop on plane two etc.)
The distance between the pattern instances would be two separate parameters and the width and height of the base sketches could also be changes separately. This would give a highly parametric way to create a multitude of lofts that could then e.g. be used to carve non-parallel pipes with variyng thickness on different places into a block. (At the point where I wrote about the parametric patterns I didn't realize yet that solvespace has currently no lofts yet, which would be incredent no. 2 - but having this could also be useful in many other scenarios outside of my specific one of course!)
Thanks for your experiment, looks very promising!

[14 more replies]

closed 2026-11-05 by phkahler