solvespace/solvespace #78: Make Sketch self-contained

whitequark, 2016-10-14:
Currently, sketch generation requires a lot of global values, which means that two sketches cannot coexist in memory. This is a blocker for many other features, and should be refactored.

Thovthe, 2022-04-15:
Is this blocked on #75 or could it conceivably be done by someone with the time and inclination?  If so ~ how much time would this take?  And is it straightforward boring work or exciting but immensely difficult?

hinell, 2022-11-01:
@Thovthe There are tonnes of problems. I think it's best to start from sketch a new SolveSpace based on CAD formats like [STEP](https://en.wikipedia.org/wiki/ISO_10303-21). The current one is quite limited. It's also has weird modelling structure where the next entity depends on the previous one and relationships can't be defined manually.

phkahler, 2023-08-09:
> Is this blocked on #75 or could it conceivably be done by someone with the time and inclination? If so ~ how much time would this take? **And is it straightforward boring work or exciting but immensely difficult?**

No it's not blocked. In fact I'd say it's almost a requriement to change the internal data structures first. Some serious consideration needs to happen though so we're not passing pointers or references around. OTOH we have to reference the globals in a lot of places now. Refactoring code to reduce or eliminate globals should not affect functionality or file format though. But then a text window (as opposed to *the* text window) would need to know what sketch it's showing, same for a main drawing window. Just my thoughts.

DjogaRo, 2024-04-25:
I haven't seen any SolveSpace code, yet, so this might be a silly suggestion.
Declaring all those globals thread_local and having one editor thread per sketch?

phkahler, 2024-04-28:
> I haven't seen any SolveSpace code, yet, so this might be a silly suggestion. Declaring all those globals thread_local and having one editor thread per sketch?

We will still need to access multiple sketches from a single thread to do a hierarchy.

DjogaRo, 2024-04-28:
That's the part I cannot eyeball, not knowing the code. Such editor threads needed some interface. Depending on the code that's there this might be an easier fix (or not) - to add a keyword to those globals' declarations and implementing such interface - than tracking down all accesses to those globals to adjust them. Given that there are multiple branches being worked on, if I got that right.
If this could be easier, it would get this here TODO out of the way, until with reaching 4.0 everythings consolidated back into master and a more thorough approach would be more feasible.
I know, I have no idea what I'm talking about not knowing this codebase. Just throwing thoughts around.

timbod7, 2024-12-24:
How would one identify which globals need to be refactored to support multiple sketches? `SK` and `SS` [here](https://github.com/solvespace/solvespace/blob/30e42ca889b20b6af65d601524085fd812c91662/src/solvespace.cpp#L11) look immediately relevant.

Beyond that, running gdb `info variables` on a debug build, produces approximately 1500 lines of output, most which looks constant or not relevant to this issue.

phkahler, 2024-12-24:
@timbod7  I'm not sure the globals need to be eliminated, but they probably should. What's really needed is the ability to load a sketch, modify parameters, and "regenerate" it.  All of the functionality in generate.cpp is currently in the SolveSpaceUI class, but most of it could be moved into the sketch class. Things like     ScheduleGenerateAll();
 should be moved up to the caller, since they don't belong in the sketch.

Another option might be to create a generator object which does the generation to a sketch. I mention this because some parts of Solvespace are written in a functional style where things are not mutated, or at least don't modify themselves. The sketch is a very mutable thing though and having it self contained makes sense. OTOH Entities know about curves but I don't think sketches do. Groups each have a SShell and SMesh, but sketches don't care about those either. A little mapping might be insightful. The solver as a library AFAICT doesn't know about a lot of stuff.

The main goal is to allow regeneration of linked sketches (recursively), so we can change parameters per-instance. See #1436 for my thoughts on that. Editing does not need to be possible on any but the top-level sketch so it could remain global.

timbod7, 2024-12-26:
After taking a closer look I see that much of the existing code assumes that handles can be resolved to values as required, and this is done by resolving the handles in the existing Sketch global, called `SK`. A small example:

```
double EntityBase::CircleGetRadiusNum() const {
    if(type == Type::CIRCLE) {
        return SK.GetEntity(distance)->DistanceGetNum();
    } else if(type == Type::ARC_OF_CIRCLE) {
        Vector c  = SK.GetEntity(point[0])->PointGetNum();
        Vector pa = SK.GetEntity(point[1])->PointGetNum();
        return (pa.Minus(c)).Magnitude();
    } else ssassert(false, "Unexpected entity type");
}
```

Possible ways to address this include:

1) threading the current sketch as an extra parameter, ie:

```
double EntityBase::CircleGetRadiusNum(Sketch &SK) const { ...
```

This would affect many, many callsites.

2) Adding a pointer to the containing Sketch on `EntityBase` (and other similar types)

Having children reference there parents like this feels awkward and somewhat messy.

3) Have a global (or more likely thread local) variable that points to the current sketch for a given operational context, and then provide a helper function to access it, ie something like:

```
double EntityBase::CircleGetRadiusNum() const {
    auto SK = useCurrentSketch();
    ...
```

timbod7, 2024-12-29:
Any suggestions from the maintainers here on what would be the best way forward? I'm hesitant to launch into a significant refactor without some hope that the resulting PR would be affected.

phkahler, 2025-01-02:
> Any suggestions from the maintainers here on what would be the best way forward? I'm hesitant to launch into a significant refactor without some hope that the resulting PR would be affected.

@timbod7  You're new to this code, so this scares me a bit. You're asking the right questions and have take time to understand some of the internals, so that's a good sign. As far as your 3 suggestions go, I lean toward 1 - threading the sketch as an extra parameter followed by 2 - Adding a pointer to the containing sketch in relevant objects.

But for starters I'd see how much progress can be made by moving functionality into the sketch itself. As I said above, most of the functionality in generate.cpp could be moved into the sketch. I wouldn't move those functions to another file (they are grouped into that file based on what they do) but I'd move them from the UI class to the sketch class. By doing that, all of those references to SK could be eliminated (the functions would be members of the SK object itself). I'm not sure how many other parts of the code could benefit from small changes like that but IMHO it's worth looking at.

I don't think we need to eliminate the global SS object at this time.

A while back I made a list of all functions in generate.cpp and which variables or other functions they call. That list follow, with "internal" meaning a function called is also in generate.cpp so might become local when move into a sketch:
```
void SolveSpaceUI::MarkGroupDirtyByEntity(hEntity he) {

	SK

void SolveSpaceUI::MarkGroupDirty(hGroup hg, bool onlyThis) {

	SK
	unsaved
	ScheduleGenerateAll()

bool SolveSpaceUI::PruneOrphans() {

	SK
	deleted

bool SolveSpaceUI::GroupsInOrder(hGroup before, hGroup after) {

	SK
	GroupExists()   --- internal

bool SolveSpaceUI::GroupExists(hGroup hg) {

	SK

bool SolveSpaceUI::EntityExists(hEntity he) {

	SK

bool SolveSpaceUI::PruneGroups(hGroup hg) {

	SK
	GroupsInOrder()  --- internal
	EntityExists()   --- internal
	deleted

bool SolveSpaceUI::PruneRequests(hGroup hg) {

	SK
	deleted

bool SolveSpaceUI::PruneConstraints(hGroup hg) {

	SK
	EntityExists()   --- internal
	deleted

void SolveSpaceUI::GenerateAll(Generate type, bool andFindFree, bool genForBBox) {

	SK, type, andFindFree, genForBBox
	GetMilliseconds()
	SS.GW.activeGroup
	SS.exportMode
	chordTol
	PruneOrphans            --- internal
	PruneGroups             --- internal
	PruneRequests           --- internal
	PruneConstraints        --- internal
	ForceReferences()       --- internal
	SolveGroupAndReport()   --- internal
	traced
	GW.Invalidate()
	GW.ClearNonexistentSelectionItems()
	SS.TW.ClearSuper()
	ScheduleShowTW()
	Message()
	deleted
	undo.cnt
	FreeAllTemporary()
	allConsistent;
	[SS.GW](http://ss.gw/).persistentDirty
	SS.centerOfMass.dirty

	dbp()

void SolveSpaceUI::ForceReferences() {

	SK

void SolveSpaceUI::UpdateCenterOfMass() {

	SK, [SS.GW](http://ss.gw/), SS.centerOfMass

void SolveSpaceUI::MarkDraggedParams() {

	SK
	sys.dragged
	SS.GW

void SolveSpaceUI::SolveGroupAndReport(hGroup hg, bool andFindFree) {

	SK
	SolveGroup()   --- internal
	TextWindow::ReportHowGroupSolved()

void SolveSpaceUI::WriteEqSystemForGroup(hGroup hg) {

	SK
	sys
	MarkDraggedParams()   --- internal, needs sys and [SS.GW](http://ss.gw/)

void SolveSpaceUI::SolveGroup(hGroup hg, bool andFindFree) {

	SK
	sys
	WriteEqSystemForGroup   --- internal, needs sys and [SS.GW](http://ss.gw/)
	FreeAllTemporary()

SolveResult SolveSpaceUI::TestRankForGroup(hGroup hg, int *rank) {

	SK
	sys
	WriteEqSystemForGroup   --- internal, needs sys and [SS.GW](http://ss.gw/)
	FreeAllTemporary()

bool SolveSpaceUI::ActiveGroupsOkay() {

	SK
```
It's been a while since I looked at it. Maybe GenerateAll could be split in two functions with only one inside the sketch? There are things like TestRankForGroup that are more group related, so perhaps that could go into the group class - and maybe a parent pointer from the group class to the enclosing sketch would be OK since there aren't many groups (I don't like the parent pointer idea for all the entities and parameters, those objects should stay small IMHO).

This is a potentially a *major* refactoring of core internal structures so planning and incremental change are warranted.
I'm hoping @ruevs will chime in here ;-)

timbod7, 2025-01-05:
Thanks for the response. 

> But for starters I'd see how much progress can be made by moving functionality into the sketch itself. As I said above, most of the functionality in generate.cpp could be moved into the sketch.

I'll have a play with and see if it makes useful progress.

closed 2026-11-05 by phkahler