solvespace/solvespace #81: "Fixed or Smaller (Bigger)" constraint for length and angle

Symbian9, 2016-10-14:
Add constraint that give a way set parametre as ≤ or ≥ to inserted value.

jwesthues, 2016-10-14:
In brief, SolveSpace's solver could be hacked to kind of support inequalities, but it was never made for that. Some earlier threads:

http://solvespace.com/forum.pl?action=viewthread&parent=211
http://solvespace.com/forum.pl?action=viewthread&parent=886

Symbian9, 2016-10-14:
Also, some time its needed prevent to change vector (for example, "line could be enlarged only from start point in left direction, and not in reverse")

Evil-Spirit, 2016-10-14:
@jwesthues, This actually was made and works fine. I even don't see any "hacks" in this solution, just the pure mathematics :)

damian0815, 2017-05-07:
Are they any plans to bring this into the core? I would love this feature.

whitequark, 2017-05-07:
No immediate plans, other tasks have a higher priority.

Symbian9, 2017-07-25:
> This actually was made and works fine. I even don't see any "hacks" in this solution, just the pure mathematics :)

@jwesthues & @Evil-Spirit, so if it work in test, why not implemented yet?

Evil-Spirit, 2017-08-01:
@jwesthues,
@whitequark,
@Symbian9,
Uff... I made it!
https://github.com/Evil-Spirit/solvespace-master/commit/6b7e99b068ac7daeef87d2aab808538fb1e38a4a

Evil-Spirit, 2017-08-01:
`case Op::NORM: return a->PartialWrt(p)->Times(a->Abs()->Sgn());`

> This actually was made and works fine. I even don't see any "hacks" in this solution, just the pure mathematics :)

> I even don't see any "hacks"

> "hacks"

Symbian9, 2017-08-01:
> Uff... I made it!

@Evil-Spirit, you are the man! Do not forget call @probonopd to this party too!

Harvie, 2019-05-20:
Can you please merge this? Looks very nice!

https://github.com/KmolYuan/python-solvespace/issues/3

whitequark, 2019-05-20:
Yeah, it's in my review queue. I did not forget about this patch though.

Harvie, 2019-05-22:
BTW if we could just enter the inequality or range like this, it would be smooth:

[image]

(it already supports some kind of basic math like subtraction, so this can be nice addition)