solvespace/solvespace #82: Represent angle as 0°-360°

Symbian9, 2016-10-14:
Represent angle as 360° for full revolution/circle.
0° must be deferent from 360°, because 0° - define start of revolution/circle, 360° - finished revolution/closed circle.

jwesthues, 2016-10-14:
[0,360) maybe someday, [0,360] no. Think, for example, of what that would mean to the present arc representation.

I'd suggest that you more closely study the mathematical structure of our current solver, and the manner in which entities and constraints are represented. This will help you to suggest enhancements that are practical in that context.

closed 2016-10-14 by jwesthues

Symbian9, 2016-10-14:
No, I mean, that for rotate element we need "Rotate" constraint.
There no need tell "Rotated to 2000°", instead "Rotate to 5.2 revolution".

Angle must be [0, 360] !

jwesthues, 2016-10-14:
Are you referring to step rotating groups? The total angle there can meaningfully equal or exceed a full revolution, since the angles between individual copies are less than that. That works in the obvious way already.

What are you actually trying to do? If you give an example of the problem that you're trying to solve, then perhaps we'll understand better.

Symbian9, 2016-10-14:
@jwesthues, I mean that rotating and angle must be two different constraints https://github.com/solvespace/solvespace/issues/83

P.S.: "New group step and repeat rotating" is from other story there.