solvespace/solvespace #83: "Rotate" constraint for rotated elements

Symbian9, 2016-10-14:
Propose make new constraint for rotating elements instead of angle.

Rotating must be defined as revolutions (3600°/360°=10 rev.).

Angle must be limited to [0°,360°].

There no need use 2000° angle for rotate, instead we must use revolutions (as engines use aka RPM).

P.S. Angle more than 360° only could be usefull for "Screw" constraint, but it's another story.

whitequark, 2016-10-14:
Please describe a specific use case in which this would be useful, and why it cannot be easily done today.

Symbian9, 2016-10-14:
Angle must be used if there no more than one revolution (360°).
Where you need angle 361° or more?
Instead for simulating rotated elements in kinematics.

Currently SolveSpace cant use more than 180°.

whitequark, 2016-10-14:
[image]

Symbian9, 2016-10-14:
@whitequark, step dimension of angle give "pendulum" if try step angle , for example, from 100 to 250. Every 180 it well reverse direction!

jwesthues, 2016-10-14:
With angles in [0,360), the solver would step across, and you could loop arbitrarily many times. That's something that could potentially be done, with some backwards-compatibility issues. [0,360] is not.

closed 2016-10-14 by jwesthues