solvespace/solvespace #84: Add arc in "sketch-in-3d"

Symbian9, 2016-10-14:
Please, add arc available while sketch in 3d.

whitequark, 2016-10-14:
> as arc is similar to circle, then its must be presented in 3d too

I don't see why is it 'must'. What is the use case? You cannot extrude entities from a sketch in 3d, and constraints-wise, point on circle and point on arc perform the exact same thing:

[image]

So there is no reason I see to have arc in 3d.

closed 2016-10-14 by whitequark

Symbian9, 2016-10-15:
Its needed for kinematics design, while there still no way constraint point on directly on circle without so many tricks https://github.com/solvespace/solvespace/issues/60, constraint point to the end of arc could be possible and easier.

jwesthues, 2016-10-17:
The reason why no arc-in-3d entity exists is that the representation used in 2d, by the arc's endpoints and the center of the circle, doesn't have an elegant counterpart in 3d. The points almost always define a plane, except that for 180 degree arcs they don't. The lack of an arc in 3d does bother me, but it bothers me less than any arc-in-3d representation that I could find.

We can of course still sketch any possible arc; a workplane is just required. That workplane's position and orientation don't even need to be fixed, and could be determined by the solver using constraints on the arc.

@Symbian9 I believe that if you submit ideas for enhancement that (a) clearly and concretely describe the problem to be solved, (b) include at least rough implementation details of your proposed solution, and (c) are generally consistent with the overall roadmap that @whitequark has outlined, then they're more likely to be tagged as such and worked on.

whitequark, 2016-10-17:
> The points almost always define a plane, except that for 180 degree arcs they don't.

Don't we have the same problem with e.g. defining a workplane through a point and two line segments, when all of the above lie on the same line?

jwesthues, 2016-10-17:
> Don't we have the same problem with e.g. defining a workplane through a point and two line segments, when all of the above lie on the same line?

There are two versions of this problem: (a) degeneracy in the way that users specify geometry with constraints they have chosen themselves, and (b) degeneracy in the solver's internal representation. The first kind is "real" and inevitable. The second kind is surprising, since it occurs even if the user's constraints aren't degenerate.

Of course, it's easy to find solver representations for the arc that don't suffer from this particular degeneracy. I just couldn't find one that also yielded the quantities that constraints were likely to care about--like the arc's endpoints--without additional (slow, non-convergent, extra-case-filled) math.

Evil-Spirit, 2016-10-17:
@jwesthues, We can just specify normals for an arcs as for circles. But we have to constrain two arc points  with normal basis to lay in u-v plane.

jwesthues, 2016-10-17:
@Evil-Spirit I think that was my favorite, pretty good aside from the large number of extra equations.

There's actually a semi-bug in 3d circles now, since they have 7 DOF, including the meaningless rotational one.

Evil-Spirit, 2016-10-17:
@jwesthues, We can think about valA as param for pt-on-circle. valA can be also constrained to some specific angle, then this rotational degree of freedom becomes meaningful. Also, constraing pt-on-circle to specific angle is the best way to specify circle quadrants or create n-gons. This approach actually was tested by me and produces good result.