solvespace/solvespace #85: Assembly tree for managing display and structure of models.

mmiscool, 2016-10-17:
This would be an enhancement. 

Having the ability to toggle on or off parts in assembly while working on it would be beneficial. 

I suggest this primarily as in my day to day job I use Siemens NX. NX is what I would consider the gold standard in parametric 3d modeling. 

This enhancement would require significant changes to how the interface works but I believe it would be beneficial. 

[image]

In NX there are 3 navigator panels that are particularly of value. 

The first is the part navigator which gives a list of features contained in a particular part. This would  be the sketches and extrudes, ect.

The second is the assembly navigator which shows a tree structure of the assembly allowing for parts to be toggled on or off with the click of a check box next to each.

The third is the constraint navigator that shows a list of constraints and can be sorted by type of constraint or by the par that constraint is linked to. 

I know that these are features that probably won't happen for quite a while but I believe that if they are explored are possible options in the future it might be determine that they have possible value.

Symbian9, 2016-10-17:
I hope, that it would be ready when for next SolveSpace 2.3.

whitequark, 2016-10-17:
1. There will be no SolveSpace 2.3.
2. With the exception of the ability to hide parts of solids, do not expect this to implemented any time soon (or ever).