solvespace/solvespace #91: Better undo history

whitequark, 2016-10-22:
Currently, only 10 are available, which can be rather irritating.

We should:
  * make that number unlimited or near-unlimited (since bare sketches, without mesh, don't really take a lot of memory);
  * save them in the savefile;
  * and have a text screen roughly describing the steps that are being undone, and allowing to switch back and forth, a la Photoshop.

Depends on #75.

wpwrak, 2017-01-31:
Especially the explanation would be very useful also for those "what just happened ?" moments.

E.g., working on a complex design, I often find that a constraint I try to apply seems to get ignored, e.g., because I accidently selected the wrong item. In such cases, I try undo and then do it again. But if my attempts keep getting ignored, I get nervous and quit, reverting to the last known to be good version. Having better feedback on how SolveSpace has interpreted my input should allow for a more confident workflow.

whitequark, 2017-01-31:
@wpwrak I don't really see how this would help. Especially, it is not an improvement over, in your case, clicking on the constraint and seeing what exactly is it attached to--which I believe you can now do with the "explorability" patch.

wpwrak, 2017-01-31:
The explorability SolveSpace has at the moment is great, and I expect it to be even greater with Evil-Spirit's pending further enhancement, but the issue is often that I'm not sure if SolveSpace even performed the operation. I.e., sometimes there is no visual feedback, or maybe there is, but not where or how I expect it. So a quick way to check what happened last should be useful.

whitequark, 2017-01-31:
Ah--fair. That will indeed improve.